fwupd/fwupd #2908: Dell XPS 13 7390: After reboot firmware update is *not* executed but drops in system check

paulmenzel, 2021-02-19:
On the Dell XPS 13 7390 with firmware 1.5.1, using Debian sid/unstable with fwupd 1.5.6, running `fwupdmgr update`, it downloads the update for UEFI firmware 1.7.0 and asks to reboot the system. It reboots, the screen is lit but nothing to see, then the speaker makes a cracking sound and it starts the UEFI firmware system diagnostics.

```
$ fwupdmgr --version
client version:	1.5.6
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.6
$ fwupdmgr get-history 
XPS 13 7390
│
├─KXG60ZNV1T02 NVMe TOSHIBA 1024GB:
│ │   Device ID:          c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│ │   Previous version:   10604106
│ │   Update State:       Success
│ │   Last modified:      2021-02-19 12:44
│ │   GUID:               8a412b95-5c07-5ce4-b63a-f9cbf254dd5d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─XG6 NVMe Device Update:
│       New version:      10604107
│       Remote ID:        lvfs
│       Summary:          XG6 NVMe SSD firmware for Dell PC
│       License:          Proprietary
│       Size:             1.8 MB
│       Created:          2020-08-21
│       Urgency:          Medium
│       Vendor:           KIOXIA Corporation
│       Description:      
│       Optimize SMART thermal temperature implementation.
│     
└─System Firmware:
  │   Device ID:          b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
  │   Previous version:   1.5.1
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2021-02-19 12:53
  │   GUID:               cbe49cca-492b-93e8-b0f4-f693501f271b
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─XPS 13 7390 System Update:
        New version:      1.7.0
        Remote ID:        lvfs
        Summary:          Firmware for the Dell XPS 13 7390
        License:          Proprietary
        Size:             12.3 MB
        Created:          2020-10-22
        Urgency:          Critical
        Vendor:           Dell Inc.
        Flags:            is-upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Firmware updates to address the Intel Technical Advisory.
        • Fixed the issue where the Data wipe option does not show in the BIOS setup after switching between RAID and AHCI mode.
        • Fixed the issue where the external monitor does not power on it is restarted.
        • Fixed the issue where the Linux Vendor Firmware Service (LVFS) does not work when Secure Boot is enabled in the BIOS setup menu.
        
        Some new functionality has also been added:
        
        • Updated the Embedded Controller Engine firmware.
        • Updated the system Power Delivery firmware.
        • Updated Enhanced Pre-Boot System Assessment (ePSA) firmware.
        • Reduced the system fan noise.
        
        Security issues fixed:
        
        • INTEL-TA-00391
        • INTEL-SA-00389
        • INTEL-SA-00381
        • INTEL-SA-00358
        • CVE-2020-26189
        • CVE-2020-8695
        • CVE-2020-8694
        • CVE-2020-8698
        • CVE-2020-8696
        • CVE-2020-0593
        • CVE-2020-0592
        • CVE-2020-0591
        • CVE-2020-0587
        • CVE-2020-12355
        • CVE-2020-12303
        • CVE-2020-8755
        • CVE-2020-8705
        • CVE-2020-8744
        • CVE-2020-8745
```

superm1, 2021-02-19:
It sounds like you have secure boot on. Try to turn it off. There is an issue fixed in this specific update for secure boot with fwupd (see the release notes you pasted).

superm1, 2021-02-19:
duplicate of https://github.com/fwupd/firmware-dell/issues/18

closed 2021-02-19 by superm1