fwupd/fwupd #2910: Setting efi_sbat_distro vars corrupts EFI program

tohojo, 2021-02-19:
**Describe the bug**
I've been having some trouble with the Arch package of fwupd, see https://bugs.archlinux.org/task/69694

The issue I'm seeing is that when trying to launch the EFI binary, I get an "unsupported program" error from systemd-boot or the EFI shell (the system firmware just fails with no error message, so that's why I resorted to those options for launching).

In the bug report linked above we narrowed it down to the `efi_sbat_distro` config. Specifically, the Arch PKGBUILD passes these parameters to 'meson build':

```
        -D efi_sbat_distro_id='arch' \
        -D efi_sbat_distro_summary='Arch Linux' \
        -D efi_sbat_distro_pkgname=${pkgname} \
        -D efi_sbat_distro_version=${pkgver} \
        -D efi_sbat_distro_url="https://archlinux.org/packages/community/x86_64/${pkgname}/"
```

When I recompile the package without those parameters the EFI binary works fine, but with them I get the "unsupported program" error. So I'm opening this issue on request of the distro maintainer.

superm1, 2021-02-19:
@hughsie  Is this the stuff that was fixed by https://github.com/fwupd/fwupd/pull/2902 perhaps?  And then also maybe the root cause of https://github.com/fwupd/firmware-dell/issues/53?

tohojo, 2021-02-19:
Sorry, I should have mentioned, we did take a detour around #2902 and that was part of the issue why things didn't work (see discussion in the bug I linked above). However, even when backporting commit ec52942bc5ad9e80c8bf905e6d0a0ccf218acc1f (which the Arch package does), the issue remains when the `efi_sbat_distro` vars are set...

superm1, 2021-02-19:
@martinezjavier  ^

superm1, 2021-02-19:
@martinezjavier is this perhaps an ordering problem?  Does the `sbat` section need to be placed at the end?  Something like this (untested - no idea if this still or now works):

```
diff --git a/plugins/uefi-capsule/efi/generate_binary.py b/plugins/uefi-capsule/efi/generate_binary.py
index 06a43725..bd8b47bf 100755
--- a/plugins/uefi-capsule/efi/generate_binary.py
+++ b/plugins/uefi-capsule/efi/generate_binary.py
@@ -84,8 +84,6 @@ def _run_objcopy(args):
             "-j",
             ".text",
             "-j",
-            ".sbat",
-            "-j",
             ".sdata",
             "-j",
             ".data",
@@ -95,6 +93,8 @@ def _run_objcopy(args):
             ".dynsym",
             "-j",
             ".rel*",
+            "-j",
+            ".sbat",
             tfd.name,
             args.outfile,
         ]
```

chrisccoulson, 2021-02-19:
I've just done a local build of fwupd with a SBAT section, and it seems that the .sbat section in the resulting binary is not correctly aligned:
```
$ readpe plugins/uefi-capsule/efi/fwupdx64.efi 
DOS Header                                    
    Magic number:                    0x5a4d (MZ)          
    Bytes in last page:              144       
    Pages in file:                   3                    
    Relocations:                     0             
    Size of header in paragraphs:    4
    Minimum extra paragraphs:        0                                                                                 
    Maximum extra paragraphs:        65535                                                                             
    Initial (relative) SS value:     0                                                                                 
    Initial SP value:                0xb8 
    Initial IP value:                0        
    Initial (relative) CS value:     0                    
    Address of relocation table:     0x40     
    Overlay number:                  0                                                                                 
    OEM identifier:                  0             
    OEM information:                 0    
    PE header offset:                0x80                                                                              
COFF/File header            
    Machine:                         0x8664 IMAGE_FILE_MACHINE_AMD64                                                   
    Number of sections:              8                                                                                 
    Date/time stamp:                 0 (Tue, 07 Jan 6900713 23:40:48 UTC)
    Symbol Table offset:             0xc600                                                                            
    Number of symbols:               322        
    Size of optional header:         0xf0    
    Characteristics:                 0x206                
    Characteristics names                      
                                         IMAGE_FILE_EXECUTABLE_IMAGE
                                         IMAGE_FILE_LINE_NUMS_STRIPPED
                                         IMAGE_FILE_DEBUG_STRIPPED
Optional/Image header                                                                                                  
    Magic number:                    0x20b (PE32+)                                                                     
    Linker major version:            2         
    Linker minor version:            34         
    Size of .text section:           0x7600               
    Size of .data section:           0x4800    
    Size of .bss section:            0                                                                                 
    Entrypoint:                      0x4000        
    Address of .text section:        0x4000
    ImageBase:                       0                                                                                 
    Alignment of sections:           0x1000                                                                            
    Alignment factor:                0x200                                                                             
    Major version of required OS:    0                                                                                 
    Minor version of required OS:    0              
    Major version of image:          0
    Minor version of image:          0
    Major version of subsystem:      0
    Minor version of subsystem:      0
    Size of image:                   0x13000
    Size of headers:                 0x2c8
    Checksum:                        0x15bf0
    Subsystem required:              0xa (IMAGE_SUBSYSTEM_EFI_APPLICATION)
    DLL characteristics:             0
    DLL characteristics names
    Size of stack to reserve:        0
    Size of stack to commit:         0
    Size of heap space to reserve:   0
    Size of heap space to commit:    0
Data directories
    Directory
        IMAGE_DIRECTORY_ENTRY_BASERELOC: 0xc000 (10 bytes)
import directory not found
export directory not found
Sections
    Section
        Name:                            .sbat
        Virtual Address:                 0
        Physical Address:                0xc9
[... 65 lines cut ...]
    Section
        Name:                            /4
        Virtual Address:                 0x11e70
        Physical Address:                0x18
        Size:                            0x200 (512 bytes)
        Pointer To Data:                 0xbe70
        Relocations:                     0
        Characteristics:                 0x40400040
        Characteristic Names
                                             IMAGE_SCN_CNT_INITIALIZED_DATA
                                             IMAGE_SCN_MEM_READ
    Section
        Name:                            .dynsym
        Virtual Address:                 0x12000
        Physical Address:                0x270
        Size:                            0x400 (1024 bytes) 
        Pointer To Data:                 0xc200
        Relocations:                     0
        Characteristics:                 0x40400040
        Characteristic Names
                                             IMAGE_SCN_CNT_INITIALIZED_DATA
                                             IMAGE_SCN_MEM_READ
```

Note the on-disk size of 520-bytes and the offset of 0x2c8 in to the image. For a valid PE image, both of these need to be a multiple of the file alignment (0x200 in this case). I guess it's possible that this might trip up the loader, although I'm speculating at this point and it's a bit late in the evening for me to start testing whether this is the issue.

hughsie, 2021-02-20:
@vathpela any ideas?

martinezjavier, 2021-02-20:
@superm1 I don’t think the sections have to be in a certain order. It seems to me that @chrisccoulson is right and this may be an alignment problem.

I thought that objcopy would do the proper alignment but @vathpela mentioned that you need to do it explicitly (i.e: objcopy --set-section-alignment .sbat=512 --add-section .sbat=sbat.csv).

We had also had rhboot/shim#281 reported, which I believe may be related with alignment issues as well.

I don’t have my laptop at hand now but I can take a look to both issues later.

hughsie, 2021-02-20:
@chrisccoulson can you try with https://github.com/fwupd/fwupd/pull/2913 please? Thanks.

chrisccoulson, 2021-02-20:
> @chrisccoulson can you try with #2913 please? Thanks.

Thanks - I managed to confirm the bug this morning. That PR looks ok, but it's not quite sufficient. There's still another issue that's causing the loader to think that the binary is invalid, and I'm trying to figure out why.

martinezjavier, 2021-02-20:
@chrisccoulson thanks a lot for looking at this. Did you see the shim issue I mentioned? Do you think that may be the same issue ?

chrisccoulson, 2021-02-20:
Ah, I think the VirtualAddress = 0 is problematic too as the loader puts the header there, so I suspect it fails here in edk:

https://github.com/tianocore/edk2/blob/4f4d862c1c7232a18347616d94c343c929657fdb/MdePkg/Library/BasePeCoffLib/BasePeCoff.c#L478

hughsie, 2021-02-20:
Does moving the .sbat section in the objcopy argv make it nonzero? I don't have s laptop here to look.

chrisccoulson, 2021-02-20:
> Does moving the .sbat section in the objcopy argv make it nonzero? I don't have s laptop here to look.

It doesn't. It seems like you can use objcopy to move sections to different virtual addresses, but then you need to pick a virtual address to move it to which requires some idea of the section layout created by the linker.

I think the way to fix this is to do what shim does - create an object file with the .sbat section and then have the linker take care of giving it a more appropriate virtual address. I've got a WIP branch which does this, and I've verified that I can execute fwupdx64.efi with the edk2 loader. I'll submit a PR shortly once I've cleaned it up.

hughsie, 2021-02-20:
> create an object file with the .sbat section

We had that originally, but it didn't work for aarch64 as it didn't know about the specific pe object type. iiuc.

chrisccoulson, 2021-02-20:
> > create an object file with the .sbat section
> 
> We had that originally, but it didn't work for aarch64 as it didn't know about the specific pe object type. iiuc.

Oh, interesting. I wonder how this works in shim. In any case, I created https://github.com/fwupd/fwupd/pull/2914, which seems to do the trick on x86-64, at least.

vathpela, 2021-02-20:
> > create an object file with the .sbat section
> 
> We had that originally, but it didn't work for aarch64 as it didn't know about the specific pe object type. iiuc.

Yeah, but you link that in to your .so as elf, before the objcopy invocation.

martinezjavier, 2021-02-20:
> Yeah, but you link that in to your .so as elf, before the objcopy invocation.

Indeed. What we originally had (that didn't work for aarch64) was to add a section to the PE binary directly. Then we were adding it to the `fwupd.so` after linking, but that also was causing issues due misalignment.

It's a pity that we need to pull all these linker scripts only for this...

martinezjavier, 2021-02-21:
I [commented](https://github.com/fwupd/fwupd/pull/2914#issuecomment-782780063) in #2914 about possible solutions that wouldn't require adding a bespoke linker scripts to `fwupd`. But I'm unsure how unreliable would be to make the assumptions mentioned there.

Basically the other options I mentioned are:

1) Set a custom address for the `.sbat` section (using the `objcopy --change-section-vma` option) in the VMA range that's not used when linking with the default linker script from `gnu-efi`.

2) Modify the default `gnu-efi` linker script to make it aware of the `.sbat` section, to avoid adding a custom one to `fwupd`.

martinezjavier, 2021-02-21:
@hughsie maybe a middle ground could be to have a set of `efi_sbat_section_align` and `efi_sbat_section_addr` meson options and use sane-ish defaults (i.e: `512` and `0x0x2000`) but allow distros to override if they don't ship the default `efi.lds` script ?

I can propose a PR on that vein if you think is worth it.

hughsie, 2021-02-21:
> but allow distros to override

Out of interest, why would 0x2000 be unsuitable for any other distros? Do distros ship different efi.lds files compared to Fedora? I've updated https://github.com/fwupd/fwupd/pull/2913 with change-section-vma.

> Modify the default gnu-efi linker efi.lds script

I would much prefer this, and then perhaps check for the .sbat section in the fwupd meson.build and refuse to build if the sbat distro ID is set.

martinezjavier, 2021-02-21:
> Out of interest, why would 0x2000 be unsuitable for any other distros? Do distros ship different efi.lds files compared to Fedora? I've updated #2913 with change-section-vma.

We have a few downstream patches in `gnu-efi` (although I think are mostly fixes that were never picked by upstream and not changes to the linker scripts), I don't know what's the state of other distros. That's why I proposed the option to allow distros to override a default value if wasn't suitable for this. Maybe it doesn't make sense, my knowledge of all this is quite superficial.

> I would much prefer this, and then perhaps check for the .sbat section in the fwupd meson.build and refuse to build if the sbat distro ID is set.

Yes, I was also leaning towards this option. After all, the `.sbat` data section is not `fwupd` specific, so it is reasonable to expect that the system wide EFI linker script would be aware of that section and where to put it.

hughsie, 2021-02-21:
> I was also leaning towards this option

So what about we grep for `.sbat` in `/usr/lib/gnuefi/x64/efi.lds` and if not found we fall back to adding change-section-vma? If a distro is laying out the binary differently then they can then add the patch to their custom efi.lds

martinezjavier, 2021-02-21:
> So what about we grep for `.sbat` in `/usr/lib/gnuefi/x64/efi.lds` and if not found we fall back to adding change-section-vma? If a distro is laying out the binary differently then they can then add the patch to their custom efi.lds

That sounds reasonable to me.

hughsie, 2021-02-21:
Do we still need set-section-alignment if efi.lds has a .sbat section?

martinezjavier, 2021-02-21:
> Do we still need set-section-alignment if efi.lds has a .sbat section?

No, we don't need any special handling of the section if the linker script takes care of everything.

hughsie, 2021-02-21:
Okay, PR updated.

martinezjavier, 2021-02-21:
@hughsie regardless of the final solution I think that the part from @chrisccoulson patch that builds an object file with the `.sbat` section and then link it instead of using a temp file is something that we want.

hughsie, 2021-02-21:
My gut instinct is to merge the PR from @chrisccoulson (once the nits are fixed) and then lobby the gnu-efi project to include the new .sbat in the linker scripts. Once that's upstream and there's a tarball release of that we up the dep in fwupd and delete the linker scripts we have in-tree.

martinezjavier, 2021-02-21:
@hughsie agreed.

superm1, 2021-03-08:
This was fixed in fwupd 1.5.7.

closed 2021-03-08 by superm1