fwupd/fwupd #2915: Fails to recognize a second Logitech Unifying Receiver

thx1111, 2021-02-21:
Arch Linux
fwupd 1.5.5-1

$ fwupdmgr --version
client version: 1.5.5
compile-time dependency versions
        gusb:   0.3.5

Unknown                  [***************************************]
daemon version: 1.5.5

$ solaar show       
Unifying Receiver
  Device path  : /dev/hidraw0
  USB id       : 046d:C52B
  Serial       : xxxxxxxx
    Firmware   : 12.10.B0032
    Bootloader : 02.14
    Other      : AA.AA
  Has 5 paired device(s) out of a maximum of 6.
  Notifications: wireless (0x000100)
  Device activity counters: 3=70, 5=192
...
Unifying Receiver
  Device path  : /dev/hidraw7
  USB id       : 046d:C52B
  Serial       : yyyyyyyy
    Firmware   : 12.01.B0019
    Bootloader : 02.14
  Has 0 paired device(s) out of a maximum of 6.
  Notifications: wireless (0x000100)
  Device activity counters: (empty)

$ fwupdmgr get-devices --show-all-devices
...
[ only the first Unifying Receiver is shown ]

$ lsusb
Bus 007 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver

Of course, either device can be unplugged to work with the remaining device, so maybe this is more of a feature request.  In contrast, for instance, though, multiple hard drives are detected without problem.

Why is more than one Unifying Receiver plugged, you ask?  I simply wanted to update and configure the second receiver before moving it to another machine.  Instead, I had to remove the configured device that connected the keyboard and mouse I was using, and dig-out a wired keyboard and mouse to use while the second receiver was getting the update.  It's just inconvenient.

hughsie, 2021-02-21:
Hmm, this worked for me the last time I checked. Can you attach the output of `sudo fwupdtool get-devices --plugins logitech_hidpp --verbose` when both are installed please.

thx1111, 2021-02-21:
Thanks.  Hmm - I expect the part of interest may be:
```
...
15:04:18:0608 FuDevice             using b9d17735cbfcab51db9a9872394bd3210a4148c7 for DEVPATH=/devices/pci0000:00/0000:00:1d.1/usb7/7-2/7-2:1.2                             
15:04:18:0816 FuPlugin             emit added from logitech_hidpp: b9d17735cbfcab51db9a9872394bd3210a4148c7                                                                 
15:04:18:0816 FuDeviceList         ignoring device b9d17735cbfcab51db9a9872394bd3210a4148c7 [logitech_hidpp:FuLogitechHidPpRuntime] existing device 6a773ff28060afc71fca7cf0c24c2dc39816e68f [logitech_hidpp:FuLogitechHidPpRuntime] already exists  
...
```
The "ignoring device ..." line displays in red.

If the second Unifying Receiver is plugged into a USB hub, instead of a socket on the motherboard, the path and device ID  change, but there is again the corresponding "ignoring device ..." line.

hughsie, 2021-02-22:
> If the second Unifying Receiver is plugged into a USB hub,

The reason it's not adding is because logitech_hidpp does not use FWUPD_DEVICE_FLAG_NO_GUID_MATCHING and the GUIDs constructed from the second device match exactly from the first as they're constructed from the VID and PID, rather than the device path.

I'm working on a patch now.

closed 2021-02-25 by hughsie