fwupd/fwupd #2916: Does not find ESP on MBR only system

paulmenzel, 2021-02-21:
On Debian sid/unstable with fwupd* 1.5.6-1, `fwupdmgr get-devices` returns `Unable to determine EFI system partition location` on a system with only a valid MBR and GPT, and only lists the hard disk drive. The ESP has the type 0xef in the MBR.

```
$ fwupdmgr --version
client version:	1.5.6
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.6
$ sudo /usr/bin/fwupdtool esp-list --verbose
13:49:52:0409 FuDebug              Verbose debugging enabled (on console 1)
13:49:52:0426 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda1, type: 0xef, internal: 1, fs: vfat
13:49:52:0430 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda2, type: 0x83, internal: 1, fs: ext4
13:49:52:0433 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda3, type: 0x83, internal: 1, fs: crypto_LUKS
13:49:52:0433 FuMain               no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b, falling back to ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
13:49:52:0446 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda1, type: 0xef, internal: 1, fs: vfat
13:49:52:0450 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda2, type: 0x83, internal: 1, fs: ext4
13:49:52:0452 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda3, type: 0x83, internal: 1, fs: crypto_LUKS
no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b: no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
$ sudo fdisk /dev/sda

Welcome to fdisk (util-linux 2.36.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): p
Disk /dev/sda: 931,51 GiB, 1000204886016 bytes, 1953525168 sectors
Disk model: CT1000MX500SSD4
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: dos
Disk identifier: 0x64fda70b

Device     Boot   Start        End    Sectors  Size Id Type
/dev/sda1  *       2048     526335     524288  256M ef EFI (FAT-12/16/32)
/dev/sda2        526336    1050623     524288  256M 83 Linux
/dev/sda3       1050624 1953525167 1952474544  931G 83 Linux
```

```
$ fwupdmgr get-devices --show-all-devices
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
MS-7A37
│
├─300 Series Chipset SATA Controller:
│     Device ID:          5299e802217cae06dbae3c38b1311bee84b7d318
│     Current version:    02
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              2752f8e3-f453-5b2c-b26e-fbe10c23be07
│                         8b139035-0f55-5b27-b1bc-c2c0064addd4
│                         ef4a96a0-45d1-5cff-90d3-e6f387d130e7
│                         17f97e61-e971-53d5-bb9e-6bcef9c6c23f
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─AMD Ryzen 3 2200G with Radeon Vega Graphics    :
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         375b07af-58d5-5172-ab62-510c2229bf79
│                         2f8267cd-b115-5138-b04d-3cb13badf673
│     Device Flags:       • Internal device
│
└─CT1000MX500SSD4:
      Device ID:          bf97ef88a4eda3055c079a95e0636fe2887d5436
      Zusammenfassung:    ATA Drive
      Current version:    M3CR020
      Vendor:             Micron (ATA:0x1344|OUI:00a075)
      GUIDs:              7fe72eb9-09fe-510e-a062-eb07f185035b
                          1d4f3437-add2-559d-bd88-0a10880e59d3
                          49811b52-642d-5307-bc6f-32e2033fb26a
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```

hughsie, 2021-02-21:
What's the output of `udisksctl dump` please?

paulmenzel, 2021-02-21:
```
$ udiskctl dump
/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks1
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                vfat
                                ntfs
                                exfat
                                xfs
                                reiserfs
                                nilfs2
                                btrfs
                                minix
                                udf
                                f2fs
                                swap
    Version:                    2.9.2

/org/freedesktop/UDisks2/block_devices/dm_2d0:
  org.freedesktop.UDisks2.Block:
    Configuration:              [('fstab', {'fsname': <b'UUID=8883f733-2248-47e0-90b9-ee5384f18d62'>, 'dir': <b'/'>, 'type': <b'auto'>, 'opts': <b'def
aults,errors=remount-ro'>, 'freq': <0>, 'passno': <1>})]
    CryptoBackingDevice:        '/org/freedesktop/UDisks2/block_devices/sda3'
    Device:                     /dev/dm-0
    DeviceNumber:               65024
    Drive:                      '/'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          false
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-dm-name-sda3_crypt
    IdLabel:                    
    IdType:                     ext4
    IdUUID:                     8883f733-2248-47e0-90b9-ee5384f18d62
    IdUsage:                    filesystem
    IdVersion:                  1.0
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/mapper/sda3_crypt
    ReadOnly:                   false
    Size:                       999664869376
    Symlinks:                   /dev/disk/by-id/dm-name-sda3_crypt
                                /dev/disk/by-id/dm-uuid-CRYPT-LUKS1-78c5b7b8e8c644b6ae783124a3308f3e-sda3_crypt
                                /dev/disk/by-uuid/8883f733-2248-47e0-90b9-ee5384f18d62
                                /dev/mapper/sda3_crypt
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.Filesystem:
    MountPoints:        /
    Size:               999664869376

/org/freedesktop/UDisks2/block_devices/sda:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/sda
    DeviceNumber:               2048
    Drive:                      '/org/freedesktop/UDisks2/drives/CT1000MX500SSD4_1817E1392ADB'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
[... 279 lines cut ...]
    AamSupported:                               false
    AamVendorRecommendedValue:                  0
    ApmEnabled:                                 true
    ApmSupported:                               true
    PmEnabled:                                  true
    PmSupported:                                true
    ReadLookaheadEnabled:                       true
    ReadLookaheadSupported:                     true
    SecurityEnhancedEraseUnitMinutes:           2
    SecurityEraseUnitMinutes:                   2
    SecurityFrozen:                             true
    SmartEnabled:                               true
    SmartFailing:                               false
    SmartNumAttributesFailedInThePast:          0
    SmartNumAttributesFailing:                  0
    SmartNumBadSectors:                         0
    SmartPowerOnSeconds:                        17251200
    SmartSelftestPercentRemaining:              0
    SmartSelftestStatus:                        success
    SmartSupported:                             true
    SmartTemperature:                           320.15000000000003
    SmartUpdated:                               1613926392
    WriteCacheEnabled:                          true
    WriteCacheSupported:                        true
```

closed 2021-02-22 by hughsie