fwupd/fwupd #2926: 04f3:0c1a ELAN:Fingerprint unusable after upgrade

filariow, 2021-02-23:
I've updated the firmware of my Elan Fingerprint and from that moment it is unusable.
It won't work neither in the UEFI BIOS.

In my Arch Linux the module is still recognized by `lsusb`
```
Bus 001 Device 009: ID 04f3:0c1a Elan Microelectronics Corp. ELAN:Fingerprint
```

If I run `fwupdmgr get-devices --show-all-devices` or `fwupdagent get-devices` I can not find the Elan device.
Any advice on how to force  firmware

superm1, 2021-02-23:
Can you please share `fwupdmgr get-history`?

filariow, 2021-02-23:
i'm sorry but I lost it trying to fix. Is anything else I can do?

hughsie, 2021-02-23:
> I've updated the firmware of my Elan Fingerprint

Where did you get the firmware _from_? All the Elan firmware is currently embargoed on the LVFS.

filariow, 2021-02-23:
I've simply run

hughsie, 2021-02-23:
I cant see any firmware on the LVFS with a GUID that matches `USB\VID_04F3&PID_0C1A`

> The output was not nice

Can you remember anything at all about the update? How did you loose the history? I'm struggling to work out what happened.

filariow, 2021-02-23:
I was trying to update the firmware for my fingerprint reader in order to have it to work on my linux pc.
I tried the following commands:
- `fwupdmgr refresh`
- `fwupdmgr get-devices`: it listed the ELan device also
- `fwupdmgr get-updates`
- `fwupdmgr update`
- `fwupdmgr update ELAN_ID`
- `fwupdmgr verify-update ELAN_ID` 

after I messed a little with this command I'm no more able to list the Elan device with `fwupdmgr get-devices`

hughsie, 2021-02-23:
> I've a Xiaomi Mi Notebook Pro 2017

That makes it even more confusing, as Xiaomi isn't even on the LVFS.

filariow, 2021-02-23:
Ok, that's really not encouraging... Can you suggest me something to do or to look at to fix my touchpad?

hughsie, 2021-02-23:
> fix my touchpad...

Is this a touchpad or fingerprint reader? Did firmware actually install? Was the device working before you tried to update the firmware?

filariow, 2021-02-23:
It's the touchpad and the fingerprint, they are a single piece of hardware.
From my understand it did not install.
Yes it was working perfectly. The fingerprint was not working correctly under Linux (Manjaro and Arch) but was good in Windows 10

hughsie, 2021-02-23:
Does the fingerprint reader work in Windows now? I'm really not sure how to debug this further to be honest.

filariow, 2021-02-23:
No, neither the fingerprint nor the touchpad. And the touchpad is also not working in UEFI BIOS where it was supposed to work.

Is there a way I can force a flash with `fwupdmgr` for the Elan device?

hughsie, 2021-02-23:
> Is there a way I can force a flash with fwupdmgr for the Elan device

No, as fwupd doesn't match your hardware to any firmware on the LVFS. Some things to try and work out what actually happened -- what's the output of:

 * `sudo fwupdtool hwids`
 * `sudo fwupdtool get-devices --plugins elantp --verbose`

It's possible your touchpad is in i2c recovery mode. You could try adding your hwid (any of them!) to the elantp.quirk file (and installing fwupd) and then trying `sudo fwupdtool attach --plugins elantp --verbose`

filariow, 2021-02-23:
Here the output of the first two commands.

```
➜ sudo fwupdtool hwids
Computer Information
--------------------
BiosVendor: INSYDE Corp.
BiosVersion: XMAKB5R0P0E07
BiosMajorRelease: 106
BiosMinorRelease: 117
Manufacturer: Timi
Family: Timibook
ProductName: TM1701
ProductSku: TM1701-16771
EnclosureKind: a
BaseboardManufacturer: Timi
BaseboardProduct: TM1701

Hardware IDs
------------
{3bc8b32b-8116-54a0-8567-139942bf8890}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{62ebc292-93e8-524f-b0aa-cce40ee98379}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{415bae7a-e0bb-5fce-889c-80d4ed603fc7}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{fa9259f0-35c3-5446-a488-2a0716bb7d73}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{a20293eb-4398-51d1-9c71-e2c3bd224494}   <- Manufacturer + Family + ProductName + ProductSku
{b9820218-8772-5955-b94a-098dec16bf50}   <- Manufacturer + Family + ProductName
{9f1e06a6-55a0-5889-a246-e250f3e4141e}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{0a82f65a-b2e7-5293-ba88-699b62bb093e}   <- Manufacturer + ProductSku
{64deec0f-51a1-5278-8aaa-a7d5a0974fcf}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{c2462c50-f34a-5369-bf04-ed9cd8c3f3ea}   <- Manufacturer + ProductName
{37c04c7d-be39-5a98-95f6-3a86870cfc37}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{fdaa0f43-caaa-580e-a57d-4297952a6115}   <- Manufacturer + Family
{b5b1baa0-8af1-571e-8040-85532425cbea}   <- Manufacturer + EnclosureKind
{a65e0573-82d6-5f9e-bc7b-caceeaab46d7}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{702303a2-07c5-5620-813f-34560cb4a9be}   <- Manufacturer

~
➜ sudo fwupdtool get-devices --plugins elantp --verbose
11:58:28:0270 FuDebug              Verbose debugging enabled (on console 1)
11:58:28:0275 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
11:58:28:0275 FuConfig             loading config values from /etc/fwupd/daemon.conf
11:58:28:0282 FuConfig             using autodetected max archive size 18.4 EB
11:58:28:0283 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
11:58:28:0284 FuRemoteList         loading /usr/share/fwupd/metainfo
11:58:28:0284 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
11:58:28:0284 XbSilo               file: 23929d1e-3a1f-2197-4412-cecd590512e8, current:23929d1e-3a1f-2197-4412-cecd590512e8, cached: (null)
11:58:28:0284 XbSilo               loading silo with file contents
11:58:28:0284 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
11:58:28:0284 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
11:58:28:0285 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
11:58:28:0285 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
11:58:28:0285 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
11:58:28:0285 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
11:58:28:0285 FuRemoteList         ignoring unfound remote fwupd
11:58:28:0285 FuRemoteList         ordering lvfs-testing=lvfs+1
11:58:28:0285 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
11:58:28:0285 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
11:58:28:0286 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
11:58:28:0286 Jcat                 loaded 1 certificates
11:58:28:0286 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
11:58:28:0286 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:58:28:0286 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
11:58:28:0286 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
11:58:28:0286 Jcat                 loaded 1 certificates
11:58:28:0286 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:58:28:0286 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
11:58:28:0286 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
11:58:28:0286 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
11:58:28:0293 FuEngine             client certificate exists and working
11:58:28:0293 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
[... 112 lines cut ...]
ElantpIcPageCount = 0x380
ElantpIapPassword = 0x1EA5

[DeviceInstanceId=ELANTP\ICTYPE_10]
ElantpIcPageCount = 0x400
ElantpIapPassword = 0x1EA5

[DeviceInstanceId=ELANTP\ICTYPE_11]
ElantpIcPageCount = 0x500
ElantpIapPassword = 0x1EA5

[DeviceInstanceId=ELANTP\ICTYPE_13]
ElantpIcPageCount = 0x800
ElantpIapPassword = 0x1EA5

[DeviceInstanceId=ELANTP\ICTYPE_14]
ElantpIcPageCount = 0x400
ElantpIapPassword = 0x1EA5

[DeviceInstanceId=ELANTP\ICTYPE_15]
ElantpIcPageCount = 0x400
ElantpIapPassword = 0x1EA5
```

Thank you very much, for your time and patience! 🙇🏾‍♂️

filariow, 2021-02-23:
Furthermore I can see the following logs using dmesg

```
➜ sudo dmesg | grep i2c
[11379.308651] i2c_hid i2c-ETD2303:00: supply vdd not found, using dummy regulator
[11379.308680] i2c_hid i2c-ETD2303:00: supply vddl not found, using dummy regulator
[11379.445998] input: ETD2303:00 04F3:0400 as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ETD2303:00/0018:04F3:0400.000C/input/input39
[11379.446583] hid-generic 0018:04F3:0400.000C: input,hidraw0: I2C HID v1.00 Mouse [ETD2303:00 04F3:0400] on i2c-ETD2303:00
```

hughsie, 2021-02-23:
I'm guessing here, as I don't actually know what happened. I think you need to add:

[HwId=fa9259f0-35c3-5446-a488-2a0716bb7d73]
Flags = elantp-recovery

to /usr/share/fwupd/quirks.d/elantp.quirk then restart your computer, then see if the device appears in `sudo fwupdtool get-devices --plugins elantp --verbose`

filariow, 2021-02-23:
Did it, but it seems not to be working. Should I try with different `HwId`s?

```
➜ sudo fwupdtool get-devices --plugins elantp --verbose
12:29:11:0840 FuDebug              Verbose debugging enabled (on console 1)
12:29:11:0851 FuConfig             loading config values from /etc/fwupd/daemon.conf
12:29:11:0851 FuConfig             using autodetected max archive size 18.4 EB
12:29:11:0853 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
12:29:11:0853 FuRemoteList         loading /usr/share/fwupd/metainfo
12:29:11:0853 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:29:11:0853 XbSilo               file: 23929d1e-3a1f-2197-4412-cecd590512e8, current:23929d1e-3a1f-2197-4412-cecd590512e8, cached: (null)
12:29:11:0853 XbSilo               loading silo with file contents
12:29:11:0853 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
12:29:11:0854 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
12:29:11:0854 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
12:29:11:0854 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
12:29:11:0854 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
12:29:11:0854 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
12:29:11:0854 FuRemoteList         ignoring unfound remote fwupd
12:29:11:0854 FuRemoteList         ordering lvfs-testing=lvfs+1
12:29:11:0854 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
12:29:11:0854 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
12:29:11:0854 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:29:11:0855 Jcat                 loaded 1 certificates
12:29:11:0855 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:29:11:0855 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:29:11:0855 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
12:29:11:0855 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:29:11:0855 Jcat                 loaded 1 certificates
12:29:11:0855 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:29:11:0855 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:29:11:0855 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
12:29:11:0855 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
12:29:11:0864 FuEngine             client certificate exists and working
12:29:11:0864 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
12:29:11:0864 FuHistory            got schema version of 6
12:29:11:0864 FuHwids              smbios property Manufacturer=Timi
12:29:11:0864 FuHwids              smbios property EnclosureKind=a
12:29:11:0864 FuHwids              smbios property Family=Timibook
12:29:11:0864 FuHwids              smbios property ProductName=TM1701
12:29:11:0864 FuHwids              smbios property ProductSku=TM1701-16771
12:29:11:0864 FuHwids              smbios property BiosVendor=INSYDE Corp.
12:29:11:0864 FuHwids              smbios property BiosVersion=XMAKB5R0P0E07
12:29:11:0864 FuHwids              smbios property BiosMajorRelease=6a
12:29:11:0864 FuHwids              smbios property BiosMinorRelease=75
12:29:11:0864 FuHwids              smbios property BaseboardManufacturer=Timi
12:29:11:0864 FuHwids              smbios property BaseboardProduct=TM1701
12:29:11:0869 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
12:29:11:0869 XbSilo               file: c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, current:c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, cached: (null)
12:29:11:0869 XbSilo               loading silo with file contents
12:29:11:0870 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
12:29:11:0870 XbSilo               file: 5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, current:5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, cached: (null)
12:29:11:0870 XbSilo               loading silo with file contents
12:29:11:0871 FuEngine             528 components now in silo
12:29:11:0874 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_elantp.so)
12:29:11:0875 FuPlugin             added udev subsystem watch of i2c-dev
12:29:11:0875 FuPlugin             added udev subsystem watch of hidraw
12:29:11:0875 FuEngine             plugins disabled: iommu, linux_tainted, synaptics_rmi, synaptics_mst, csr, cpu, system76_launch, wacom_raw, pci_bcr, upower, colorhug, ep963x, dell_dock, tpm_eventlog, jabra, ccgx, dell_esrt, hailuck, linux_sleep, dell, thunderbolt, ebitdo, goodixmoc, fresco_pd, uefi_recovery, vli, bios, bcm57xx, synaptics_prometheus, fastboot, thelio_io, uefi_capsule, uefi_pk, emmc, uefi_dbx, steelseries, amt, wacom_usb, superio, nvme, logind, acpi_facp, pci_mei, dfu, flashrom, nitrokey, acpi_dmar, linux_lockdown, optionrom, rts54hub, logitech_hidpp, pixart_rf, redfish, ata, tpm, altos, cros_ec, rts54hid, solokey, linux_swap, msr, synaptics_cxaudio
12:29:11:0875 FuEngine             Emitting PropertyChanged('Status'='loading')
Loading…                 [-                                      ]12:29:11:0875 FuEngine             using plugins: elantp
12:29:11:0886 FuPlugin             fu_plugin_device_created(elantp)
12:29:11:0887 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
No detected devices
12:29:11:0891 FuMain               No detected devices
```

hughsie, 2021-02-23:
What does `cat /sys/class/i2c-adapter/i2c-*/name` print? I have no idea if arch loads enough stuff to actually populate those device entries.

filariow, 2021-02-23:
Here it is!

```
➜ sudo cat /sys/class/i2c-adapter/i2c-*/name
SMBus I801 adapter at 4040
i915 gmbus dpb
i915 gmbus dpd
AUX A/port A
AUX C/port C
Synopsys DesignWare I2C adapter
Synopsys DesignWare I2C adapter
nvkm-0000:01:00.0-bus-0001
nvkm-0000:01:00.0-bus-0002
nvkm-0000:01:00.0-bus-0004
nvkm-0000:01:00.0-aux-0004
nvkm-0000:01:00.0-bus-0005
nvkm-0000:01:00.0-aux-0005
i915 gmbus dpc
```

hughsie, 2021-02-23:
Can you just delete the entire fu_plugin_device_created function in fu-plugin-elantp.c, recompile, install and then `sudo fwupdtool get-devices --plugins elantp --verbose` again?

filariow, 2021-02-23:
Here it is!

```
➜ git diff | cat
diff --git a/plugins/elantp/fu-plugin-elantp.c b/plugins/elantp/fu-plugin-elantp.c
index 8fcb6e8f..70ddee53 100644
--- a/plugins/elantp/fu-plugin-elantp.c
+++ b/plugins/elantp/fu-plugin-elantp.c
@@ -12,20 +12,6 @@
 #include "fu-elantp-hid-device.h"
 #include "fu-elantp-i2c-device.h"

-gboolean
-fu_plugin_device_created (FuPlugin *plugin, FuDevice *dev, GError **error)
-{
-       if (fu_device_get_specialized_gtype (dev) == FU_TYPE_ELANTP_I2C_DEVICE &&
-           !fu_plugin_has_custom_flag (plugin, "elantp-recovery")) {
-               g_set_error_literal (error,
-                                    FWUPD_ERROR,
-                                    FWUPD_ERROR_NOT_SUPPORTED,
-                                    "not required");
-               return FALSE;
-       }
-       return TRUE;
-}
-
 void
 fu_plugin_init (FuPlugin *plugin)
 {

fwupd on  master [!?] at ☸️  pga-aks-dev (dev)
➜ sudo /usr/local/bin/fwupdtool get-devices --plugins elantp --verbose
13:45:57:0560 FuDebug              Verbose debugging enabled (on console 1)
13:45:57:0564 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
13:45:57:0564 FuConfig             loading config values from /etc/fwupd/daemon.conf
13:45:57:0564 FuConfig             using autodetected max archive size 18.4 EB
13:45:57:0566 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
13:45:57:0566 FuRemoteList         loading /usr/share/fwupd/metainfo
13:45:57:0566 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
13:45:57:0566 XbSilo               file: 23929d1e-3a1f-2197-4412-cecd590512e8, current:23929d1e-3a1f-2197-4412-cecd590512e8, cached: (null)
13:45:57:0566 XbSilo               loading silo with file contents
13:45:57:0566 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
13:45:57:0566 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
13:45:57:0566 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
13:45:57:0567 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
13:45:57:0567 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
13:45:57:0567 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
13:45:57:0567 FuRemoteList         ignoring unfound remote fwupd
13:45:57:0567 FuRemoteList         ordering lvfs-testing=lvfs+1
13:45:57:0567 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
13:45:57:0567 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
13:45:57:0567 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
13:45:57:0567 Jcat                 loaded 1 certificates
13:45:57:0567 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
13:45:57:0567 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
13:45:57:0567 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
13:45:57:0567 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
13:45:57:0567 Jcat                 loaded 1 certificates
13:45:57:0567 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
13:45:57:0567 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
13:45:57:0567 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
13:45:57:0567 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
13:45:57:0573 FuEngine             client certificate exists and working
13:45:57:0573 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
13:45:57:0574 FuHistory            got schema version of 6
13:45:57:0574 FuHwids              smbios property Manufacturer=Timi
13:45:57:0574 FuHwids              smbios property EnclosureKind=a
13:45:57:0574 FuHwids              smbios property Family=Timibook
13:45:57:0574 FuHwids              smbios property ProductName=TM1701
13:45:57:0574 FuHwids              smbios property ProductSku=TM1701-16771
[... 7 lines cut ...]
13:45:57:0579 XbSilo               file: c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, current:c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, cached: (null)
13:45:57:0579 XbSilo               loading silo with file contents
13:45:57:0579 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
13:45:57:0579 XbSilo               file: 5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, current:5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, cached: (null)
13:45:57:0579 XbSilo               loading silo with file contents
13:45:57:0580 FuEngine             528 components now in silo
13:45:57:0586 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_elantp.so)
13:45:57:0586 FuPlugin             added udev subsystem watch of i2c-dev
13:45:57:0586 FuPlugin             added udev subsystem watch of hidraw
13:45:57:0586 FuEngine             elantp has incorrect built version 6611b18844e0b419301f484e8857667f94bb92d6321a0fcb53d8cb1cb2cd0172
13:45:57:0586 FuEngine             plugins disabled: iommu, linux_tainted, synaptics_rmi, synaptics_mst, csr, cpu, system76_launch, wacom_raw, pci_bcr, upower, colorhug, ep963x, dell_dock, tpm_eventlog, jabra, ccgx, dell_ [... cut ...]
13:45:57:0605 FuDevice             failed to set quirk key GType=FuLogitechHidPpPeripheral: device GType FuLogitechHidPpPeripheral not supported
13:45:57:0606 FuPlugin             fu_plugin_device_created(elantp)
13:45:57:0607 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
WARNING: This tool has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: This package has not been validated, it may not work properly.
No detected devices
13:45:57:0617 FuMain               No detected devices
```

superm1, 2021-02-23:
> Where did you get the firmware from? All the Elan firmware is currently embargoed on the LVFS.

I still don't understand - what firmware was flashed.  Nothing on https://fwupd.org/lvfs/devices/ has Elan listed.  Did you enable a different remote?

>That makes it even more confusing, as Xiaomi isn't even on the LVFS.

FWIW I once saw a Xiaomi laptop that tried to identify as a Dell (loaded some Dell kernel modules).  My impression is they're not the highest quality firmware compared to larger OEMs.

>No, neither the fingerprint nor the touchpad. And the touchpad is also not working in UEFI BIOS where it was supposed to work.

My _suspicion_ at the moment is the wrong firmware got flashed to your device, but I don't understand what it was before and what it is now.

hughsie, 2021-02-23:
Lets go back a bit:

    fwupdmgr refresh
    fwupdmgr get-devices: it listed the ELan device also
    fwupdmgr get-updates
    fwupdmgr update
    fwupdmgr update ELAN_ID
    fwupdmgr verify-update ELAN_ID

Which of these commands did anything? The `fwupdmgr update` and `fwupdmgr update ELAN_ID` commands would only work if there was a matching release on the LVFS. I'm tempted to guess that `verify-update` put the device into bootloader mode, and the only way I know how to get an elaptp device out of bootloader is to actually flash a new firmware. :/

filariow, 2021-02-23:
> > Where did you get the firmware from? All the Elan firmware is currently embargoed on the LVFS.
> 
> I still don't understand - what firmware was flashed. Nothing on https://fwupd.org/lvfs/devices/ has Elan listed. Did you enable a different remote?

No I didn't add nor enabled a different remote.

superm1, 2021-02-23:
>I'm tempted to guess that verify-update put the device into bootloader mode

Ah yeah.... that sounds plausible.

>only way I know how to get an elaptp device out of bootloader is to actually flash a new firmware. :/

Maybe @jinglewu can advise if there is anything about this specific model that can be done to help?  Is the touchpad when in recovery mode not on I2C address 0x15 perhaps?

hughsie, 2021-02-23:
The thing is, just detaching should keep the device in HID mode, but with the bootloader flag set as it's in EAP. The i2c recovery is only needed if the firmware flash failed, which I don't think happened.

hughsie, 2021-02-23:
@filariow if I were you I'd remove the AC adaptor, battery (and also the internal extra one if there's one installed too) then wait 30 minutes and try again. @jinglewu might know a proper fix, but I've even tried the https://github.com/jinglewu/etphidiap/blob/master/etphid_updater.c code and that even needs a firmware write to allow IAP reset and then switch to TP PTP.

filariow, 2021-02-23:
Thank you @hughsie, I'll try tonight and let you know!

filariow, 2021-02-23:
Unfortunately no luck 😢
I removed the battery and the AC, the bios was resetted (I had to set timedate in BIOS in order to make the arch bootable again), but the touchpad is still not responding neither in BIOS nor obviously in Arch.

Other ideas? 😟

hughsie, 2021-02-24:
> Other ideas

I have two ideas:

 * Wait for @jinglewu to return from holiday
 * Ask the laptop vendor for the elantp firmware -- flashing the firmware will return the device to runtime mode -- you might be able to find the touchpad firmware in the win32 updater tools somewhere.

Richard.

filariow, 2021-02-24:
Ok, I found the .cat file in the [driver zip](https://cdn.cnbj1.fds.api.mi-img.com/mibook-drivers/20190110/EFD1.23.5.1101_USB_Driver_WHQL.zip) I downloaded from the [official xiaomi site](https://www.mi.com/service/bijiben/drivers/15).

At this point I should flash it into the Fingerprint right? How can I be sure of the command to execute in order to flash it. My biggest to concern is to be sure on which is the right ID in order to not flash the wrong device...

hughsie, 2021-02-24:
Well, I'm confused why it's showing up as a USB device, but it's now showing up as a hidraw device, e.g. `HIDRAW\VEN_04F3` -- I'd much rather wait for @jinglewu before we make things `worse`.

Of course, being able to permanently disable a touchpad by sending 5 bytes to a USB device _as a user_ (rather than as root) seems like a pretty big failure on the shoulders of Xiaomi...

I'm starting to wonder if that catch-all vendor match of `[DeviceInstanceId=HIDRAW\VEN_04F3]` was such a good idea afterall. I think if we keep that the minimum we need is some way of getting the hardware out of bootloader mode without actually flashing a firmware file...

superm1, 2021-02-24:
> Well, I'm confused why it's showing up as a USB device, but it's now showing up as a hidraw device, e.g. HIDRAW\VEN_04F3 -- I'd much rather wait for @jinglewu before we make things worse.

I'd guess this:
https://github.com/fwupd/fwupd/blob/master/plugins/elantp/fu-elantp-hid-device.c#L61

@filariow if you set `FWUPD_PROBE_VERBOSE=1` in your environment that you launch your fwupdtool command can you see if you see a message `not i2c-hid touchpad`?

> I'm starting to wonder if that catch-all vendor match of [DeviceInstanceId=HIDRAW\VEN_04F3] was such a good idea afterall. I think if we keep that the minimum we need is some way of getting the hardware out of bootloader mode without actually flashing a firmware file...

The other possibility is a better interrogation during probe.

superm1, 2021-02-24:
Now a few ifs.
* If that's the the reason it's not enumerating
* If it's actually the same API to reattach from bootloader (https://github.com/fwupd/fwupd/blob/master/plugins/elantp/fu-elantp-hid-device.c#L495)
* It's the same register to tell you're in bootloader (https://github.com/fwupd/fwupd/blob/master/plugins/elantp/fu-elantp-hid-device.c#L139)
* If nothing else dangerous happens running the rest of the probe routine on this device

Then you might be able to get it to reattach by removing that model number check and running `fwupdtool attach` on the device.

filariow, 2021-02-24:
> @filariow if you set `FWUPD_PROBE_VERBOSE=1` in your environment that you launch your fwupdtool command can you see if you see a message `not i2c-hid touchpad`?

@superm1 yes, I do. I attach the output of `sudo FWUPD_PROBE_VERBOSE=1 /usr/local/bin/fwupdtool get-devices --plugins elantp --verbose &> fwupdtool-verbose.log` to this comment.

[fwupdtool-verbose.log](https://github.com/fwupd/fwupd/files/6038366/fwupdtool-verbose.log)

superm1, 2021-02-24:
>@superm1 yes, I do.

OK.  So now my question then is how did this possibly probe before?  Did the PID change for the device when it detached?  Can you look through your old kernel logs to see if you can find something that mentions `04f3` from back when it "worked"?
Like from my side I see stuff like this for a USB dock:
```
[57293.256585] usb 1-1.5: New USB device found, idVendor=413c, idProduct=b06e, bcdDevice= 1.01
```

I'm wondering if that `idProduct` field changed for you.

Given the state of your device right now, I'm personally of the opinion how much more harm can running the probing routine and attempting the detach we have right now possibly do?  @hughsie can speak up if he thinks that's still too dangerous and we should wait for @jinglewu .

hughsie, 2021-02-24:
> hughsie can speak up if he thinks that's still too dangerous

I don't think the hardware can get any *more* broken than it is already...

superm1, 2021-02-24:
> I don't think the hardware can get any more broken than it is already...

OK so remove likes 61 through 68 here: https://github.com/fwupd/fwupd/blob/master/plugins/elantp/fu-elantp-hid-device.c#L61
And then try `fwupdtool attach`

hughsie, 2021-02-24:
> And then try fwupdtool attach

If that works it's a good sign, but elantp is weird that at least for my hardware you actually need to write some firmware to get out of IAP mode.

superm1, 2021-02-24:
> If that works it's a good sign, but elantp is weird that at least for my hardware you actually need to write some firmware to get out of IAP mode.

Do you know is there signature verification?  If so, `dd if=/dev/random of=foo.bin bs=128` is _some firmware_ and will fail verification and maybe get out then?

hughsie, 2021-02-24:
> maybe get out then

Good question. I think that would fail the CRC block check, although I'm not sure if that's enough of a firmware update to allow attaching back into runtime mode. If it is, we could certainly add that as a workaround to make the standalone attach work.

filariow, 2021-02-24:
> OK. So now my question then is how did this possibly probe before? Did the PID change for the device when it detached? Can you look through your old kernel logs to see if you can find something that mentions `04f3` from back when it "worked"?

Long story short: I have a Manjaro installation on a disk, on a new disk I bought I installed Arch (where I used fwupdmgr). After that I installed W10 to update the BIOS and made the Arch installation unbootable. So I reinstalled Arch and lost the history of fwupdmgr. Now I `chroot`ed on Manjaro and read the `journalctl` from 01 Feb 2021.

```
[root@fra-minotebook /]# journalctl --since="2021-02-01 00:43:54" | grep 04f3
Feb 01 09:18:13 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 01 11:33:01 fra-mipro kernel: vboxdrv: 000000004f340b25 VMMR0.r0
Feb 01 15:22:35 fra-mipro kernel: vboxdrv: 000000004f340b25 VMMR0.r0
Feb 01 23:37:56 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 02 09:12:26 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 03 09:09:29 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 03 21:46:43 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 04 09:35:05 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 05 09:21:29 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 06 14:19:59 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 07 00:56:49 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 07 17:36:53 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 08 09:02:36 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 09 09:32:34 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 09 13:37:38 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 10 09:31:02 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 11 09:23:55 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 11 18:04:05 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 11 18:19:09 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 12 08:18:58 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 13 11:05:20 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 14 11:38:23 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 14 22:13:35 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 15 09:23:40 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 16 09:13:51 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 16 22:50:43 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 16 23:15:56 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 17 09:03:37 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 09:27:53 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 18:16:05 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 18:54:26 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 19:01:39 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 22:05:03 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 22:05:40 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 23:06:53 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 23:15:49 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 23:16:34 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 23:17:08 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 18 23:17:39 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 19 00:43:55 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 19 00:56:25 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 19 10:27:36 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 19 11:39:19 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
[... cut ...]
Feb 19 14:42:29 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 19 15:01:38 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 21 11:34:00 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 21 16:54:12 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
Feb 21 17:15:21 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42
```

filariow, 2021-02-24:
Sorry @hughsie  and @superm1 I did not understood if I have to 
> remove likes 61 through 68 here: https://github.com/fwupd/fwupd/blob/master/plugins/elantp/fu-elantp-hid-device.c#L61
And then try fwupdtool attach
or do something else or nothing 😅

superm1, 2021-02-24:
>Feb 05 09:21:29 fra-mipro kernel: usb 1-8: New USB device found, idVendor=04f3, idProduct=0c1a, bcdDevice= 1.42

Well that's not helpful to illustrate how you got into this conundrum.  The device doesn't actually probe, currently and it's the same PID as before?  Hmm.

>or do something else or nothing 😅

Compile with this:
```
diff --git a/plugins/elantp/fu-elantp-hid-device.c b/plugins/elantp/fu-elantp-hid-device.c
index d1710248..be850fc7 100644
--- a/plugins/elantp/fu-elantp-hid-device.c
+++ b/plugins/elantp/fu-elantp-hid-device.c
@@ -57,16 +57,6 @@ fu_elantp_hid_device_probe (FuDevice *device, GError **error)
                return FALSE;
        }

-       /* i2c-hid */
-       if (fu_udev_device_get_model (FU_UDEV_DEVICE (device)) < 0x3000 ||
-           fu_udev_device_get_model (FU_UDEV_DEVICE (device)) >= 0x4000) {
-               g_set_error_literal (error,
-                                    FWUPD_ERROR,
-                                    FWUPD_ERROR_NOT_SUPPORTED,
-                                    "not i2c-hid touchpad");
-               return FALSE;
-       }
-
        /* set the physical ID */
        return fu_udev_device_set_physical_id (FU_UDEV_DEVICE (device), "hid", error);
 }
```

and then after you build/install, then run

`# sudo fwupdtool attach --verbose --plugins elanp`

And cross your fingers.

filariow, 2021-02-24:
Ok. should I discard the previous changes I @hughsie asked me to apply on the codebase?

```
diff --git a/plugins/elantp/fu-plugin-elantp.c b/plugins/elantp/fu-plugin-elantp.c
index 8fcb6e8f..70ddee53 100644
--- a/plugins/elantp/fu-plugin-elantp.c
+++ b/plugins/elantp/fu-plugin-elantp.c
@@ -12,20 +12,6 @@
 #include "fu-elantp-hid-device.h"
 #include "fu-elantp-i2c-device.h"

-gboolean
-fu_plugin_device_created (FuPlugin *plugin, FuDevice *dev, GError **error)
-{
-       if (fu_device_get_specialized_gtype (dev) == FU_TYPE_ELANTP_I2C_DEVICE &&
-           !fu_plugin_has_custom_flag (plugin, "elantp-recovery")) {
-               g_set_error_literal (error,
-                                    FWUPD_ERROR,
-                                    FWUPD_ERROR_NOT_SUPPORTED,
-                                    "not required");
-               return FALSE;
-       }
-       return TRUE;
-}
-
 void
 fu_plugin_init (FuPlugin *plugin)
 {
```

superm1, 2021-02-24:
Yeah I think discard them now.

filariow, 2021-02-24:
I did it and following the output

```
➜ sudo /usr/local/bin/fwupdtool attach --verbose --plugins elanp
20:58:16:0106 FuDebug              Verbose debugging enabled (on console 1)
20:58:16:0110 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
20:58:16:0111 FuConfig             loading config values from /etc/fwupd/daemon.conf
20:58:16:0111 FuConfig             using autodetected max archive size 18.4 EB
20:58:16:0112 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
20:58:16:0112 FuRemoteList         loading /usr/share/fwupd/metainfo
20:58:16:0112 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
20:58:16:0112 XbSilo               file: 23929d1e-3a1f-2197-4412-cecd590512e8, current:23929d1e-3a1f-2197-4412-cecd590512e8, cached: (null)
20:58:16:0112 XbSilo               loading silo with file contents
20:58:16:0112 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
20:58:16:0112 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
20:58:16:0112 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
20:58:16:0112 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
20:58:16:0113 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
20:58:16:0113 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
20:58:16:0113 FuRemoteList         ignoring unfound remote fwupd
20:58:16:0113 FuRemoteList         ordering lvfs-testing=lvfs+1
20:58:16:0113 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
20:58:16:0113 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
20:58:16:0113 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
20:58:16:0113 Jcat                 loaded 1 certificates
20:58:16:0113 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
20:58:16:0113 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:58:16:0113 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
20:58:16:0113 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
20:58:16:0113 Jcat                 loaded 1 certificates
20:58:16:0113 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:58:16:0113 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
20:58:16:0113 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
20:58:16:0113 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
20:58:16:0120 FuEngine             client certificate exists and working
20:58:16:0120 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
20:58:16:0121 FuHistory            got schema version of 6
20:58:16:0121 FuHwids              smbios property Manufacturer=Timi
20:58:16:0121 FuHwids              smbios property EnclosureKind=a
20:58:16:0121 FuHwids              smbios property Family=Timibook
20:58:16:0121 FuHwids              smbios property ProductName=TM1701
20:58:16:0121 FuHwids              smbios property ProductSku=TM1701-16771
20:58:16:0121 FuHwids              smbios property BiosVendor=INSYDE Corp.
20:58:16:0121 FuHwids              smbios property BiosVersion=XMAKB5R0P0E07
20:58:16:0121 FuHwids              smbios property BiosMajorRelease=6a
20:58:16:0121 FuHwids              smbios property BiosMinorRelease=75
20:58:16:0121 FuHwids              smbios property BaseboardManufacturer=Timi
20:58:16:0121 FuHwids              smbios property BaseboardProduct=TM1701
20:58:16:0127 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
20:58:16:0127 XbSilo               file: c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, current:c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, cached: (null)
20:58:16:0127 XbSilo               loading silo with file contents
20:58:16:0127 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
20:58:16:0127 XbSilo               file: 5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, current:5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, cached: (null)
20:58:16:0127 XbSilo               loading silo with file contents
20:58:16:0128 FuEngine             528 components now in silo
20:58:16:0132 FuEngine             plugins disabled: iommu, linux_tainted, synaptics_rmi, synaptics_mst, csr, cpu, system76_launch, wacom_raw, pci_bcr, upower, colorhug, ep963x, dell_dock, tpm_eventlog, jabra, ccgx, dell_esrt, hailuck, linux_sleep, dell, thunderbolt, ebitdo, goodixmoc, fresco_pd, uefi_recovery, vli, elantp, bios, bcm57xx, synaptics_prometheus, fastboot, thelio_io, uefi_capsule, uefi_pk, emmc, uefi_dbx, steelseries, amt, wacom_usb, superio, nvme, logind, acpi_facp, pci_mei, dfu, flashrom, nitrokey, acpi_dmar, linux_lockdown, optionrom, rts54hub, logitech_hidpp, pixart_rf, redfish, ata, tpm, altos, cros_ec, rts54hid, solokey, linux_swap, msr, synaptics_cxaudio
20:58:16:0132 FuEngine             Emitting PropertyChanged('Status'='loading')
Loading…                 [-                                      ]20:58:16:0141 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
No detected devices
20:58:16:0152 FuMain               No detected devices

```

I noticed `20:58:16:0110 FuMain               Failed to stop daemon: failed to find fwupd.service: ` should I run `sudo systemctl start fwupd` and retry?

superm1, 2021-02-24:
Can you run that with `FWUPD_PROBE_VERBOSE=1` again?

superm1, 2021-02-24:
> I noticed 20:58:16:0110 FuMain Failed to stop daemon: failed to find fwupd.service: should I run sudo systemctl start fwupd and retry?

No that's OK.  It's just to make sure it doesn't conflict.  Expected output

filariow, 2021-02-24:
```
➜ sudo FWUPD_PROBE_VERBOSE=1 /usr/local/bin/fwupdtool attach --verbose --plugins elanp
21:03:50:0528 FuDebug              Verbose debugging enabled (on console 1)
21:03:50:0533 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
21:03:50:0533 FuConfig             loading config values from /etc/fwupd/daemon.conf
21:03:50:0533 FuConfig             using autodetected max archive size 18.4 EB
21:03:50:0535 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
21:03:50:0535 FuRemoteList         loading /usr/share/fwupd/metainfo
21:03:50:0535 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
21:03:50:0535 XbSilo               file: 23929d1e-3a1f-2197-4412-cecd590512e8, current:23929d1e-3a1f-2197-4412-cecd590512e8, cached: (null)
21:03:50:0535 XbSilo               loading silo with file contents
21:03:50:0535 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
21:03:50:0535 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
21:03:50:0536 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
21:03:50:0536 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
21:03:50:0536 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
21:03:50:0536 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
21:03:50:0536 FuRemoteList         ignoring unfound remote fwupd
21:03:50:0536 FuRemoteList         ordering lvfs-testing=lvfs+1
21:03:50:0536 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
21:03:50:0536 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
21:03:50:0536 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
21:03:50:0536 Jcat                 loaded 1 certificates
21:03:50:0536 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
21:03:50:0536 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:03:50:0536 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
21:03:50:0536 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
21:03:50:0537 Jcat                 loaded 1 certificates
21:03:50:0537 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:03:50:0537 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
21:03:50:0537 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
21:03:50:0537 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
21:03:50:0543 FuEngine             client certificate exists and working
21:03:50:0543 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
21:03:50:0544 FuHistory            got schema version of 6
21:03:50:0544 FuHwids              smbios property Manufacturer=Timi
21:03:50:0544 FuHwids              smbios property EnclosureKind=a
21:03:50:0544 FuHwids              smbios property Family=Timibook
21:03:50:0544 FuHwids              smbios property ProductName=TM1701
21:03:50:0544 FuHwids              smbios property ProductSku=TM1701-16771
21:03:50:0544 FuHwids              smbios property BiosVendor=INSYDE Corp.
21:03:50:0544 FuHwids              smbios property BiosVersion=XMAKB5R0P0E07
21:03:50:0544 FuHwids              smbios property BiosMajorRelease=6a
21:03:50:0544 FuHwids              smbios property BiosMinorRelease=75
21:03:50:0544 FuHwids              smbios property BaseboardManufacturer=Timi
21:03:50:0544 FuHwids              smbios property BaseboardProduct=TM1701
21:03:50:0549 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
21:03:50:0549 XbSilo               file: c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, current:c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, cached: (null)
21:03:50:0549 XbSilo               loading silo with file contents
21:03:50:0550 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
21:03:50:0550 XbSilo               file: 5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, current:5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, cached: (null)
21:03:50:0550 XbSilo               loading silo with file contents
21:03:50:0550 FuEngine             528 components now in silo
21:03:50:0553 FuEngine             plugins disabled: iommu, linux_tainted, synaptics_rmi, synaptics_mst, csr, cpu, system76_launch, wacom_raw, pci_bcr, upower, colorhug, ep963x, dell_dock, tpm_eventlog, jabra, ccgx, dell_esrt, hailuck, linux_sleep, dell, thunderbolt, ebitdo, goodixmoc, fresco_pd, uefi_recovery, vli, elantp, bios, bcm57xx, synaptics_prometheus, fastboot, thelio_io, uefi_capsule, uefi_pk, emmc, uefi_dbx, steelseries, amt, wacom_usb, superio, nvme, logind, acpi_facp, pci_mei, dfu, flashrom, nitrokey, acpi_dmar, linux_lockdown, optionrom, rts54hub, logitech_hidpp, pixart_rf, redfish, ata, tpm, altos, cros_ec, rts54hid, solokey, linux_swap, msr, synaptics_cxaudio
21:03:50:0553 FuEngine             Emitting PropertyChanged('Status'='loading')
Loading…                 [-                                      ]21:03:50:0553 FuEngine             usb added FuUsbDevice:
Unknown Device
  Flags:                none
  PhysicalId:           usb:02:00

[... cut ...]
  InstanceIds:          USB\VID_1D6B&PID_0002&REV_0510
  InstanceIds:          USB\VID_1D6B&PID_0002
  Flags:                none
  VendorId:             USB:0x1D6B
  Version:              5.10
  VersionFormat:        bcd
  PhysicalId:           usb:01:00

21:03:50:0563 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
No detected devices
21:03:50:0573 FuMain               No detected devices
```

superm1, 2021-02-24:
Oh there's a typo.

>➜ sudo FWUPD_PROBE_VERBOSE=1 /usr/local/bin/fwupdtool attach --verbose --plugins elanp

Should be `elantp`

filariow, 2021-02-24:
you're right! here the new output

```
➜ sudo FWUPD_PROBE_VERBOSE=1 /usr/local/bin/fwupdtool attach --verbose --plugins elantp
21:06:59:0446 FuDebug              Verbose debugging enabled (on console 1)
21:06:59:0452 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
21:06:59:0452 FuConfig             loading config values from /etc/fwupd/daemon.conf
21:06:59:0452 FuConfig             using autodetected max archive size 18.4 EB
21:06:59:0453 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
21:06:59:0453 FuRemoteList         loading /usr/share/fwupd/metainfo
21:06:59:0453 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
21:06:59:0453 XbSilo               file: 23929d1e-3a1f-2197-4412-cecd590512e8, current:23929d1e-3a1f-2197-4412-cecd590512e8, cached: (null)
21:06:59:0453 XbSilo               loading silo with file contents
21:06:59:0454 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
21:06:59:0454 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
21:06:59:0454 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
21:06:59:0454 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
21:06:59:0454 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
21:06:59:0454 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
21:06:59:0454 FuRemoteList         ignoring unfound remote fwupd
21:06:59:0454 FuRemoteList         ordering lvfs-testing=lvfs+1
21:06:59:0454 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
21:06:59:0454 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
21:06:59:0454 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
21:06:59:0454 Jcat                 loaded 1 certificates
21:06:59:0455 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
21:06:59:0455 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:06:59:0455 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
21:06:59:0455 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
21:06:59:0455 Jcat                 loaded 1 certificates
21:06:59:0455 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:06:59:0455 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
21:06:59:0455 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
21:06:59:0455 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
21:06:59:0463 FuEngine             client certificate exists and working
21:06:59:0463 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
21:06:59:0463 FuHistory            got schema version of 6
21:06:59:0464 FuHwids              smbios property Manufacturer=Timi
21:06:59:0464 FuHwids              smbios property EnclosureKind=a
21:06:59:0464 FuHwids              smbios property Family=Timibook
21:06:59:0464 FuHwids              smbios property ProductName=TM1701
21:06:59:0464 FuHwids              smbios property ProductSku=TM1701-16771
21:06:59:0464 FuHwids              smbios property BiosVendor=INSYDE Corp.
21:06:59:0464 FuHwids              smbios property BiosVersion=XMAKB5R0P0E07
21:06:59:0464 FuHwids              smbios property BiosMajorRelease=6a
21:06:59:0464 FuHwids              smbios property BiosMinorRelease=75
21:06:59:0464 FuHwids              smbios property BaseboardManufacturer=Timi
21:06:59:0464 FuHwids              smbios property BaseboardProduct=TM1701
21:06:59:0469 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
21:06:59:0469 XbSilo               file: c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, current:c29e50bc-5de9-ae5c-5b16-8aa8a3ce29ee, cached: (null)
21:06:59:0469 XbSilo               loading silo with file contents
21:06:59:0470 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
21:06:59:0470 XbSilo               file: 5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, current:5a13ecf4-e4cb-b2c4-4f86-6d41a3cfc6de, cached: (null)
21:06:59:0470 XbSilo               loading silo with file contents
21:06:59:0470 FuEngine             528 components now in silo
21:06:59:0474 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_elantp.so)
21:06:59:0475 FuPlugin             added udev subsystem watch of i2c-dev
21:06:59:0475 FuPlugin             added udev subsystem watch of hidraw
21:06:59:0475 FuEngine             elantp has incorrect built version 6611b18844e0b419301f484e8857667f94bb92d6321a0fcb53d8cb1cb2cd0172
21:06:59:0475 FuEngine             plugins disabled: iommu, linux_tainted, synaptics_rmi, synaptics_mst, csr, cpu, system76_launch, wacom_raw, pci_bcr, upower, colorhug, ep963x, dell_dock, tpm_eventlog, jabra, ccgx, dell_esrt, hailuck, linux_sleep, dell, thunderbolt, ebitdo, goodixmoc, fresco_pd, uefi_recovery, vli, bios, bcm57xx, synaptics_prometheus, fastboot, thelio_io, uefi_capsule, uefi_pk, emmc, uefi_dbx, steelseries, amt, wacom_usb, superio, nvme, logind, acpi_facp, pci_mei, dfu, flashrom, nitrokey, acpi_dmar, linux_lockdown, optionrom, rts54hub, logitech_hidpp, pixart_rf, r [... cut ...]
21:06:59:0487 FuPlugin             fu_plugin_device_created(elantp)
21:06:59:0487 FuEngine             elantp ignoring: not i2c-hid touchpad
21:06:59:0487 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
WARNING: This tool has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: This package has not been validated, it may not work properly.
No detected devices
21:06:59:0490 FuMain               No detected devices
```

superm1, 2021-02-24:
It's still showing:

> 21:06:59:0487 FuEngine             elantp ignoring: not i2c-hid touchpad

Did you make sure you did `ninja install`  after you built?

filariow, 2021-02-24:
Installed it correctly and executed. Here the log.

```
➜ sudo FWUPD_PROBE_VERBOSE=1 /usr/bin/fwupdtool attach --verbose --plugins elantp
21:23:52:0271 FuDebug              Verbose debugging enabled (on console 1)
21:23:52:0275 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
21:23:52:0275 FuConfig             loading config values from /etc/fwupd/daemon.conf
21:23:52:0275 FuConfig             using autodetected max archive size 18.4 EB
21:23:52:0276 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
21:23:52:0276 FuRemoteList         loading /usr/share/fwupd/metainfo
21:23:52:0276 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
21:23:52:0276 XbSilo               file: 23929d1e-3a1f-2197-4412-cecd590512e8, current:9e8d3dbf-1408-d8ec-9d05-eff6b3589ed8, cached: (null)
21:23:52:0276 XbSilo               compiling /usr/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs.metainfo.xml:ctime=1614201795.600590…
21:23:52:0277 XbSilo               compiling /usr/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs-testing.metainfo.xml:ctime=1614201795.600590…
21:23:52:0287 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
21:23:52:0288 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
21:23:52:0288 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
21:23:52:0288 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
21:23:52:0288 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
21:23:52:0288 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
21:23:52:0288 FuRemoteList         ignoring unfound remote fwupd
21:23:52:0288 FuRemoteList         ordering lvfs-testing=lvfs+1
21:23:52:0288 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
21:23:52:0288 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
21:23:52:0288 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
21:23:52:0288 Jcat                 loaded 1 certificates
21:23:52:0288 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
21:23:52:0288 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:23:52:0288 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
21:23:52:0288 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
21:23:52:0289 Jcat                 loaded 1 certificates
21:23:52:0289 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:23:52:0289 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
21:23:52:0289 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
21:23:52:0289 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
21:23:52:0295 FuEngine             client certificate exists and working
21:23:52:0295 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
21:23:52:0295 FuHistory            got schema version of 6
21:23:52:0295 FuHwids              smbios property Manufacturer=Timi
21:23:52:0295 FuHwids              smbios property EnclosureKind=a
21:23:52:0295 FuHwids              smbios property Family=Timibook
21:23:52:0295 FuHwids              smbios property ProductName=TM1701
21:23:52:0295 FuHwids              smbios property ProductSku=TM1701-16771
21:23:52:0295 FuHwids              smbios property BiosVendor=INSYDE Corp.
21:23:52:0295 FuHwids              smbios property BiosVersion=XMAKB5R0P0E07
21:23:52:0295 FuHwids              smbios property BiosMajorRelease=6a
21:23:52:0295 FuHwids              smbios property BiosMinorRelease=75
21:23:52:0295 FuHwids              smbios property BaseboardManufacturer=Timi
21:23:52:0295 FuHwids              smbios property BaseboardProduct=TM1701
21:23:52:0296 FuQuirks             skipping invalid file elantp.quirk.bkp
21:23:52:0301 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
21:23:52:0301 XbSilo               file: 00000000-0000-0000-0000-000000000000, current:b8196285-5b8f-06fd-004d-adcf9e3fa15e, cached: (null)
21:23:52:0301 XbSilo               compiling /usr/share/fwupd/quirks.d/altos.quirk:ctime=1614201795.687262…
21:23:52:0301 XbSilo               compiling /usr/share/fwupd/quirks.d/ata.quirk:ctime=1614201795.687262…
21:23:52:0301 XbSilo               compiling /usr/share/fwupd/quirks.d/bcm57xx.quirk:ctime=1614201795.687262…
21:23:52:0302 XbSilo               compiling /usr/share/fwupd/quirks.d/ccgx-ids.quirk:ctime=1614201795.687262…
21:23:52:0303 XbSilo               compiling /usr/share/fwupd/quirks.d/ccgx.quirk:ctime=1614201795.687262…
21:23:52:0303 XbSilo               compiling /usr/share/fwupd/quirks.d/colorhug.quirk:ctime=1614201795.687262…
21:23:52:0304 XbSilo               compiling /usr/share/fwupd/quirks.d/cpu.quirk:ctime=1614201795.687262…
21:23:52:0304 XbSilo               compiling /usr/share/fwupd/quirks.d/cros-ec.quirk:ctime=1614201795.687262…
21:23:52:0304 XbSilo               compiling /usr/share/fwupd/quirks.d/dell-dock.quirk:ctime=1614201795.687262…
[... cut ...]
21:23:52:0503 XbSilo               ignoring for OR statement: indexed string '2cb5d14c-127f-51a7-9bba-72f4e39992b5' was unfound
21:23:52:0503 XbSilo               ignoring for OR statement: indexed string '3852e430-731f-55fa-a0e1-f2ff3b818c9f' was unfound
WARNING: This package has not been validated, it may not work properly.
Selected device: ETD2303:00 04F3:0400
21:23:52:0506 FuEngine             Emitting PropertyChanged('Status'='device-restart')
Restarting device…       [ -                                     ]% 
```

superm1, 2021-02-24:
and?🥁 🥁 🥁 🥁 🥁 🥁 🥁 🥁 🥁 🥁 🥁 🥁

filariow, 2021-02-24:
and it's still not working 😢 do I have to reboot?

superm1, 2021-02-24:
No, rebooting shouldn't be required  I guess that means this idea didn't work. 😭

hughsie, 2021-02-24:
I really think you need to use install-blob, not attach. Either the windows fw or just a file of the right size.

superm1, 2021-02-24:
I don't see the Windows FW in that zip.  So your only real option then is this (which should fail the CRC check as @hughsie said)

```
# dd if=/dev/random of=foo.bin bs=57344 count=1
# sudo fwupdtool install-blob foo.bin --plugins elantp --verbose
```

filariow, 2021-02-24:
```
fra-minotebook# sudo fwupdtool install-blob foo.bin --plugins elantp --verbose
21:38:53:0300 FuDebug              Verbose debugging enabled (on console 1)
21:38:53:0301 FuCommon             reading foo.bin with 57344 bytes
21:38:53:0304 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
21:38:53:0305 FuConfig             loading config values from /etc/fwupd/daemon.conf
21:38:53:0305 FuConfig             using autodetected max archive size 18.4 EB
21:38:53:0305 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
21:38:53:0306 FuRemoteList         loading /usr/share/fwupd/metainfo
21:38:53:0306 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
21:38:53:0306 XbSilo               file: 9e8d3dbf-1408-d8ec-9d05-eff6b3589ed8, current:9e8d3dbf-1408-d8ec-9d05-eff6b3589ed8, cached: (null)
21:38:53:0306 XbSilo               loading silo with file contents
21:38:53:0306 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
21:38:53:0306 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
21:38:53:0306 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
21:38:53:0306 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
21:38:53:0306 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
21:38:53:0306 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
21:38:53:0306 FuRemoteList         ignoring unfound remote fwupd
21:38:53:0306 FuRemoteList         ordering lvfs-testing=lvfs+1
21:38:53:0306 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
21:38:53:0306 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
21:38:53:0306 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
21:38:53:0307 Jcat                 loaded 1 certificates
21:38:53:0307 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
21:38:53:0307 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:38:53:0307 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
21:38:53:0307 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
21:38:53:0307 Jcat                 loaded 1 certificates
21:38:53:0307 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:38:53:0307 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
21:38:53:0307 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
21:38:53:0307 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
21:38:53:0313 FuEngine             client certificate exists and working
21:38:53:0313 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
21:38:53:0313 FuHistory            got schema version of 6
21:38:53:0313 FuHwids              smbios property Manufacturer=Timi
21:38:53:0313 FuHwids              smbios property EnclosureKind=a
21:38:53:0313 FuHwids              smbios property Family=Timibook
21:38:53:0313 FuHwids              smbios property ProductName=TM1701
21:38:53:0313 FuHwids              smbios property ProductSku=TM1701-16771
21:38:53:0313 FuHwids              smbios property BiosVendor=INSYDE Corp.
21:38:53:0313 FuHwids              smbios property BiosVersion=XMAKB5R0P0E07
21:38:53:0313 FuHwids              smbios property BiosMajorRelease=6a
21:38:53:0313 FuHwids              smbios property BiosMinorRelease=75
21:38:53:0313 FuHwids              smbios property BaseboardManufacturer=Timi
21:38:53:0313 FuHwids              smbios property BaseboardProduct=TM1701
21:38:53:0313 FuQuirks             skipping invalid file elantp.quirk.bkp
21:38:53:0318 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
21:38:53:0318 XbSilo               file: b8196285-5b8f-06fd-004d-adcf9e3fa15e, current:b8196285-5b8f-06fd-004d-adcf9e3fa15e, cached: (null)
21:38:53:0318 XbSilo               loading silo with file contents
21:38:53:0319 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
21:38:53:0319 XbSilo               file: 7476d6a3-d014-9c77-ce05-ba0a5401ca1f, current:7476d6a3-d014-9c77-ce05-ba0a5401ca1f, cached: (null)
21:38:53:0319 XbSilo               loading silo with file contents
21:38:53:0319 FuEngine             530 components now in silo
21:38:53:0323 FuPlugin             init(/usr/lib/fwupd-plugins-3/libfu_plugin_elantp.so)
21:38:53:0323 FuPlugin             added udev subsystem watch of i2c-dev
21:38:53:0323 FuPlugin             added udev subsystem watch of hidraw
21:38:53:0323 FuEngine             plugins disabled: iommu, linux_tainted, synaptics_rmi, synaptics_mst, cpu, system76_launch, wacom_raw, pci_bcr, upower, colorhug, ep963x, dell_dock, tpm_eventlog, jabra, ccgx, dell_esrt, hailuck, linux_sleep, dell, thunderbolt, ebitdo, goodixmoc, fresco_pd, uefi_recovery, vli, bios, bcm57xx, synaptics_prometheus, dfu_csr, fastboot, thelio_io, uefi_capsule, uefi_pk, emmc, uefi_dbx, steelseries, amt, wacom_usb, superio, nvme, logind, acpi_facp, pci_mei, dfu, nitrokey, acpi_dmar, linux_lockdown, optionrom, rts54hub, logitech_hidpp, pixart_rf, redfish, ata, tpm, altos, cros_ec, rts54hid, [... cut ...]
  Icon:                 input-touchpad
  Created:              2021-02-24
  Modified:             2021-02-24
  PhysicalId:           HID_PHYS=i2c-ETD2303:00
  FirmwareSizeMin:      57344
  FirmwareSizeMax:      57344
  Order:                0
  Priority:             1
  PossiblePlugin:       elantp
  ModuleId:             0x7d
  Pattern:              0x1
  FwPageSize:           0x0
  IcPageCount:          0x380
  IapType:              0x0
  IapCtrl:              0xc1

firmware incompatible, got 0xa04a, expected 0x007d
```

[35 more replies]

closed 2021-03-12 by hughsie