fwupd/fwupd #2940: Writing to FuVliPdDevice seems not to be working

hughsie, 2021-02-25:
I'm using the Lenovo VL103 no-power USB-C dongle `USB\VID_17EF&PID_7223` for testing, and then using:

    sudo FWUPD_VLI_USBHUB_VERBOSE=1 ./src/fwupdtool --verbose --plugins vli install-blob VL103_App1-R99_54_07_01-20181015-SL-PS186-4Lan+I2C-VDM.bin

This is identified as a "dual-image" device, and so the `fu_vli_pd_device_write_dual_firmware()` gets selected. It then writes 0x8000 bytes @0x28000, verifying each write as it goes:

	12:10:48:0499 FuPluginVliUsbhub    writing 0x8000 bytes @0x28000
	12:10:48:0499 FuPluginVliUsbhub    writing 0x20 block @0x28020
	12:10:48:0501 FuPluginVliUsbhub    writing 0x20 block @0x28040
	12:10:48:0504 FuPluginVliUsbhub    writing 0x20 block @0x28060
	12:10:48:0506 FuPluginVliUsbhub    writing 0x20 block @0x28080
	....
	12:10:48:0808 FuPluginVliUsbhub    writing 0x20 block @0x28fc0
	12:10:48:0811 FuPluginVliUsbhub    writing 0x20 block @0x28fe0
	12:10:48:0813 FuPluginVliUsbhub    writing 0x20 block @0x29000

and then the write fails as the data is not read back as written, and I get the error:

    failed to write block 0x80: got 0x07, expected 0x06 @ 0x0005

The address of `0x29000` and block number of `0x80` are certainly suspicious and make me wonder if the device is actually dual image... and looking at the quirk I see:

    [DeviceInstanceId=USB\VID_17EF&PID_7223]
    Plugin = vli
    GType = FuVliPdDevice

so what's happened is that when the device booted from PD mode to BILLBOARD we switched to a different (generic) VID and PID:

    [DeviceInstanceId=USB\VID_2109&PID_0103]
    Plugin = vli
    GType = FuVliPdDevice
    Flags = dual-image
    DeviceKind = VL103

...and this is what's being used for the actual update.

So my questions are:

 * is VID_17EF&PID_7223 dual image?
 * are **all** VID_2109&PID_0103 billboards actually dual bank? In which case the first bullet must be false.
 * should we move the dual-image to the non-billboard VID/PID? If so, which non-billboard (as in the OEM IDs) are actually dual bank? VID_17EF&PID_7212 is already marked as dual bank correctly.

I suspect what we need to do is move the dual-image from the VID_2109&PID_0103 to the VID_17EF&PID_7223, and I also suspect we added it to the BB device to work around a bug I fixed a few days ago: https://github.com/fwupd/fwupd/commit/7b2621c3de970b9b7e5378a3937933e9d92a07be

Seems sane? Thanks.

hughsie, 2021-04-11:
@memily did https://github.com/fwupd/fwupd/pull/2982 fix this too? Thanks.

SherlockChu219, 2021-04-12:
Hi Richard,
VL103 can support 2 copies firmware in SPI flash.

VL103 checks FW1(0x20000-0x28000) first. If correct, it boots from FW1, else VL103 checks FW2(0x28000-0x30000) and boots from FW2 if it is correct.
If both FWs are incorrect, it boots from rom-code with ID 2109/0103.
So, the flow of update VL103 with 2 copies FW should be:
(1) check CRC of FW1
(2) if CRC is correct, update FW2 first(using sector erase), then update FW1.
(3) if CRC is incorrect, update FW1 first(using sector erase), then update FW2.
Applying the above rule will not result in an error with VID/PID 2019/0103, the VID/PID will remain the correct 17EF/7223 during the entire update process.

hughsie, 2021-04-12:
Thanks for the info @SherlockChu219. If i understand correctly then we do need to tag VID_17EF&PID_7223 as dual-bank then. I'll do a PR for @memily to review, thanks.

closed 2021-04-13 by hughsie