fwupd/fwupd #2946: fwupd suggests wrong firmware updates on LG gram laptop

pjb7687, 2021-02-26:
**Describe the bug**
On an LG laptop (which is not mine, model name 14ZD970-GX70K), the software update suggests updating the system firmware to the wrong one, "ThinkPad X1 Yoga 3rd Corporate ME Update" (version 184.80.3746). This is also shown in the snap 'gnome-firmware', which also suggests the wrong firmware is one of the releases of the current system firmware "14ZD970-GX70K System Firmware" (version 1.21.37898).

[image]
[image]

After confirming the update, the laptop restarted, went into the firmware update mode, and then stuck on it forever. After force rebooting the machine, (thankfully) it was returned back to the normal Ubuntu session.

**Steps to Reproduce**
On Ubuntu 18.04.5 LTS, the software updater notifies the update.
[image]

**Expected behavior**
No new firmware updates suggested.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:  1.2.10
compile-time dependency versions
  gusb:  0.2.11
  efivar:  34
daemon version:  1.2.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
It was installed by default. Therefore via apt, I think.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
14ZD970-GX70K System Firmware
  DeviceId:             4be1d82544d8b1332b6189d4368d1e5a03e4883c
  Guid:                 e20bafd3-9914-4f4f-9537-3129e090eb3c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              1.21.37898
  VersionLowest:        1.21.37898
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2021-02-25
  UpdateState:          success

Unifying Receiver
  DeviceId:             b0c34fcaa7f7fb80dfc0d931d18e93e88479b186
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.11_B0036
  VersionBootloader:    BOT03.02_B0009
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2021-02-25

HD Graphics 620
  DeviceId:             bbbf1ce3d1cf15550c3760b354592040292415bb
  Guid:                 2886e312-afd8-5510-8993-12d568a85e00
  Guid:                 38f6c009-a25c-5b50-b3b5-fef4f9c6b846
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Version:              02
[... 328 lines cut ...]
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.15.0-135-generic
├── initrd.img-4.15.0-136-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-4.15.0-135-generic
├── System.map-4.15.0-136-generic
├── vmlinuz-4.15.0-135-generic
└── vmlinuz-4.15.0-136-generic

5 directories, 288 files
```

**Additional questions**
- Operating system and version: Ubuntu 18.04.5 LTS
- Have you tried rebooting? Yes
- Is this a regression? This first happened on 24 Feb 2021. It didn't happen at least a week ago.
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? This isn't a Lenovo system

hughsie, 2021-02-26:
Okay, this a first. I think LG has accidentally used the same GUID as Lenovo did. It's tricky as LG are not shipping updates on the LVFS. I'll think of a few ways we can fix this but many thanks for reporting this!

hughsie, 2021-02-26:
So this can never happen again I've accelerated something that I've been planning to do for some time: https://lists.linuxfoundation.org/pipermail/lvfs-announce/2021-February/000056.html

This means that an OEM like Lenovo could never accidentally ship a firmware update for a different OEM, or in this case it means the requirement would fail on the LG machine that's accidentally using the GUID defined by Lenovo.

Can you try `fwupdmgr refresh --force` and then see if the update is still suggested? Thanks.

pjb7687, 2021-02-26:
Thanks! But unfortunately it didn't work.

hughsie, 2021-02-26:
> But unfortunately it didn't work

Doh! Okay, lets try something else. Can you `fwupdmgr refresh --force` again please, this time I've added a HWID requirement on the firmware too.

pjb7687, 2021-02-26:
It worked! Great! Thank you so much.

hughsie, 2021-02-26:
Excellent -- I still need to find the right people in 3 different companies to work out what's the root cause. Random GUIDs just shouldn't clash, ever!

superm1, 2021-02-26:
>Random GUIDs just shouldn't clash, ever!

My favorite quote from stack overflow on the topic: `There are so many GUIDs that if you generate several trillion of them randomly, you're still more likely to get hit by a meteorite than to have even one collision`

>root cause

Without talking to anyone my hypothetical guess is some reference code from company A made it into production code at company B and company C with ZERO changes.  And if that's the case you better pray that company A's reference code doesn't ever become production code for one of company A's products...

closed 2021-02-26 by superm1