fwupd/fwupd #2956: Dell WD19 upgrade issue

fredoche, 2021-03-01:
**Describe the bug**
I cannot update to the latest firmware version. gnome-software, the stand-alone  "firmware" program, and CLI fwupdmgr all prompt for an update. When told to proceed, they look like they are performing the update, yet, the firmware on the WD19TB stays the same (TB ver 43 instead of 60).
yet,
when running wupdmgr get-devices --filter=~internal , it says that the devices have been updated successfully, which may, or may not, get your analytics maybe off place.
``` Devices that have been updated successfully:

 • VMM5331 in Dell dock (05.04.03 → 05.04.06)
 • WD19TB (01.00.00.04 → 01.00.00.05)
 • Package level of Dell dock (01.00.15.00 → 01.00.15.01)
 • Thunderbolt controller in Dell dock (43.00 → 60.00)
```

**Steps to Reproduce**
run fwupdmgr twice in a row.
```
root@grey-matter fred]# fwupdmgr  update
Devices with no available firmware updates: 
 • Thunderbolt host controller
Upgrade available for Thunderbolt controller in Dell dock from 43.00 to 60.00
Thunderbolt controller in Dell dock must remain connected for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Téléchargement…          [***************************************]
Décompression…           [***************************************]
Authentification…        [***************************************]
Authentification…        [***************************************]
Mise à jour de Thunderbolt controller in Dell dock…**************]
Écriture…                [***************************************]
Successfully installed firmware
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
 • Micron 2200S NVMe 512GB
 • UEFI dbx
[root@grey-matter fred]# fwupdmgr  update
Devices with no available firmware updates: 
 • Thunderbolt host controller
Upgrade available for Thunderbolt controller in Dell dock from 43.00 to 60.00
Thunderbolt controller in Dell dock must remain connected for the duration of the update to avoid damage. Continue with update? [Y|n]:
```

**Expected behavior**
the second fwupmgr should be no-op , and the firmware should show as updated.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
[root@grey-matter fred]# fwupdmgr --version
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.7

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Fedora 33 default, so dnf I guess

**fwupd device information**
Please provide the output of the external fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --filter=~internal
XPS 13 7390
│
└─WD19TB:
  │   Device ID:          570867f4ddda9915445da59bd85cd0e0e507270e
[... 199 lines cut ...]
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0965 FuPlugin             destroy(synaptics_mst)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0965 FuPlugin             destroy(logind)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0965 FuPlugin             destroy(upower)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0965 FuPlugin             destroy(modem_manager)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0968 FuPlugin             destroy(linux_lockdown)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0968 FuPlugin             destroy(tpm_eventlog)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0969 FuPlugin             destroy(linux_swap)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0969 FuPlugin             destroy(dell)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0969 FuPlugin             destroy(flashrom)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0969 FuPlugin             destroy(linux_tainted)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0970 FuPlugin             destroy(redfish)
mars 01 10:42:14 grey-matter fwupd[3010]: 09:42:14:0970 FuPlugin             destroy(uefi_capsule)
mars 01 10:42:14 grey-matter systemd[1]: fwupd.service: Succeeded.
mars 01 10:42:14 grey-matter systemd[1]: Stopped Firmware update daemon.
mars 01 10:42:24 grey-matter systemd[1]: Starting Firmware update daemon...
mars 01 10:42:24 grey-matter fwupd[5358]: 09:42:24:0617 FuRemoteList         loading /usr/share/fwupd/metainfo
mars 01 10:42:24 grey-matter fwupd[5358]: 09:42:24:0617 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb

```

**Additional questions**
- Operating system and version: fedora 33 workstation
- Have you tried unplugging the dock or any peripherals from your machine? yes
- Have you tried to power cycle the dock from the AC adapter? yes
- Is this a regression? I dont know

fredoche, 2021-03-01:
For the record, after writing this report, so , for like 5 minutes, after changing the log level, while every peripheral (eth, usb, hdmi) disconnected, now, "fwupdmgr  update" doesnt prompt for the same update. I'll check now if the update has been applied correctly after reboot and a/c disconnect and will let you know

fredoche, 2021-03-01:
Aaand its updated! So some patience is required. Maybe mashing the update button kind of start the update all over again, maybe I got unlucky, maybe the announced duration (22 seconds) for update is kind of optimistic, maybe the device got scared when I reached out for help on github, I really can't tell. In any case my firmware is updated, maybe I fell into an unfortunate UX issue or just got plain unlucky. Thanks, keep up the good work

superm1, 2021-03-01:
Well glad it got done, but let's find out what happened..

When it "wasn't working", did you happen to unplug the dock in-between attempts?  And if you did - did you unplug from host or from wall AC adapter?

The flow is _supposed_ to be:
1. `fwupdmgr update`
2. You get a message telling you the update finishes next time you unplug dock from host
3. You unplug dock from host
4. Plug dock back in
4. Fully updated

superm1, 2021-03-01:
@hughsie did the update message get lost in the metadata again?  There was a new release just pushed out yesterday that @fredoche has run here, I wonder if it's missing.

hughsie, 2021-03-01:
> did the update message get lost in the metadata again

Looks okay here.

superm1, 2021-03-01:
> Looks okay here.

I fetched the stable metadata, and yeah I see it in the EC as well.
[image]

BTW I noticed this in requires, not sure how that is happening in stable, nor how it was possible for @fredoche to update with this here.
```
    <requires>
      <firmware compare="eq" version="XXX:NEVER_GOING_TO_MATCH">vendor-id</firmware>
    </requires>
```

>and I dont believe gnome-software or firmware told me either

Gnome-software doesn't do anything with it _currently_:
https://gitlab.gnome.org/GNOME/gnome-software/-/issues/806

>flow
```
Successfully installed firmware
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
 • Micron 2200S NVMe 512GB
 • UEFI dbx
[root@grey-matter fred]# fwupdmgr  update
Devices with no available firmware updates: 
 • Thunderbolt host controller
Upgrade available for Thunderbolt controller in Dell dock from 43.00 to 60.00
Thunderbolt controller in Dell dock must remain connected for the duration of the update to avoid damage. Continue with update? [Y|n]:
```
@hughsie  I wonder if we're missing removing the updatable flag or lying about the version.  I _think_ what happened is device is in needs-activation right now but version number doesn't update so it still looks like it needs an update.
Which approach is better in your mind?

superm1, 2021-03-01:
>       <firmware compare="eq" version="XXX:NEVER_GOING_TO_MATCH">vendor-id</firmware>

The number of downloads is also really low, so I worry about this.

hughsie, 2021-03-01:
What's the vendor ID supposed to be?

superm1, 2021-03-01:
Everything but the TBT controller: `USB:0x413C`.  TBT controller either of : `THUNDERBOLT:0x00D4|TBT:0x00D4`

hughsie, 2021-03-02:
> Everything but the TBT controller: USB:0x413C. TBT controller either of : THUNDERBOLT:0x00D4|TBT:0x00D4

The `dell` account has all those already. Remember it's the uploader ODM that needs the vendor-id set; I've assigned the ODM the requried USB vendor ID.

superm1, 2021-03-05:
As mentioned above this is caused by fwupd leaving the device in `updatable` state but also not bumping the version. 
This behavior is fixed by https://github.com/fwupd/fwupd/commit/3df7dd370888b49670e374f8c213f35430ad7b42 which will be in fwupd 1.5.8.  On any earlier version of fwupd, if you simply call `fwupdtool activate` or unplug the USB-C cable then it will complete the update.

closed 2021-03-05 by superm1