fwupd/fwupd #2975: Dell WD19 upgrade issue - the update message is unclear

ckruetze, 2021-03-05:
**Describe the bug**
Update message is unclear.
I have now tried about 5 times to update an WD19. I'm aware of https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-WD19*:-Device-ID-was-not-found but that is not the problem. The problem is the update message.
"Successfully installed firmware: Updates will complete after the dock is unplugged.  The dock will be unavailable for approximately 1 minute."
The message is not clear in a way that at least I and a colleague I asked tried to unplug the WD19 power cable. We didn't realise that we should unplug the USB Type C cable connecting the dock to the laptop.

I guess it is not a bug with the update process but more with the documentation. As when I realised that it was the USB cable I should unplug everything worked as expected.

**Steps to Reproduce**
Update a Dell WD19 dock and read the message

**Expected behavior**
A more clear message. If possible change the text to something like "... Updates will complete after the USB cable connecting the dock and laptop is unplugged..."

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Not sure it was installed with Ubuntu I think

**fwupd device information**
Please provide the output of the external fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --filter=~internal
Precision 5540
│
├─WD19:
│ │   Device ID:           70baeee61c76c1ff376c0b74a696fc7c7213c734
│ │   Summary:             High performance dock
│ │   Current version:     01.00.00.05
│ │   Minimum Version:     01.00.00.00
│ │   Vendor:              Dell Inc. (USB:0x413C)
│ │   Install Duration:    1 minute
│ │   Serial Number:       H2ZJSV2/3138161177336244
│ │   Update Message:      Updates will complete after the dock is unplugged.  The dock will be unavailable for approximately 1 minute.
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:         5c6c8dedd16aa060fe7996087358c98f1a04e1cb
│ │     Summary:           A representation of dock update status
│ │     Current version:   01.00.15.01
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  5 seconds
│ │     GUID:              8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:      • Updatable
│ │                        • Supported on remote server
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │   
│ ├─RTS5413 in Dell dock:
│ │     Device ID:         3b7d38c59fa333512ced0c08664219fa22c90b8d
│ │     Summary:           USB 3.1 Generation 1 Hub
│ │     Current version:   01.21
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  14 seconds
│ │     GUIDs:             86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4 ← USB\VID_413C&PID_B06F&REV_0101
│ │                        b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
│ │                        ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device is usable for the duration of the update
│ │   
│ ├─RTS5487 in Dell dock:
│ │     Device ID:         c33d3b2a5870a4c5d95ee7c41784dec1f68078e8
│ │     Summary:           USB 3.1 Generation 2 Hub
│ │     Current version:   01.47
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  3 seconds
│ │     GUIDs:             707c63d2-e597-5c40-84db-9b1bb4c48d96 ← USB\VID_413C&PID_B06E&REV_0101
│ │                        acfcd89b-105d-55b9-b85b-08bf8508f38c ← USB\VID_413C&PID_B06E
│ │                        568ffa1e-a0db-5287-9ea3-872b60f7730b ← USB\VID_413C&PID_B06E&hub
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device is usable for the duration of the update
│ │   
│ └─VMM5331 in Dell dock:
│       Device ID:         c108504056150122cb663b10de22aa18147fa21e
│       Summary:           Multi Stream Transport controller
│       Current version:   05.04.06
│       Vendor:            Dell Inc. (USB:0x413C)
│       Install Duration:  6 minutes
│       GUID:              89fec0b6-6b76-5008-b82c-5e5c6c164007 ← MST-panamera-vmm5331-259
│       Device Flags:      • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│                          • Device is usable for the duration of the update
│     
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.11_B0032
      Bootloader Version:  BOT01.04_B0016
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:        • Updatable
                           • Supported on remote server

```

**Dock SKU**
Please mention which module is installed in your WD19.

- [ ] WD19 (Single-C)
- [X] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**
Please describe all devices connected to the dock.  Be as specific as possible,
including USB devices, hubs, monitors, and downstream type-C devices.

**Additional questions**
- Operating system and version: Ubuntu 20.10
- Have you tried unplugging the dock or any peripherals from your machine? YES
- Have you tried to power cycle the dock from the AC adapter? YES
- Is this a regression? NO

joakim-tjernlund, 2021-03-05:
Having to physically unplug anything is a bug. It prevents  IT admins to manage upgrades remotely.

superm1, 2021-03-05:
> Having to physically unplug anything is a bug. It prevents IT admins to manage upgrades remotely.

On the contrary no IT admin should be remotely applying updates to anything without the flag `FWUPD_DEVICE_FLAG_USABLE_DURING_UPDATE`.  This flag indicates that the device will apply all the updates but there is no risk for data loss.  Users can connect any number of peripherals behind a dock and actively be using them.

This flag was introduced into fwupd **because** of the update flow [introduced](https://patents.google.com/patent/US20200143057A1/en?inventor=mario+limonciello&oq=mario+limonciello) for the WD19 on Linux.  All updates will be staged into secure regions on the device and updates completed when any of these occur:

1. The administrator intentionally calls `fwupdtool activate`.
2. The system is rebooted.  `fwupdtool activate` will be called by `systemd`.
3. The USB-C port is unplugged from the host.  Such as a user undocking.

> A more clear message. If possible change the text to something like "... Updates will complete after the USB cable connecting the dock and laptop is unplugged..."

I guess I see 3 possible approaches to helping this confusion:

1. `Updates will complete next time you undock or reboot`
2. `Updates will complete after the USB cable is unplugged`
3. Remove the message all together.  Everyone _eventually_ unplugs or reboots.

joakim-tjernlund, 2021-03-05:
> On the contrary no IT admin should be remotely applying updates to anything without the flag `FWUPD_DEVICE_FLAG_USABLE_DURING_UPDATE`. This flag indicates that the device will apply all the updates [...]

I see, my mistake. Sorry for the noise.

ckruetze, 2021-03-05:
> I guess I see 3 possible approaches to helping this confusion:
> 
>     1. `Updates will complete next time you undock or reboot`
> 
>     2. `Updates will complete after the USB cable is unplugged`
> 
>     3. Remove the message all together.  Everyone _eventually_ unplugs or reboots.

I think 1. or 2. would work. I think 2. is the clearest. Regarding your message in 1. would the update actually work with a simple reboot? The dock would have a constant connection so I'm not sure if that is enough.

No 3. I wouldn't find so useful. As after an update I'm sure I'm not the only one who checks if it worked. And without unplugging anything the update would just run again and again. I think a message stating that it will be completed after the next undock or unplug of the USB cable would be best.

superm1, 2021-03-05:
> I think 1. or 2. would work. I think 2. is the clearest. Regarding your message in 1. would the update actually work with a simple reboot? The dock would have a constant connection so I'm not sure if that is enough.

Normally - yes.  `fwupdtool activate` is called by default on the shutdown path for systems using systemd.  If you pretend that's not there, it depends on whether the host PD controller resets when the host resets.

To avoid those potential corner cases (no systemd or PD controller doesn't reset when host resets) I think I lean on 2 as well.

@hughsie to update this I think it needs LVFS admin rights.

Here's what I think we're landing on (but feel free to add more input if you disagree)

```
Updates will complete next time the dock USB cable is unplugged.  The dock will be unavailable for approximately 1 minute.
```

hughsie, 2021-03-05:
> I think it needs LVFS admin rights.

I suspect it's more "we didn't wire up a text box" -- give me a couple of hours.

hughsie, 2021-03-05:
Is this the `Update Message`? If so that's changable on the LVFS in the component updates panel, no?

superm1, 2021-03-08:
> Is this the `Update Message`? If so that's changable on the LVFS in the component updates panel, no?

Double checked it and didn't see it present, no.

superm1, 2021-03-08:
Oh the box is missing while in stable, but present while in embargo.  I moved it back, modified it and moved it to stable.
Thanks~

closed 2021-03-08 by superm1