fwupd/fwupd #2976: fwupdmgr 1.5.7 doesn't show any T470 related devices anymore

Roxxor91, 2021-03-05:
**Describe the bug**
I am using Debian Unstable, and was very happy with updating my Lenovo T470 via the LVFS. But now since the upgrade to 1.5.7 it only shows my Intel SSD.

**Steps to Reproduce**
Upgrade to fwupdmgr 1.5.7

**Expected behavior**
Show BIOS version, Intel ME, etc.

**fwupd version information**
```fwupdmgr --version
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5
daemon version:	1.5.7
```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Debian Unstable apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
20HES01100
│
├─HD Graphics 620:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              164325d0-cc40-51e0-9935-b1fabe2a686b ← PCI\VEN_8086&DEV_5916&SUBSYS_17AA2245&REV_02
│                         943b8145-a3aa-5e38-9a26-6f3dd2bb1f0f ← PCI\VEN_8086&DEV_5916&SUBSYS_17AA2245
│                         2886e312-afd8-5510-8993-12d568a85e00 ← PCI\VEN_8086&DEV_5916&REV_02
│                         38f6c009-a25c-5b50-b3b5-fef4f9c6b846 ← PCI\VEN_8086&DEV_5916
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─INTEL SSDPEKNW020T8:
│     Device ID:          c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Zusammenfassung:    NVM Express Solid State Drive
│     Current version:    002C
│     Vendor:             Intel Corporation (NVME:0x8086)
│     Serial Number:      BTNH90250CJ32P0C
│     GUIDs:              c5fe8b70-dc9a-5c3b-9634-659091d29812 ← NVME\VEN_8086&DEV_F1A8&SUBSYS_8086390D&REV_03
│                         1122104f-b10a-5f32-bc13-7a1ac0f52ea2 ← NVME\VEN_8086&DEV_F1A8&SUBSYS_8086390D
│                         c6cd9ab0-8f20-512e-9e1f-1af55b8454b9 ← NVME\VEN_8086&DEV_F1A8&REV_03
│                         82741c78-f5dc-5c23-a152-00de5799edc8 ← NVME\VEN_8086&DEV_F1A8
│                         7c0505be-b609-566e-bf94-561e441a9171 ← INTEL SSDPEKNW020T8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i5-7200U CPU @ 2.50GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000de
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         9ca69899-3716-5857-9fd3-882a5c73236f ← CPUID\PRO_0&FAM_06&MOD_8E&STP_9
│     Device Flags:       • Internal device
│   
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Current version:    71.4.17568.3603
      Vendor:             ST Microelectronics (TPM:STM)
      GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
                          84df3581-f896-54d2-bd1a-372602f04c32 ← TPM\VEN_STM&DEV_0001
                          bfaed10a-bbc1-525b-a329-35da2f63e918 ← TPM\VEN_STM&MOD_
                          70b7b833-7e1a-550a-a291-b94a12d0f319 ← TPM\VEN_STM&DEV_0001&VER_2.0
                          06f005e9-cb62-5d1a-82d9-13c534c53c48 ← TPM\VEN_STM&MOD_&VER_2.0
      Device Flags:       • Internal device
```

**Additional questions**
- Operating system and version: Debian Unstable
- Have you tried rebooting? Yes
- Is this a regression? Yes

hughsie, 2021-03-05:
What does `fwupdmgr get-plugins` say?

Roxxor91, 2021-03-05:
```
linux_tainted:
  Flags:                • Aktiviert

uefi_pk:
  Flags:                • Disabled

linux_lockdown:
  Flags:                • Aktiviert

cros_ec:
  Flags:                • Aktiviert

ebitdo:
  Flags:                • Aktiviert

logind:
  Flags:                • Aktiviert

goodixmoc:
  Flags:                • Aktiviert

tpm:
  Flags:                • Aktiviert

jabra:
  Flags:                • Aktiviert

system76_launch:
  Flags:                • Aktiviert

ccgx:
  Flags:                • Aktiviert

pixart_rf:
  Flags:                • Aktiviert

colorhug:
  Flags:                • Aktiviert

synaptics_rmi:
  Flags:                • Aktiviert

thelio_io:
  Flags:                • Aktiviert

steelseries:
  Flags:                • Aktiviert

uefi_dbx:
  Flags:                • Disabled

altos:
  Flags:                • Aktiviert

solokey:
  Flags:                • Aktiviert

synaptics_cxaudio:
  Flags:                • Aktiviert

rts54hid:
  Flags:                • Aktiviert

logitech_hidpp:
  Flags:                • Aktiviert

optionrom:
  Flags:                • Aktiviert

linux_sleep:
  Flags:                • Aktiviert

acpi_dmar:
  Flags:                • Aktiviert

fastboot:
  Flags:                • Aktiviert

bcm57xx:
  Flags:                • Aktiviert

elantp:
  Flags:                • Aktiviert

flashrom:
  Flags:                • Aktiviert

pci_mei:
  Flags:                • Aktiviert

rts54hub:
  Flags:                • Aktiviert

amt:
  Flags:                • Disabled

upower:
  Flags:                • Aktiviert

dell_dock:
  Flags:                • Aktiviert

pci_bcr:
  Flags:                • Aktiviert

dell_esrt:
  Flags:                • Disabled
                        • Required hardware was not found

cpu:
  Flags:                • Aktiviert

synaptics_prometheus:
  Flags:                • Aktiviert

nvme:
  Flags:                • Aktiviert

bios:
  Flags:                • Disabled

fresco_pd:
  Flags:                • Aktiviert

wacom_raw:
  Flags:                • Aktiviert

superio:
  Flags:                • Aktiviert

ata:
  Flags:                • Aktiviert

wacom_usb:
  Flags:                • Aktiviert

acpi_facp:
  Flags:                • Aktiviert

redfish:
  Flags:                • Disabled
                        • Required hardware was not found

iommu:
  Flags:                • Aktiviert

synaptics_mst:
  Flags:                • Aktiviert

emmc:
  Flags:                • Aktiviert

vli:
  Flags:                • Aktiviert

linux_swap:
  Flags:                • Aktiviert

nitrokey:
  Flags:                • Aktiviert

hailuck:
  Flags:                • Aktiviert

dfu:
  Flags:                • Aktiviert

ep963x:
  Flags:                • Aktiviert

dfu_csr:
  Flags:                • Aktiviert

thunderbolt:
  Flags:                • Aktiviert

msr:
  Flags:                • Aktiviert

tpm_eventlog:
  Flags:                • Disabled

uefi_capsule:
  Flags:                • Aktiviert

dell:
  Flags:                • Disabled
                        • Required hardware was not found

uefi_recovery:
  Flags:                • Disabled
                        • Required hardware was not found
```

hughsie, 2021-03-05:
> uefi_dbx Disabled

This is weird. Are you still booting in EFI mode? What does `sudo fwupdtool get-devices --verbose` output?

Roxxor91, 2021-03-05:
```
12:17:36:0695 FuDebug              Verbose debugging enabled (on console 1)
12:17:36:0708 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
12:17:36:0709 FuConfig             loading config values from /etc/fwupd/daemon.conf
12:17:36:0709 FuConfig             using autodetected max archive size 18,4 EB
12:17:36:0712 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
12:17:36:0713 FuRemoteList         loading /usr/share/fwupd/metainfo
12:17:36:0713 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:17:36:0713 XbSilo               file: 5c258d7c-b352-6800-e9eb-7fae2bd89ed4, current:5c258d7c-b352-6800-e9eb-7fae2bd89ed4, cached: (null)
12:17:36:0713 XbSilo               loading silo with file contents
12:17:36:0713 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
12:17:36:0714 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
12:17:36:0714 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
12:17:36:0715 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
12:17:36:0715 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
12:17:36:0715 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.dpkg-old
12:17:36:0715 FuRemoteList         ignoring unfound remote fwupd
12:17:36:0715 FuRemoteList         ordering lvfs-testing=lvfs+1
12:17:36:0716 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:17:36:0716 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:17:36:0716 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
12:17:36:0716 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:17:36:0716 Jcat                 loaded 1 certificates
12:17:36:0716 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
12:17:36:0716 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:17:36:0716 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:17:36:0716 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
12:17:36:0716 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:17:36:0717 Jcat                 loaded 1 certificates
12:17:36:0717 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
12:17:36:0717 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
12:17:36:0743 FuEngine             client certificate exists and working
12:17:36:0743 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
12:17:36:0744 FuHistory            got schema version of 6
12:17:36:0744 FuHwids              smbios property Manufacturer=LENOVO
12:17:36:0744 FuHwids              smbios property EnclosureKind=a
12:17:36:0744 FuHwids              smbios property Family=ThinkPad T470
12:17:36:0744 FuHwids              smbios property ProductName=20HES01100
12:17:36:0744 FuHwids              smbios property ProductSku=LENOVO_MT_20HE_BU_Think_FM_ThinkPad T470
12:17:36:0744 FuHwids              smbios property BiosVendor=LENOVO
12:17:36:0744 FuHwids              smbios property BiosVersion=N1QET90W (1.65 )
12:17:36:0744 FuHwids              smbios property BiosMajorRelease=01
12:17:36:0744 FuHwids              smbios property BiosMinorRelease=41
12:17:36:0744 FuHwids              smbios property BaseboardManufacturer=LENOVO
12:17:36:0744 FuHwids              smbios property BaseboardProduct=20HES01100
12:17:36:0755 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
12:17:36:0755 XbSilo               file: 00bd024e-8baf-399c-824d-b4879bbe28c7, current:00bd024e-8baf-399c-824d-b4879bbe28c7, cached: (null)
12:17:36:0755 XbSilo               loading silo with file contents
12:17:36:0757 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
12:17:36:0757 XbSilo               file: f48625b6-2a0d-a9bf-806e-385c120d962c, current:f48625b6-2a0d-a9bf-806e-385c120d962c, cached: (null)
12:17:36:0757 XbSilo               loading silo with file contents
12:17:36:0759 FuEngine             703 components now in silo
12:17:36:0778 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_linux_tainted.so)
12:17:36:0779 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_uefi_pk.so)
12:17:36:0780 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_linux_lockdown.so)
12:17:36:0781 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_cros_ec.so)
12:17:36:0782 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_ebitdo.so)
12:17:36:0783 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_logind.so)
12:17:36:0784 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_goodixmoc.so)
12:17:36:0808 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_tpm.so)
12:17:36:0808 FuPlugin             added udev subsystem watch of tpm
[... cut ...]
12:17:37:0745 FuPlugin             destroy(logind)
12:17:37:0745 FuPlugin             destroy(upower)
12:17:37:0746 FuPlugin             destroy(linux_lockdown)
12:17:37:0746 FuPlugin             destroy(tpm_eventlog)
12:17:37:0746 FuPlugin             destroy(dell)
12:17:37:0746 FuPlugin             destroy(flashrom)
12:17:37:0747 FuPlugin             destroy(linux_swap)
12:17:37:0747 FuPlugin             destroy(linux_tainted)
12:17:37:0747 FuPlugin             destroy(redfish)
12:17:37:0747 FuPlugin             destroy(uefi_capsule)
```

Roxxor91, 2021-03-05:
I am sorry for wasting your time, somehow it seems my booting sequence changed from UEFI to legacy. From the beginning I had both enabled, so UEFI + Legacy boot. I never changed it. Now after you asked if I am booting through UEFI, I enabled UEFI only, deactivating LEgay for good, and my devices now show up again.
No clue why it changed, but the issue is solved. Thank you. My mistake, fwupdmgr works as it should!

closed 2021-03-05 by Roxxor91