fwupd/fwupd #2987: UEFI firmware of MSI mainboard not listed

paulmenzel, 2021-03-09:
**Describe the bug**
On the [MSI B350M MORTAR (MS-7A37)](https://de.msi.com/Motherboard/support/B350M-MORTAR.html) with an UEFI firmware, I am sure, in the past `fwupdmgr get-devices` listed the firmware, but said that UEFI capsules wasn’t enabled. With version 1.5.7 (not sure since when) the item is not listed anymore.

```
$ fwupdmgr --version
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.7

$ fwupdmgr get-devices --show-all-devices
MS-7A37
│
├─300 Series Chipset SATA Controller:
│     Device ID:          5299e802217cae06dbae3c38b1311bee84b7d318
│     Current version:    02
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              2752f8e3-f453-5b2c-b26e-fbe10c23be07
│                         8b139035-0f55-5b27-b1bc-c2c0064addd4
│                         ef4a96a0-45d1-5cff-90d3-e6f387d130e7
│                         17f97e61-e971-53d5-bb9e-6bcef9c6c23f
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─AMD Ryzen 3 2200G with Radeon Vega Graphics    :
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         375b07af-58d5-5172-ab62-510c2229bf79
│                         2f8267cd-b115-5138-b04d-3cb13badf673
│     Device Flags:       • Internal device
│
└─CT1000MX500SSD4:
      Device ID:          bf97ef88a4eda3055c079a95e0636fe2887d5436
      Zusammenfassung:    ATA Drive
      Current version:    M3CR020
      Vendor:             Micron (ATA:0x1344|OUI:00a075)
      GUIDs:              7fe72eb9-09fe-510e-a062-eb07f185035b
                          1d4f3437-add2-559d-bd88-0a10880e59d3
                          49811b52-642d-5307-bc6f-32e2033fb26a
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```

The mainboard firmware is not listed.

```
$ sudo efibootmgr -v
BootCurrent: 0000
Timeout: 1 seconds
BootOrder: 0000,0001
Boot0000* debefistub	HD(1,MBR,0x64fda70b,0x800,0x80000)/File(\EFI\DEBEFI\VMLINUZ)r.o.o.t.=.U.U.I.D.=.8.8.8.3.f.7.3.3.-.2.2.4.8.-.4.7.e.0.-.9.0.b.9.-.e.e.5.3.8.4.f.1.8.d.6.2. .r.o. .q.u.i.e.t. .n.o.i.s.a.p.n.p. .c.r.y.p.t.o.m.g.r...n.o.t.e.s.t.s. .r.a.n.d.o.m...t.r.u.s.t._.c.p.u.=.o.n. .s.e.l.i.n.u.x.=.0. .a.p.p.a.r.m.o.r.=.0. .k.m.e.m.l.e.a.k.=.o.f.f. .i.n.i.t.r.d.=.\.E.F.I.\.d.e.b.e.f.i.\.I.N.I.T.R.D.
Boot0001* debian	HD(1,MBR,0x64fda70b,0x800,0x80000)/File(\EFI\DEBIAN\GRUBX64.EFI)
```

```
$ sudo ls -lR /boot/efi/
/boot/efi/:
insgesamt 32769
-rwx------ 1 root root 16777216 25. Jan 2019  E7A37AMS.1J0
-rwx------ 1 root root 16777216  8. Mär 2019  E7A37AMS.1K0
drwx------ 4 root root      512  2. Apr 2020  EFI

/boot/efi/EFI:
insgesamt 1
drwx------ 2 root root 512  2. Apr 2020  debefi
drwx------ 2 root root 512 16. Jul 2018  debian

/boot/efi/EFI/debefi:
insgesamt 25432
-rwx------ 1 root root  7801303  8. Mär 19:57 INITRD
-rwx------ 1 root root 18241024  8. Mär 19:57 VMLINUZ

/boot/efi/EFI/debian:
insgesamt 132
-rwx------ 1 root root 135168  4. Mär 11:52 grubx64.efi
```

hughsie, 2021-03-09:
Can you attach the output of `sudo fwupdtool get-devices --show-all -v` please.

superm1, 2021-03-09:
> I am sure, in the past fwupdmgr get-devices listed the firmware, but said that UEFI capsules wasn’t enabled

I'm a bit confused, it sounds like this is WAD then, no?  We shouldn't be listing the UEFI firmware here if it's not updatable and there is nothing you can do about it.

paulmenzel, 2021-03-09:
On a Dell Latitude E7250 at least a warning is shown:

```
$ fwupdmgr get-devices 
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Latitude E7250
│
├─SAMSUNG SSD PM851 mSATA 256GB:
│     Device ID:          0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
│     Zusammenfassung:    ATA Drive
│     Current version:    EXT4AD0Q
│     Vendor:             Samsung (ATA:0x144D|OUI:002538)
│     GUIDs:              16edd6b3-f18c-5eee-be5a-e7cb89d592f2
│                         cae6f248-5571-598e-8d17-226faea2805e
│                         3be83754-dd2c-556e-8c44-2fd08f362ade
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Zusammenfassung:    UEFI Revocation Database
      Current version:    0
      Minimum Version:    0
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      GUIDs:              c6682ade-b5ec-57c4-b687-676351208742
                          f8ba2887-9411-5c36-9cee-88995bb39731
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation
    
$ fwupdmgr get-devices --show-all-devices
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Latitude E7250
│
├─Event Log:
│     Device ID:          58bd405f31c48e6eca290b425f530a94c91e955c
│     GUID:               a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│     Device Flags:       • Internal device
│   
├─HD Graphics 5500:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    09
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              5f5a0a08-69b7-5b63-91fc-c3e20c2f0b58
│                         3c729b01-2419-57c8-8dda-052bab778c92
│                         19e1f63e-bef0-5582-b236-364069bcd0d5
│                         64c0ea55-d861-5dd0-aa6b-bcc307f57ed4
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel AMT [unprovisioned]:
│     Device ID:          e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Zusammenfassung:    Hardware and firmware technology for remote out-of-band management
│     Current version:    10.0.56.3002
│     Bootloader Version: 10.0.56.3002
│     Vendor:             Intel Corporation
│     GUID:               2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:       • Internal device
│   
├─Intel(R) Core™ i7-5600U CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000002f
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         bcad68bd-c33a-56cd-a8a6-b9a78d62ee6e
│                         a702f59f-10c2-5561-976c-b94c89cb42d8
│     Device Flags:       • Internal device
│   
├─SAMSUNG SSD PM851 mSATA 256GB:
│     Device ID:          0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
│     Zusammenfassung:    ATA Drive
│     Current version:    EXT4AD0Q
│     Vendor:             Samsung (ATA:0x144D|OUI:002538)
│     GUIDs:              16edd6b3-f18c-5eee-be5a-e7cb89d592f2
│                         cae6f248-5571-598e-8d17-226faea2805e
│                         3be83754-dd2c-556e-8c44-2fd08f362ade
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Zusammenfassung:    UEFI Revocation Database
      Current version:    0
      Minimum Version:    0
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      GUIDs:              c6682ade-b5ec-57c4-b687-676351208742
                          f8ba2887-9411-5c36-9cee-88995bb39731
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation
    
```

superm1, 2021-03-09:
This flag gets set by Dell plugin:
https://github.com/fwupd/fwupd/blob/117f857899d22a0d3d6414bd2239ccce1684820e/plugins/dell/fu-plugin-dell.c
On Dell side we have the way to turn it on.

hughsie, 2021-04-11:
> Can you attach the output of sudo fwupdtool get-devices --show-all -v please.

Still this please.

paulmenzel, 2021-04-20:
I am sorry for the long delay.

[Output of `sudo fwupdtool get-devices --show-all -v`](https://github.com/fwupd/fwupd/files/6341906/fwupdtool.get-devices--show-all-v.txt)

hughsie, 2021-04-20:
> 08:20:53:0217 FuPluginUefiCapsule  failed to add /sys/firmware/efi/esrt/entries/entry0: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid

What does this list please: `sudo cat /sys/firmware/efi/esrt/entries/entry*/fw_class`

paulmenzel, 2021-04-20:
$ sudo cat /sys/firmware/efi/esrt/entries/entry*/fw_class
    00000000-0000-0000-0000-000000000000

hughsie, 2021-04-20:
Hmm, so fwupd is doing exactly the right thing I'm afraid to say; `00000000-0000-0000-0000-000000000000` isn't valid, but it's possible we were creating a device for that in the past. I'd say you need to flash a new firmware using something like winflash or the MSI tools, and then fingers crossed a new firmware would actually create the ESRT entries. I don't think it's a fwupd bug at all, sorry.

closed 2021-04-20 by hughsie

paulmenzel, 2021-04-20:
It’s the [latest firmware from July 2020](https://de.msi.com/Motherboard/support/B350M-MORTAR.html#down-bios). Linux logs:

    [    0.000000] efi: ACPI 2.0=0x5c04a000 ACPI=0x5c04a000 SMBIOS=0x5cdd9000 MEMATTR=0x58695698 ESRT=0x5869bf98 RNG=0x5cdf3f18
    […]
    [    0.000000] esrt: Reserving ESRT space from 0x000000005869bf98 to 0x000000005869bfd0.

paulmenzel, 2021-04-20:
Ok, so the verbose log also contains:

    08:20:53:0217 FuPluginUefiCapsule  failed to add /sys/firmware/efi/esrt/entries/entry0: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid

paulmenzel, 2021-04-20:
… similar to [Invalid ESRT GUID error shortly after boot](https://askubuntu.com/questions/1215052/invalid-esrt-guid-error-shortly-after-boot).

paulmenzel, 2021-04-20:
From [Populating the ESRT table](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/populating-the-esrt-table):

> The EFI System Resource Table (ESRT) provides a mechanism for identifying integrated device and system firmware resources for the purposes of targeting firmware updates to those resources. Each entry in the ESRT describes a device or system firmware resource that can be targeted by a firmware update driver package. Each firmware resource that can be updated by a firmware update driver package must be described by exactly one entry in the ESRT to enable firmware updates to be deployed and installed.

So, my system does not support UEFI Capsule updates, so it’s expected.

I would still have thought, fwupdmgr listing the device with a note, that UEFI Capsule updates are unsupported.

superm1, 2021-04-29:
> From [Populating the ESRT table](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/populating-the-esrt-table):
>
>
> So, my system does not support UEFI Capsule updates, so it’s [...]

We used to do something like this in the past where a dummy device was present.  That was removed later on though as part of https://github.com/fwupd/fwupd/commit/7bcb8d4385665700f21b246747b69ab5aff0c136#diff-39ab7e5ff545c0c59f5c8a5492e6aa4b4d1282f93663542155eafcad43e8e9e6L37

To be honest I'm really not sure it's worth efforts to try to add anything else for this use case of system in UEFI mode but no capsules supported.  The whole point of fwupd is to be able to _update_ firmware.
So in this case it's effectively read only information, and trivially possible to read the information manually by `/sys/class/dmi/id/bios_version`.