fwupd/fwupd #3012: cros-ec-usb-device: Retry write process in the event of failures due to interrupted usb connection

bleungatchromium, 2021-03-11:
**Describe the bug**
A clear and concise description of what the bug is.
During development of quiche, I've observed that the RO firmware's USB endpoint will occasionally reset briefly upon rebooting into it. This is enough to interrupt a write in progress, which causes the cros-ec plugin to fail write, and bail out entirely.

**Steps to Reproduce**
Steps to reproduce the behavior.
Flash quiche from RW -> RO -> RW several times. Eventually should fail like this:
04:49:33:0957 FuEngine             Emitting PropertyChanged('Status'='device-write')
Writing?                 [                     \                 ]04:49:33:0957 FuPluginCrosEc       trimmed 584 trailing bytes
04:49:33:0957 FuPluginCrosEc       sending 0x2fdb8 bytes to 0x10000
04:49:33:0970 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0970 FuPluginCrosEc       failed to flush to idle
04:49:33:0970 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0970 FuPluginCrosEc       failed to flush to idle
04:49:33:0970 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0970 FuPluginCrosEc       failed to flush to idle
04:49:33:0971 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0971 FuPluginCrosEc       failed to flush to idle
04:49:33:0971 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0971 FuPluginCrosEc       failed to flush to idle
04:49:33:0972 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0972 FuPluginCrosEc       failed to flush to idle
04:49:33:0972 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0972 FuPluginCrosEc       failed to flush to idle
04:49:33:0972 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0972 FuPluginCrosEc       failed to flush to idle
04:49:33:0972 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0972 FuPluginCrosEc       failed to flush to idle
04:49:33:0973 FuDevice             fu_device_retry_full: assertion 'error != NULL' failed
04:49:33:0973 FuPluginCrosEc       failed to flush to idle

**Expected behavior**
A clear and concise description of what you expected to happen.
Retry. Actually, just rerunning fwupdtool install-blob again with the device sitting idle in the RO section continues the update 

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Self built as a part of chromiumos.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version: Chrome OS
- Have you tried rebooting? Not relevant.
- Is this a regression? Nope.

hughsie, 2021-03-11:
>  04:49:33:0970 FuDevice fu_device_retry_full: assertion 'error != NULL' failed

This concerns me a bit; this shouldn't happen. Can you reproduce under gdb, with `G_DEBUG=fatal-criticals` set please -- and then backtrace on the critical warning. It looks like the root cause is not returning from somewhere when setting the error.

hughsie, 2021-03-11:
also, maybe grab the `error->message` when in the right stack frame in gdb pls.

bleungatchromium, 2021-03-11:
Ok, working on getting you that info.

bleungatchromium, 2021-03-11:
Ok, I found the problem.

Looks like I pass NULL as error into retry in 3 places in my code. I'll put up a change to fix that.

bleungatchromium, 2021-03-11:
That isn't the root cause of the problem though, only extra spam around the retry that fails. (definitely worth fixing, as the retry logic was busted before).

The root cause is that the underling USB device disconnects and then reenumerates, so we are operating on a stale handle. That's why going around again with ANOTHER_WRITE_REQUIRED helps.

bleungatchromium, 2021-03-11:
Actuall, @hughsie, i just saw the change you pushed to https://github.com/fwupd/fwupd/commit/11b49a559256b73df9c1a89bbfef5f9eab4d952a

That looks like it addresses the cause of the assertion too in a better way. I'll sanity check it real quick.

PR #3013 (2021-03-13): cros-ec: Allow a failure to write a section, do another write
The RO partition's USB endpoint has been observed to occasionally hiccup
(usb disconnect and reenumeration) upon reboot and initial attach, causing a
failure to update a section. Rather that making this fatal, let's set
ANOTHER_WRITE_REQUIRED and allow us to handle the new USB device after
this event.

fixes #3012

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit f7832847 (2021-03-13): cros-ec: Allow a failure to write a section, do another write

The RO partition's USB endpoint has been observed to occasionally hiccup
(usb disconnect and reenumeration) upon reboot and initial attach, causing a
failure to update a section. Rather that making this fatal, let's set
ANOTHER_WRITE_REQUIRED and allow us to handle the new USB device after
this event.

fixes #3012
commit 1c691e26 (2021-03-24): cros-ec: Allow a failure to write a section, do another write

The RO partition's USB endpoint has been observed to occasionally hiccup
(usb disconnect and reenumeration) upon reboot and initial attach, causing a
failure to update a section. Rather that making this fatal, let's set
ANOTHER_WRITE_REQUIRED and allow us to handle the new USB device after
this event.

fixes #3012
closed 2021-03-13 by bleungatchromium