fwupd/fwupd #3019: Establish locking mechanism to prevent two simultaneous instances of fwupdtool

djcampello, 2021-03-11:
**Is your feature request related to a problem? Please describe.**
If two asynchronous instances of fwupdtool try to access the database these leads to issues

**Describe the solution you'd like**
Simple file locking mechanism should suffice

**Describe alternatives you've considered**
File locking external to fwupd wrapping fwupdtool invocations

hughsie, 2021-03-11:
>  If two asynchronous instances of fwupdtool try to access the database these leads to issues

What are you trying to do? The database was really designed for multiple readers and one writer (the daemon) -- although I don't see a problem with two things trying to write to sqlite at the same time.

>  File locking external to fwupd wrapping fwupdtool invocations

Hmm, I think the solution here is to use the daemon which will serialize access to both the database, and more importantly **the hardware** too.

djcampello, 2021-03-11:
I had some SQLite errors showing up when a `fwupdtool get-devices` executed while a `fwupdtool update` was also executing. My guess is that the later was writing the database to mark the device as "pending for activation" when the former executed. 

I agree with the daemon solving the problem. While this is not the case for us, I am looking to have a temporary solution to serialize multiple pending fwupdtool update executions (all triggered from upstart). I was thinking on a file lock that I could wait on.

hughsie, 2021-03-11:
> when a fwupdtool get-devices executed while a fwupdtool update was also executing

I really don't think that's safe at all. Probing the hardware while it's being updated isn't tested in any way and is probably a Bad Idea™.

> I was thinking on a file lock

I think we can do something better than that. Why don't we make fwupdtool also "own" the org.freedesktop.fwupd DBus interface for the entire time it's running? The benefits is that it would ensure the daemon quits when safe, and it would wait for other fwupdtool instances to quit before continuing to the user initiated action.

djcampello, 2021-03-11:
If a fwupdtool instance is running and the user launches another one, will it fail or hang until it can acquire the DBus interface?

hughsie, 2021-03-11:
> will it fail or hang until it can acquire the DBus interface

I believe you can configure GDBus in either way. I suspect blocking with a message is what the user would expect.

superm1, 2021-03-11:
> I believe you can configure GDBus in either way. I suspect blocking with a message is what the user would expect.

We do require you to stop `fwupd.service` if you start `fwupdtool`, and so that jives pretty well with that too.

closed 2021-04-15 by hughsie