fwupd/fwupd #3025: Documentation generation doesn't work in cross builds

rossburton, 2021-03-12:
The man pages are generated using `help2man`, which doesn't work in cross builds where you can't execute the target binaries.

hughsie, 2021-03-12:
Does `-Dman=false` work or does that still try to call `help2man`?

rossburton, 2021-03-12:
That works, but leads to no documentation.

hughsie, 2021-03-12:
I'm not sure what the solution is; i mean we switched to help2man from docbook as the documentation was just getting out of sync with reality.

superm1, 2021-03-12:
Maybe we need to use the launcher that runs tests as a launcher for help2man?

It's usually qemu based and should work in cross builds I'd expect.

hughsie, 2021-03-12:
So we'd run help2man with `exe_wrapper` -- I guess that makes sense -- for ARM it would be `qemu-aarch64` i suppose. How you make `custom_target` use `exe_wrapper` I have no idea....

hughsie, 2021-03-12:
According to the Meson folk, custom_target should be using exe_wrapper. Can you share your cross file here to see if I can reproduce?

xclaesse, 2021-03-12:
Gave it a try and that's indeed a bug in Meson, at least since 0.56 but probably since forever. run_target() would plug the exe_wrapper but not custom_target(). Unfortunately there are tones of little inconsistencies in the way command are handled by different meson APIs because they share too little common code path, I started refactoring that in 0.57 but the more I dig the worst it gets :(

Anyway.. please report in Meson :D

xclaesse, 2021-03-12:
wait, assuming you're cross building help2man, right?

hughsie, 2021-03-12:
> assuming you're cross building help2man

IIUC, we want to run the custom_target with the exe_wrapper as the help2man requires us to run the _just built_ binary.

xclaesse, 2021-03-12:
ahhh, help2man takes an executable in arguments, ok... I thought in your case help2man itself was a cross built binary (in which case Meson won't use exe_wrapper but should). 

But in your case Meson won't add the wrapper neither... dunno if it should...

xclaesse, 2021-03-12:
Even if Meson gives you the exe wrapper, I don't know if that can work because help2man seems to accept only one argument. You can dont `help2man qemu-aarch64 myapp`, can you?

hughsie, 2021-03-13:
I assumed help2man was cross-built too; e.g. `qemu-aarch64 help2man myapp` -- @rossburton is that accurate?

rossburton, 2021-03-13:
In my situation, I can get either native or cross.

xclaesse, 2021-03-15:
Even if Meson gives you the exe wrapper, I don't know if that can work because help2man seems to accept only one argument. You can't do `help2man qemu-aarch64 myapp`, can you? The only way this can work is either cross compiling help2man and run it in qemu (meson can do that) or pass a python wrapper to help2man, so when help2man execute your script with --version, the script can itself execute your app with qemu and pipe stdout.

ericonr, 2021-03-30:
Probably not what you want to hear, but in Void Linux we just steal the man pages from the native fwupd version.

hughsie, 2021-03-30:
I'm wondering if we should generate the man pages a different way entirely.

superm1, 2021-03-30:
>You can't do help2man qemu-aarch64 myapp, can you?

I would think the answer here is [qemu-user and binfmt_misc](https://ownyourbits.com/2018/06/13/transparently-running-binaries-from-any-architecture-in-linux-with-qemu-and-binfmt_misc/)

closed 2021-04-11 by hughsie