fwupd/fwupd #3036: ESP not detected when on raid

ausil, 2021-03-17:
**Describe the bug**
A clear and concise description of what the bug is.
My system's esp partition is on a raid1 and I am getting an error that there is no ESP partition

[root@kasuf ~]# mount
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime,seclabel)
devtmpfs on /dev type devtmpfs (rw,nosuid,seclabel,size=32854196k,nr_inodes=8213549,mode=755,inode64)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev,seclabel,inode64)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,seclabel,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,seclabel,size=13165440k,nr_inodes=819200,mode=755,inode64)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,seclabel,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime,seclabel)
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,nosuid,nodev,noexec,relatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
/dev/mapper/fedora_kasuf-root on / type ext4 (rw,relatime,seclabel)
selinuxfs on /sys/fs/selinux type selinuxfs (rw,nosuid,noexec,relatime)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=31,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=31966)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime,seclabel)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime,seclabel)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,seclabel,pagesize=2M)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime,seclabel)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,seclabel,size=32913596k,nr_inodes=409600,inode64)
/dev/md127 on /boot type ext4 (rw,relatime,seclabel)
/dev/md125 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=ascii,shortname=winnt,errors=remount-ro)
sunrpc on /var/lib/nfs/rpc_pipefs type rpc_pipefs (rw,relatime)
tmpfs on /run/user/217600001 type tmpfs (rw,nosuid,nodev,relatime,seclabel,size=6582716k,nr_inodes=1645679,mode=700,uid=217600001,gid=217600001,inode64)
gvfsd-fuse on /run/user/217600001/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=217600001,group_id=217600001)
portal on /run/user/217600001/doc type fuse.portal (rw,nosuid,nodev,relatime,user_id=217600001,group_id=217600001)
[root@kasuf ~]# /usr/bin/fwupdtool esp-list --verbose
15:42:29:0776 FuDebug              Verbose debugging enabled (on console 1)
15:42:29:0790 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0791 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0792 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0794 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0795 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0796 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0797 FuMain               no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b, falling back to ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
15:42:29:0809 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0810 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0812 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0814 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0815 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
15:42:29:0816 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b: no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7

**Steps to Reproduce**
Steps to reproduce the behavior.
setup the system with everything as a raid1 array 

**Expected behavior**
A clear and concise description of what you expected to happen.
ESP partition to be detected 

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.7

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
[... 306 lines cut ...]
│   └── entries
│       ├── 17ea27cffa564d2bab34ea58d202e50d-0-rescue.conf
│       ├── 17ea27cffa564d2bab34ea58d202e50d-5.11.3-300.fc34.x86_64.conf
│       ├── 17ea27cffa564d2bab34ea58d202e50d-5.11.5-300.fc34.x86_64.conf
│       └── 17ea27cffa564d2bab34ea58d202e50d-5.11.6-300.fc34.x86_64.conf
├── lost+found
├── memtest86+-5.31
├── System.map-5.11.3-300.fc34.x86_64
├── System.map-5.11.5-300.fc34.x86_64
├── System.map-5.11.6-300.fc34.x86_64
├── vmlinuz-0-rescue-17ea27cffa564d2bab34ea58d202e50d
├── vmlinuz-5.11.3-300.fc34.x86_64
├── vmlinuz-5.11.5-300.fc34.x86_64
└── vmlinuz-5.11.6-300.fc34.x86_64

15 directories, 108 files
```

**Additional questions**
- Operating system and version: Fedora 34
- Have you tried rebooting? yes 
- Is this a regression? yes 
- Are you using an NVMe disk? yes 
- Is secure boot enabled? yes 
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

hughsie, 2021-03-17:
Does this actually work when updating using a capsule? @vathpela is this something we support for firmware updates?

vathpela, 2021-03-17:
I don't see a reason it *shouldn't* work, unless the wrong mdraid metadata version was used, but then it probably wouldn't boot either.

ausil, 2021-03-17:
I was able to drop the cap file in the ESP partition and manually update it. I believe that it should work fine.

hughsie, 2021-03-18:
I didn't know EFI firmware could reconstruct a `linux_raid_member` partition. I'm not sure what to do here.

ausil, 2021-03-18:
as it is raid1 it is readable without being reconstructed

superm1, 2021-03-23:
In this case are you able to just hardcode the ESP in the `uefi_capsule.conf`?

ausil, 2021-03-27:
setting `OverrideESPMountPoint=/boot/efi` seems to be an option

superm1, 2021-03-29:
To me I think this is (currently) corner case enough that we should document on the wiki but no code changes and expected behavior.  @hughsie thoughts?

hughsie, 2021-03-29:
I agree. ESP on raid is kinda unusual...

superm1, 2021-03-29:
Documented in [wiki ](https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found) as ESP on RAID needs `uefi_capsule.conf` modifications

closed 2021-03-29 by superm1