fwupd/fwupd #3042: How to implement DFU support for some device that requires a HID-command first?

djcb, 2021-03-18:
I'm working on `fwupd` support for some microphones, and after reading all the docs (I think!) and some of the code, I have a few questions. I'm a bit new to USB programming, so please bear with me!

The device is upgradable using DFU, but it does not normally expose the DFU Interface; it needs to be sent an HID command and then wait a few secs before the interface appears.

So, I guess this precludes the device from appearing in `fwupdtool get-devices`, or? Note that I did add the device (PID/VID) to the dfu quirks file.

What would be the best way to implement the call-HID-command-then-wait-a-bit part, i.e., should I subclass `FuDfuDevice`? Are there any examples of devices doing something similar?

**fwupd version information**
```shell
build/src/fwupdmgr --version
client version:	1.5.7-124-geacee1c9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.7-123-g3b5523eb
```
Thank you in advance!

hughsie, 2021-03-18:
Hi Dirk -- I did send you email a few days ago, but it might have not arrived. Excuse me for pasting my original reply:

> The device is upgradable using DFU, but it does not normally expose the
> DFU Interface; it needs to be sent an HID command and then wait a few
> secs before the interface appears.

Quite a few other devices do this. The proper firmware USB pattern is
to expose a DFU "runtime" interface, so that fwupd can put the device
into bootloader more automatically without any additional code, but
for lots of different reasons (security through obscurity, existing
apps, interface limitations) some devices expect something custom.

> What would be the best way to implement the call-HID-command part, i.e.,
> should I subclass FuDfuDevice?

I think the only thing you need to do is implement the setup() and
detach() vfuncs -- this allows fwupd to put the device into dfu mode
and then the other plugin can "take over" and then return the device
back to runtime mode. You'd subclass FuHidDevice or FuUsbDevice.

Two good examples you can copy are
https://github.com/fwupd/fwupd/tree/master/plugins/nitrokey and
https://github.com/fwupd/fwupd/tree/master/plugins/system76-launch --
the former being a more complicated HID protocol and the latter being
a simpler not-quite-HID protocol.

The only other "gotcha" is that you need a way to get the current
firmware version from the device without detaching to DFU mode, and
that you also need to tell fwupd what the device VID/PID will be in
the bootloader mode. This is done using CounterpartGuid in plugin
quirks file.

If it helps we're also in #fwupd on freenode.

Richard.

djcb, 2021-03-18:
Oh, thank you for the super-swift reply! 

Indeed, I did not receive the message, but I guess it's useful to have it documented here so it may help others. So the trick is to subclass FuHidDevice / FuUsbDevice rather than FuDfuDevice... didn't think of that, but it makes sense.

Thanks for the hints, let me write some code :-)

closed 2021-03-22 by superm1