fwupd/fwupd #3054: Recognized devices flip flops between a subset vs the full set

sbconslt, 2021-03-22:
**Describe the bug**

As of late fwupdmgr seems to arbitrarily flip flop between recognizing only a subset (numbering 4) of the devices it should instead of the full set (numbering 7). In the past, this was always reliably consistent.

**Steps to Reproduce**

Downloading metadata with `fwupdmgr refresh --force` is sufficient to generate the differing device count.

**Expected behavior**

Previous to the appearance of this issue, the device count was entirely stable between runs of fwupdmgr, across boots, etc. What I have termed the "full set" is what should be recognized if things are working correctly end to end.

**fwupd version information**

Please provide the version of the daemon and client.
```shell
# fwupdmgr --version
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.7
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Installed with system, so via dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

Two samples follow. The first, when the subset (4 devices):

```shell
# fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 4 local devices supported

# fwupdmgr get-devices
20QD0000US
│
├─Thunderbolt host controller:
│     Device ID:          4cfc79db1dd498794d0c2f7a5eeea672c4b7a07c
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    44.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              74cdddba-f7b6-5a31-a3dc-64186b5a5491 ← THUNDERBOLT\VEN_0109&DEV_1801&REV_00
│                         18bb00e3-5d7a-55d3-831e-0e000c184b0e ← THUNDERBOLT\VEN_0109&DEV_1801
│                         c7920601-0cda-507f-851e-92229eb1d370 ← TBT-01091801-native
│                         96c6512b-0138-51da-b02f-f44f2c4c7744 ← TBT-01091801-native-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Prometheus:
│ │   Device ID:          d432baa2162a32c1554ef24bd8281953b9d07c11
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3158804
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   Serial Number:      54232081658158
│ │   GUIDs:              09e8bf16-3e69-50f5-bb66-c7a040248352 ← USB\VID_06CB&PID_00BD&REV_0000
│ │                       8088f861-6318-5b1e-9ce4-fbddbedb09ac ← USB\VID_06CB&PID_00BD
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:        8dcffb5d059857368caa2b7b22371ebd831b0c0c
[... 222 lines cut ...]
                          • Device is usable for the duration of the update
    
________________________________________________

Devices that have been updated successfully:

 • Prometheus (10.01.2959683 → 10.01.3158804)
 • WDC PC SN730 SDBQNTY-512G-1001 (11130101 → 11170101)
 • Intel Management Engine (192.68.1606 → 192.70.1652)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection):
0.	Do not upload reports at this time, but prompt again for future updates
1.	Do not upload reports, and never ask to upload reports for future updates
2.	Upload reports just this one time, but prompt again for future updates
3.	Upload reports this time and automatically upload reports after completing future updates
0
Request canceled
```

**Additional questions**

- Operating system and version: Fedora 33 Workstation, Linux x1 5.11.7-200.fc33.x86_64 #1 SMP Wed Mar 17 18:55:20 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
- Have you tried rebooting? Yes, rebooting can change the flip flop in either direction, but it can also even occur within the same boot.
- Is this a regression? Yes, it was not present until recently.

hughsie, 2021-03-23:
Super interesting, thanks for filing this issue. If you do `sudo fwupdtool --plugins uefi_capsule get-devices --verbose` do you get the 3 missing device or none at all? If none, can you upload the entire log to this issue please. If you do get 3 devices, do subsequent runs (maybe try ~10?) of `fwupdtool` ever fail? Thanks!

sbconslt, 2021-03-23:
Here is the output of `fwupdtool --plugins uefi_capsule get-devices --verbose` immediately after `fwupdmgr refresh --force` reported 4 recognized devices (the short set). Just scanning the output it certainly looks to me like the full set of devices nonetheless appears with `fwupdtool`, though I'm sure you can decipher it better than I can.

As you requested I repeated the `fwupdtool` command several more times for verification. Over a period of about 10 minutes I repeated the command a total of 5 times and captured its output. The output was identical every time besides for timestamps. I verified this by stripping the leading timestamps with `sed` and `diff`ing the output. Note that for the repeat tests, I did not rerun `fwupdmgr refresh --force`, only `fwupdtool`. Let me know if there's anything else you'd like me to try.

```
# fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 4 local devices supported

# fwupdtool --plugins uefi_capsule get-devices --verbose
20:43:28:0223 FuDebug              Verbose debugging enabled (on console 1)
20:43:28:0249 FuConfig             loading config values from /etc/fwupd/daemon.conf
20:43:28:0249 FuConfig             using autodetected max archive size 18.4 EB
20:43:28:0252 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
20:43:28:0252 FuRemoteList         loading /usr/share/fwupd/metainfo
20:43:28:0253 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
20:43:28:0253 XbSilo               file: 24dfee83-3312-9731-252b-b00c1babf27f, current:24dfee83-3312-9731-252b-b00c1babf27f, cached: (null)
20:43:28:0253 XbSilo               loading silo with file contents
20:43:28:0254 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
20:43:28:0254 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
20:43:28:0254 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
20:43:28:0255 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
20:43:28:0256 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
20:43:28:0257 FuRemoteList         ordering lvfs-testing=lvfs+1
20:43:28:0257 FuRemoteList         ignoring unfound remote fwupd
20:43:28:0257 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
20:43:28:0257 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:43:28:0257 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
20:43:28:0257 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
20:43:28:0257 Jcat                 loaded 1 certificates
20:43:28:0257 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
20:43:28:0257 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
20:43:28:0257 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:43:28:0257 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
20:43:28:0257 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
20:43:28:0257 Jcat                 loaded 1 certificates
20:43:28:0258 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
20:43:28:0258 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
20:43:28:0264 FuEngine             client certificate exists and working
20:43:28:0264 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
20:43:28:0265 FuHistory            got schema version of 6
20:43:28:0265 FuHwids              smbios property Manufacturer=LENOVO
20:43:28:0265 FuHwids              smbios property EnclosureKind=a
20:43:28:0265 FuHwids              smbios property Family=ThinkPad X1 Carbon 7th
20:43:28:0265 FuHwids              smbios property ProductName=20QD0000US
20:43:28:0265 FuHwids              smbios property ProductSku=LENOVO_MT_20QD_BU_Think_FM_ThinkPad X1 Carbon 7th
20:43:28:0265 FuHwids              smbios property BiosVendor=LENOVO
20:43:28:0265 FuHwids              smbios property BiosVersion=N2HET60W (1.43 )
20:43:28:0265 FuHwids              smbios property BiosMajorRelease=01
20:43:28:0265 FuHwids              smbios property BiosMinorRelease=2b
20:43:28:0265 FuHwids              smbios property BaseboardManufacturer=LENOVO
20:43:28:0265 FuHwids              smbios property BaseboardProduct=20QD0000US
20:43:28:0273 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
20:43:28:0273 XbSilo               file: d949eb54-a0be-ade8-ef97-54741bc0c55e, current:d949eb54-a0be-ade8-ef97-54741bc0c55e, cached: (null)
20:43:28:0273 XbSilo               loading silo with file contents
20:43:28:0273 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
20:43:28:0274 XbSilo               file: 3efa8c38-0b7a-51b0-59bb-9e0108391fff, current:3efa8c38-0b7a-51b0-59bb-9e0108391fff, cached: (null)
20:43:28:0274 XbSilo               loading silo with file contents
20:43:28:0274 FuEngine             541 components now in silo
20:43:28:0282 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_uefi_capsule.so)
[... cut ...]
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update
  VendorId:             DMI:LENOVO
  Version:              26403
  VersionLowest:        1
  VersionFormat:        number
  VersionRaw:           0x00006723
  VersionLowestRaw:     0x00000001
  Icon:                 audio-card
  Created:              2021-03-23

20:43:28:0348 FuPlugin             destroy(uefi_capsule)
```

hughsie, 2021-03-24:
Hmm, so enumeration is working correctly. Could you do the same thing again please, but this time:

 * terminal 1: `sudo /usr/libexec/fwupd/fwupd --verbose`
 * terminal 2: `fwupdmgr get-devices`

And then ctrl-c if you get the full set, and try again a few times. If you can get the UEFI-less output then the full daemon log would be great. Thanks!

sbconslt, 2021-03-25:
Naturally now that we are looking closely at fwupd, the creature senses that we are looking and the problem refuses to present itself. I have not been able to reproduce the short set (4 devices) when doing `fwupdmgr refresh --force` while having `sudo /usr/libexec/fwupd/fwupd --verbose` open in tandem, but I will keep trying.

hughsie, 2021-03-25:
@sbconslt the other thing to try is to add the --verbose in the fwupd systemd service file, to see if it's something like resource/permissions/selinux things that are being sidestepped by running it manually on the console.

sbconslt, 2021-03-26:
It certainly seems to be the case from my repeated testing, that fwupd may exhibit the issue when running as a systemd service unit, but never exhibits the issue when run manually with `/usr/libexec/fwupd/fwupd --verbose` (or, I have not produced it that way in many tries, in any case).

hughsie, 2021-03-29:
Okay, if you can get the --verbose output (in the journal?) when you see the missing devices that would be really useful. Thanks.

hughsie, 2021-04-11:
@sbconslt any update here please?

sbconslt, 2021-04-11:
I have not reproduced the issue in normal operation of `fwupdmgr refresh` in weeks. So if you want, let's close the issue.

hughsie, 2021-04-11:
Okay, excellent. Feel free to reopen if it pops up again.

closed 2021-04-11 by hughsie