fwupd/fwupd #3055: Thunderbolt device missing from firmware list, apparently due to power management

sdh4, 2021-03-23:
Thunderbolt host controller fails to appear in `fwupdmgr get-devices`. This is on a Lenovo Thinkpad X1 Yoga 2nd Gen. 

At one point after plugging in a device the controller did briefly appear, but it disappeared before fwupd was able to write firmware to it. 

If I turn on the thunderbolt power with `boltctl power on` then fwupdmgr consistently shows the Thunderbolt controller under get-devices. 

```fwupd-1.5.7-1.fc33.x86_64
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5
daemon version:	1.5.7
```
Boltmgr from bolt-0.9.1-1.fc33.x86_64
Linux version 5.11.7-200.fc33.x86_64

Installed from dnf

Rebooting does not help. Unknown if this is a regression.

Per this page https://christian.kellner.me/2019/11/14/fwupd_bolt_force_power_thunderbolt/ fwupd is supposed to force power the controller. In my case for some reason it was failing to do so (?) but manually force powering it with boltctl was sufficient

hughsie, 2021-03-23:
We removed all the force power stuff I'm afraid: https://github.com/fwupd/fwupd/commit/6358e23490acde05f1ca65c714d141bb8f6497ae

The fix is to insert a thunderbold device into the power for the duration of the update so the controller stays awake.

sdh4, 2021-03-24:
Yes; having a device plugged in was not sufficient in my case (appeared briefly but disappeared before the FW could be updated). 
The `boltctl power on` workaround worked fine.

closed 2021-04-09 by superm1