fwupd/fwupd #3081: Touchpad is missing from `fwupdmgr get-devices --show-all-devices` output

ericonr, 2021-03-29:
**Describe the bug**

With version 1.5.5, I can see my touchpad. With version 1.5.8, I can't. Even weirder, with 1.5.5 I didn't even need `--show-all-devices` for it to appear.

**Steps to Reproduce**

Run `fwupdmgr get-devices --show-all-devices` on a Dell XPS 9570.

If I diff the output between 1.5.5 and 1.5.8, the only difference is:

```diff
--- 1.5.5.out	2021-03-29 16:04:54.324530046 -0300
+++ 1.5.8.out	2021-03-29 16:04:37.982420458 -0300
@@ -78,18 +78,6 @@
 │       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
 │       Device Flags:     • Internal device
 │     
-├─Touchpad:
-│     Device ID:          881d0bf9902910255d18c15a368c4106b26fee62
-│     Current version:    1.7.2378871
-│     Bootloader Version: 54.0.0
-│     Vendor:             Synaptics (HIDRAW:0x06CB)
-│     GUIDs:              1d6705d0-3fa6-58d2-bd0e-806ed6792c18
-│                         2aa17264-7ef0-5cc5-9368-8595974a5e31
-│                         741692a4-ae19-5ecc-9e48-e644604c2345
-│                         a90f8ef5-de8d-5810-a6c2-d0b88b1e0be8
-│     Device Flags:       • Internal device
-│                         • Updatable
-│   
 └─UHD Graphics 630 (Mobile):
       Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
       Vendor:             Intel Corporation (PCI:0x8086)
```

**Expected behavior**

I should get Touchpad information.

**fwupd version information**

1.5.8

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): from source, while updating the package for Void Linux.

**Additional questions**
- Operating system and version: Void Linux
- Have you tried rebooting? No
- Is this a regression? Seems so

If necessary, I can try to bisect between released versions and then commits... Is there an easy way to make `fwupdmgr` use the daemon from the build directory, so I can avoid having to continuously reinstall the package?

hughsie, 2021-03-29:
I think this is deliberate: https://github.com/fwupd/fwupd/commit/277347688427fd1bf68d68d8d8788bf67d42e01c

i.e. we only add devices that are explicitly in the allow list. What hardware is this out of interest?

ericonr, 2021-03-29:
Ah, that makes sense, I think. Seems I was fortunate enough to not get issues after probing the touchpad.

It's a XPS 9570, and libinput tells me it's `SYNA2393:00 06CB:7A13 Touchpad`. For Windows, `Intel-HID-Event-Filter-Driver_33CDY_WIN_2.2.1.377_A11_04.EXE` is one of the drivers listed in the Dell support page for it, but I think by default it uses the built-in stuff.

hughsie, 2021-03-29:
@superm1 is that device ever likely to get firmware updates?

superm1, 2021-03-29:
Touchpad updates are incredibly rare.  But needing to patch a quirk file and then release a CAB is going to make it even harder.  I wish there was a way that we could tell what was safe to probe and not.

<crazy thought>
What if we let there be a feedback loop like we have for adding version formats?  Let the existence of a GUID on LVFS let us _probe_ that GUID from fwupd.  The problem would of course be bugs in intermediary releases, but then how about if we looked at minimum version on the device in LVFS and only let that probe feedback loop work if you're new enough?
</crazy thought>

ericonr, 2021-03-30:
For the record, I was reporting this more to be sure that it was intended/expected than as something that stops me from updating a certain piece of hardware that I need to update. It just jumped out at me as I tested the 1.5.5 -> 1.5.8 update.

I'm glad to help figure out the decisions and test, just want to be sure you folks don't spend time on it thinking this particular case is very important and/or urgent :)

hughsie, 2021-03-30:
> I wish there was a way that we could tell what was safe to probe and not.

Well, we have the ONLY_SUPPORTED quirk which means we only add the device if the metadata exists, although the chicken-and-egg problem to your suggestion is that we don't know the current version until we've probed, and we don't know if it's safe to probe until we have the current version. Same problem when you just use the GUIDs rather than the version -- any custom GUIDs added in ->probe() are not going to be there to use with ONLY_SUPPORTED...

superm1, 2021-03-31:
> Well, we have the ONLY_SUPPORTED quirk which means we only add the device if the metadata exists, although the chicken-and-egg problem to your suggestion is that we don't know the current version [...]

yeah you're right :(

closed 2021-03-31 by superm1