fwupd/fwupd #3084: Lenovo ThinkPad P1 Gen 2 is requested to get 10% of battery for updating bios but fails to flash at 11% because of the battery level

ErwanAliasr1, 2021-03-30:
**Describe the bug**
Lenovo ThinkPad P1 Gen 2 (20QUS58100) is said to need 10% of battery at least to get the bios updated.
At 11%, the firmware refuses to flash as battery is too low

**Steps to Reproduce**
- Lenovo ThinkPad P1 Gen 2 with 1.36 bios
- Put battery at 11%
- Install the bios 1.38
- Reboot
- Fw flash fail because of not enough battery

**Expected behavior**
The fwuptool does request a minimal battery level to get the fw being installed and host rebooted but this does seems in sync with the fw expectations.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
- via apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20QUS58100
?
??Thunderbolt Controller:
?     Device ID:           61b6e470639f64427b47ae7e101a778a2ecca334
?     Summary:             Unmatched performance for high-speed I/O
?     Current version:     59.00
?     Vendor:              Lenovo (TBT:0x0109)
?     GUIDs:               02ca1877-1054-5273-9f47-586b27f9abc3 ? TBT-01091813
?                          6075eaff-25d6-5a4f-91f8-6f565a644e72 ? TBT-01091813-controller0-0
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Device stages updates
?   
??Dell Universal Dock D6000:
?     Device ID:           62d5ed7ac4bfd0f40b57140818519eb35784fb8c
?     Current version:     31.27
?     Vendor:              DisplayLink (USB:0x17E9)
?     Serial Number:       1907028815
?     GUIDs:               5405d978-fd0d-5890-be13-593f009cc292 ? USB\VID_17E9&PID_6006&REV_3127
?                          1b3b6e05-bd4d-5880-b68d-c7c00b6681f8 ? USB\VID_17E9&PID_6006
?     Device Flags:        ? Updatable
?   
??SAMSUNG MZVLB1T0HBLR-000L7:
?     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
?     Summary:             NVM Express Solid State Drive
?     Current version:     4M2QEXF7
?     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
?     Serial Number:       S4EMNF0N530981
?     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d ? NVME\VEN_144D&DEV_A808&REV_00
?                          47335265-a509-51f7-841e-1c94911af66b ? NVME\VEN_144D&DEV_A808
?                          b77af583-f45c-537c-8584-fc5d1c80fa20 ? SAMSUNG MZVLB1T0HBLR-000L7
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Needs a reboot after installation
?                          ? Device is usable for the duration of the update
?   
??System Firmware:
?     Device ID:           105dc4287111ca23352a3b4759a602c10ad8bf88
?     Current version:     0.1.36
?     Minimum Version:     0.1.0
?     Vendor:              LENOVO (DMI:LENOVO)
?     GUID:                55d04ffc-714a-4457-b982-d244343e1958
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Supported on remote server
?                          ? Needs a reboot after installation
?                          ? Cryptographic hash verification is available
?                          ? Device is usable for the duration of the update
?   
??TPM:
? ?   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
? ?   Current version:     73.64.17568.6659
? ?   Vendor:              ST Microelectronics (TPM:STM)
? ?   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6 ? system-tpm
? ?                        84df3581-f896-54d2-bd1a-372602f04c32 ? TPM\VEN_STM&DEV_0001
? ?                        bfaed10a-bbc1-525b-a329-35da2f63e918 ? TPM\VEN_STM&MOD_
? ?                        70b7b833-7e1a-550a-a291-b94a12d0f319 ? TPM\VEN_STM&DEV_0001&VER_2.0
? ?                        06f005e9-cb62-5d1a-82d9-13c534c53c48 ? TPM\VEN_STM&MOD_&VER_2.0
? ?   Device Flags:        ? Internal device
? ? 
? ??Event Log:
?       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
?       Vendor:            ST Microelectronics (TPM:STM)
?       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff ? system-tpm-eventlog
?       Device Flags:      ? Internal device
?     
??TU117GLM [Quadro T1000 Mobile]:
?     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
?     Current version:     a1
?     Vendor:              NVIDIA Corporation (PCI:0x10DE)
?     GUIDs:               f71f56b0-8808-5f2c-99ae-afc499f476c3 ? PCI\VEN_10DE&DEV_1FB9&REV_A1
?                          aea1c72d-ddad-5702-9aaf-2b9bf2442a80 ? PCI\VEN_10DE&DEV_1FB9
?     Device Flags:        ? Internal device
?                          ? Cryptographic hash verification is available
?   
??UEFI Device Firmware:
?     Device ID:           b649808587169ddcf6996a64fe99777e0b97c4d4
?     Current version:     0.0.0
?     Vendor:              DMI:LENOVO
?     GUID:                466d3054-4e25-483c-96f0-d6c35e765042
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Supported on remote server
?                          ? Needs a reboot after installation
?                          ? Device is usable for the duration of the update
?   
??UEFI Device Firmware:
?     Device ID:           3a7d0b9b370f21148edec601cee9ee5864d29fb9
?     Current version:     0.1.23
?     Minimum Version:     0.1.23
?     Vendor:              DMI:LENOVO
?     GUID:                5aaaa50f-8348-43aa-befe-7fc4b9838ab5
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Supported on remote server
?                          ? Needs a reboot after installation
?                          ? Device is usable for the duration of the update
?   
??UEFI Device Firmware:
?     Device ID:           95426c1ada8b5820052af61f71e12134286adffb
?     Current version:     1.1.20484
?     Minimum Version:     0.0.1
?     Vendor:              DMI:LENOVO
?     GUID:                7ab93135-c544-4c88-ba2e-08ec6adb9e6f
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Needs a reboot after installation
?                          ? Device is usable for the duration of the update
?   
??UEFI Device Firmware:
?     Device ID:           dbb2bdd29afc1ffb5b3eede855d6a0c3985dde63
?     Current version:     0.73.64
?     Minimum Version:     0.73.64
?     Vendor:              DMI:LENOVO
?     GUID:                85a03cdf-539f-448b-80be-df1cf3438963
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Needs a reboot after installation
?                          ? Device is usable for the duration of the update
?   
??UHD Graphics 630 (Mobile):
      Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
      Vendor:              Intel Corporation (PCI:0x8086)
      GUIDs:               12c60b98-52cd-500a-b99e-087aa6b47f0b ? PCI\VEN_8086&DEV_3E9B&REV_00
                           db677673-1102-5481-a11e-6f397e123ced ? PCI\VEN_8086&DEV_3E9B
      Device Flags:        ? Internal device
                           ? Cryptographic hash verification is available

```

**Additional questions**
- Operating system and version: ubuntu 20
- Have you tried rebooting? yes
- Is this a regression? Maybe

hughsie, 2021-03-30:
@mrhpearson do you know what the policy is for Lenovo? @superm1 same question for Dell. I suspect we want to make the theshold something like 50% for system firmware updates (e.g. BIOS) and something like 10% for peripherals, but I'm open to ideas.

superm1, 2021-03-30:
@hughsie see commit https://github.com/fwupd/fwupd/commit/e396f62d74b6cdb0000660b72fba7510a93fe627 for the BIOS details (IE the default value used by fwupd matches it right now).   I recall we used to have a bigger value like 30 or 50% but there was complaints that internal test cases against the expected behavioral spec failed.

Since across OEMs there is no common value, I think what we're going to probably end up with is some special cases in code (IE Dell it's 10%, Lenovo it's X% and if not those it's 50%).

The main reason for a value in the first place is that if power is unplugged the update doesn't get interrupted by dying quickly.  I don't see a problem with leaving it 10% for peripherals.

hughsie, 2021-03-30:
> I think what we're going to probably end up with is some special cases in code/quirks

This is something I pondered with @hardboprobot for the battery_level stuff. I'll do a PR and see what it looks like.

hughsie, 2021-03-30:
https://github.com/fwupd/fwupd/pull/3086 for the peripheral case, not going to work in the upower case.

ErwanAliasr1, 2021-03-31:
Shouldn't the vendor be able to override the default value when submitting a firmware?

Please note the upgrade succeed when I retried at 90% of battery

mrhpearson, 2021-03-31:
I got some feedback from the firmware team.
In summary: Battery >= 25% with an exception for Battery FW where the AC has to be connected (I don't think we have any batter FW on LVFS yet....but for the future)

As a note - there are Microsoft guidelines (https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/seamless-crisis-prevention-and-recovery). Potentially this makes 25% a good default where there isn't something explicit? It's what our firmware team is based from and I'm sure we're not alone.

hughsie, 2021-03-31:
> Shouldn't the vendor be able to override the default value when submitting a firmware?

I personally think that's too much control. I think a healthy level of configurability would be for the plugin author to define the appropriate level. I'm still pondering something better.

closed 2021-04-07 by hughsie