fwupd/fwupd #3094: Remove fu_plugin_request_recoldplug()?

hughsie, 2021-04-01:
Nothing seems to use `fu_plugin_request_recoldplug()` but its plumbed in pretty deep into fwupd. Removing the functionality would remove a lot of dead and complex code. @superm1 do you think this something we want to keep around for the future, or can it be nuked?

hughsie, 2021-04-01:
Same question for fu_context_set_coldplug_delay() I guess too.

superm1, 2021-04-01:
The main reason we added them initially was to support `dell` plugin calling force power on some devices through BIOS interface.  We also did it for thunderbolt force power.  Those are both removed for `$reasons` now.

I went back and forth whether it makes sense to remove, and I think i'm in agreement.  We want all devices to come in via udev/usb/BT backends.  We should have events for all of them.

hughsie, 2021-04-01:
I agree the recoldplug stuff was super fragile. Removed in https://github.com/fwupd/fwupd/pull/3095

closed 2021-04-01 by hughsie