fwupd/fwupd #3099: pixart_rf plugin: Error message (not critical) when fwupdtool is run on Chromebook with an Elan Touchpad

bleungatchromium, 2021-04-02:
**Describe the bug**
A clear and concise description of what the bug is.
When running fwupdtool limited to just the pixart_rf plugin, an error message is observed, apparently when the tool is looking at other devices on the system.

```
FuDevice             failed to set quirk key GType=FuElantpHidDevice: device GType FuElantpHidDevice not supported                                        
```

**Steps to Reproduce**
Steps to reproduce the behavior.
```
fwupdtool --plugins=pixart_rf get-devices
```

**Expected behavior**
A clear and concise description of what you expected to happen.
No error message.

**fwupd version information**
Please provide the version of the daemon and client.
```
# fwupdtool --version
client version: 1.5.8
compile-time dependency versions
        gusb:   0.3.5

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed with Chrome OS, from Chrome OS's ebuild, pegged to fwupd 1.5.8.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version: Google Chrome OS, R91-13881.0.0 (Official Build) dev-channel volteer test
- Have you tried rebooting? Doesn't help.
- Is this a regression? No.

bleungatchromium, 2021-04-02:
Chromebook in question is an in-development model, and it has two i2c based elan touch devices.

One i2c elan touchpad, one i2c touchscreen (uses hid-multitouch).

```
[    2.164708] elan_i2c i2c-ELAN0000:00: Elan Touchpad: Module ID: 0x00bf, Firmware: 0x0001, Sample: 0x0001, IAP: 0x0000
[    2.164769] input: Elan Touchpad as /devices/pci0000:00/0000:00:19.1/i2c_designware.5/i2c-19/i2c-ELAN0000:00/input/input5
[    2.221232] input: ELAN90FC:00 04F3:2A49 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-15/i2c-ELAN90FC:00/0018:04F3:2A49.0001/input/input6
[    2.221286] input: ELAN90FC:00 04F3:2A49 UNKNOWN as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-15/i2c-ELAN90FC:00/0018:04F3:2A49.0001/input/input7
[    2.221306] input: ELAN90FC:00 04F3:2A49 UNKNOWN as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-15/i2c-ELAN90FC:00/0018:04F3:2A49.0001/input/input8
[    2.221327] input: ELAN90FC:00 04F3:2A49 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-15/i2c-ELAN90FC:00/0018:04F3:2A49.0001/input/input9
[    2.221349] input: ELAN90FC:00 04F3:2A49 Stylus as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-15/i2c-ELAN90FC:00/0018:04F3:2A49.0001/input/input10
[    2.221374] hid-multitouch 0018:04F3:2A49.0001: input,hidraw0: I2C HID v1.00 Device [ELAN90FC:00 04F3:2A49] on i2c-ELAN90FC:00
```

hughsie, 2021-04-02:
The warning is correct, but in this case, invalid. In case you were wondering:

	#2  0x00007ffff7f29abf in fu_device_set_quirk_kv (self=0x538480, key=0x7fffe84c9035 "GType", value=0x7fffe84cef32 "FuElantpHidDevice", error=0x7fffffffcf98) at ../libfwupdplugin/fu-device.c:1260
	#3  0x00007ffff7f29c65 in fu_device_quirks_iter_cb (ctx=0x496500, key=0x7fffe84c9035 "GType", value=0x7fffe84cef32 "FuElantpHidDevice", user_data=0x538480) at ../libfwupdplugin/fu-device.c:1308
	#4  0x00007ffff7f43c10 in fu_quirks_lookup_by_id_iter (self=0x496500, group=0x518dc0 "1f96f71f-0d28-5450-8d7e-045223496f4d", iter_cb=0x7ffff7f29c0e <fu_device_quirks_iter_cb>, user_data=0x538480)
	    at ../libfwupdplugin/fu-quirks.c:441
	#5  0x00007ffff7f25cb9 in fu_context_lookup_quirk_by_id_iter (self=0x483d60, guid=0x518dc0 "1f96f71f-0d28-5450-8d7e-045223496f4d", iter_cb=0x7ffff7f29c0e <fu_device_quirks_iter_cb>, user_data=0x538480)
	    at ../libfwupdplugin/fu-context.c:473
	#6  0x00007ffff7f29d35 in fu_device_add_guid_quirks (self=0x538480, guid=0x518dc0 "1f96f71f-0d28-5450-8d7e-045223496f4d") at ../libfwupdplugin/fu-device.c:1322
	#7  0x00007ffff7f2a14c in fu_device_add_instance_id_full (self=0x538480, instance_id=0x523e20 "HIDRAW\\VEN_04F3", flags=FU_DEVICE_INSTANCE_FLAG_ONLY_QUIRKS) at ../libfwupdplugin/fu-device.c:1478
	#8  0x00007ffff7f51309 in fu_usb_device_probe (device=0x538480, error=0x7fffffffd2b0) at ../libfwupdplugin/fu-usb-device.c:328
	#9  0x00007ffff7f2e94a in fu_device_probe (self=0x538480, error=0x7fffffffd2b0) at ../libfwupdplugin/fu-device.c:3424
	#10 0x000000000043d547 in fu_engine_backend_device_added_cb (backend=0x4966a0, device=0x538480, self=0x49c040) at ../src/fu-engine.c:5935
	#11 0x00007ffff7cc7c2f in g_closure_invoke () at /lib64/libgobject-2.0.so.0
	#12 0x00007ffff7ce3ea6 in signal_emit_unlocked_R () at /lib64/libgobject-2.0.so.0
	#13 0x00007ffff7ce576a in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
	#14 0x00007ffff7ce5983 in g_signal_emit () at /lib64/libgobject-2.0.so.0
	#15 0x0000000000446996 in fu_backend_device_added (self=0x4966a0, device=0x538480) at ../src/fu-backend.c:46
	#16 0x000000000044abe7 in fu_usb_backend_device_added_cb (ctx=0x4d86f0, usb_device=0x506040, backend=0x4966a0) at ../src/fu-usb-backend.c:31
	#17 0x00007ffff7cc7c2f in g_closure_invoke () at /lib64/libgobject-2.0.so.0
	#18 0x00007ffff7ce3ea6 in signal_emit_unlocked_R () at /lib64/libgobject-2.0.so.0
	#19 0x00007ffff7ce576a in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
	#20 0x00007ffff7ce5983 in g_signal_emit () at /lib64/libgobject-2.0.so.0
	#21 0x00007ffff7a4cd5b in g_usb_context_add_device () at /lib64/libgusb.so.2
	#22 0x00007ffff7a4cf3b in g_usb_context_rescan () at /lib64/libgusb.so.2
	#23 0x00007ffff7a4d048 in g_usb_context_enumerate () at /lib64/libgusb.so.2
	#24 0x000000000044ad69 in fu_usb_backend_coldplug (backend=0x4966a0, error=0x7fffffffdd90) at ../src/fu-usb-backend.c:73
	#25 0x0000000000446c53 in fu_backend_coldplug (self=0x4966a0, error=0x7fffffffdd90) at ../src/fu-backend.c:93
	#26 0x000000000043ed91 in fu_engine_load (self=0x49c040, flags=14, error=0x7fffffffdf78) at ../src/fu-engine.c:6394
	#27 0x0000000000418acf in fu_util_start_engine (priv=0x48d930, flags=14, error=0x7fffffffdf78) at ../src/fu-tool.c:179
	#28 0x0000000000419f8a in fu_util_get_devices (priv=0x48d930, values=0x478460, error=0x7fffffffdf78) at ../src/fu-tool.c:668
	#29 0x0000000000423b5e in fu_util_cmd_array_run (array=0x48da00, priv=0x48d930, command=0x7fffffffe71d "get-devices", values=0x7fffffffe488, error=0x7fffffffdf78) at ../src/fu-util-common.c:551
	#30 0x000000000042131c in main (argc=2, argv=0x7fffffffe478) at ../src/fu-tool.c:3255

so we enumerate all USB devices, find the elantp device, match the VID quirk entry, get the GType, fail to find the gtype (as elantp is not enabled, and so didn't register it) and then we show the warning.

bleungatchromium, 2021-04-02:
Ah that makes sense.

In fact, the warning goes away completely if I add elantp to the list of plugins.
fwupdtool --plugins=elantp,pixart_rf get-devices

Thanks for the explanation!

Benson

hughsie, 2021-04-02:
Ahh, I still think we can handle the error better. I have a few ideas.

closed 2021-04-02 by hughsie

bleungatchromium, 2021-04-02:
Thanks, @hughsie !