fwupd/fwupd #3104: "unsupported" on Microsoft Surface Pro 3

sphh, 2021-04-04:
**Describe the bug**
Installing a UEFI update for a Microsoft Surface Pro 3 tablet/laptop works well, but during a reboot I get the error message:
```
Found update fwupd-ux-capsule
Found update fwupd-5a2d987b-cb39-42fe-a4cf-d5d0abae3a08-0
Adding new capsule
Adding new capsule
WARNING: Could not apply capsules: Unsupported
start_image() returned unsupported
```
as reported by another user in https://github.com/linux-surface/surface-uefi-firmware/issues/2.

**Steps to Reproduce**
1. Prepare the firmware updates according to https://github.com/linux-surface/surface-uefi-firmware for the downloaded update file `SurfacePro3_Win10_18362_1902002_0.msi` gives me these four firmware file:
```
SurfacePro3_EC_38.16.85.0.cab
SurfacePro3_TOUCH_426.27.66.0.cab
SurfacePro3_SAM_3.9.850.0.cab
SurfacePro3_UEFI_3.11.2650.0.cab
```
2. Install the UEFI firmware gives no warning/error:
```
fwupdmgr install SurfacePro3_UEFI_3.11.2650.0.cab
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on System Firmware…                                   ]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: 
```
3. Reboot.
4. Get the error message mentioned above, but boot resumes to Linux.

**Expected behavior**
Install the UEFI firmware update.

On a side note, the procedure outlined above worked for a Surface Pro 5 (2017).

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed with the default package manager (apt, synaptic).

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Surface Pro 3
│
├─Haswell-ULT Integrated Graphics Controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     09
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               61820cd9-3622-5ddc-9ba4-c1308215d704
│                          84a13add-7bfc-54a2-bec5-3461340c6773
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─SAMSUNG MZMTE256HMHP-000MV:
[... 374 lines cut ...]
│       ├──fs.mod
│       └──std.mod
├── initrd.img -> initrd.img-5.4.0-70-generic
├── initrd.img-5.11.10-surface
├── initrd.img-5.11.7-surface
├── initrd.img-5.4.0-70-generic
├── initrd.img.old -> initrd.img-5.11.7-surface
├── lost+found [error opening dir]
├── System.map-5.11.10-surface
├── System.map-5.11.7-surface
├── System.map-5.4.0-70-generic
├── vmlinuz -> vmlinuz-5.4.0-70-generic
├── vmlinuz-5.11.10-surface
├── vmlinuz-5.11.7-surface
├── vmlinuz-5.4.0-70-generic
└── vmlinuz.old -> vmlinuz-5.11.7-surface
```

**Additional questions**
- Operating system and version: Linux Mint 20.1
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? `ls /dev/nvme*` gave `No such file or directory`, so I assume: No
- Is secure boot enabled? Tried both (enabled and disabled), got the same error message.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

superm1, 2021-04-04:
This seems to be a firmware bug to me, I don't believe there is anything fwupd can do about it.

superm1, 2021-04-04:
At that point fwupd has tried to call the BIOS UpdateCapsule routine and got that return code.

hughsie, 2021-04-04:
Specifically, `RT->UpdateCapsule` is returning `Unsupported`

sphh, 2021-04-04:
I just tried with an older firmware (`SurfacePro3_UEFI_RS3RS4RS5__3.11.2550.0` and I got the same result. The next one without a funny name is `SurfacePro3_UEFI_3.11.2150.0`, but I cannot try that out, because this FW is already installed (from an update from Windows before I kicked that OS out).

I just double checked the firmware file `.cap` sizes from the downloaded update and that one in the generated .cab file. They are exactly the same. I also checked the md5sums for those two files. They are the same, so I believe we can assume that the firmware itself was not tampered with.

The only difference I noticed is, that the `firmware.inf` file is different in the generated `.cab` file and in the downloaded updated file. Looking into the conversion skript, I found, that the `.inf` file is converted with the help of `dos2unix` and when I do it, I get the following output:
```
dos2unix: converting UTF-16LE file UEFI.inf to UTF-8 Unix format...
```
so it looks ok to me.

If that is a firmware bug, than the update would not be installable from Windows. Is that correct?

Any ideas, what I could try?

hughsie, 2021-04-05:
On Sun, 4 Apr 2021 at 23:20, sphh ***@***.***> wrote:
> I just tried with an older firmware (SurfacePro3_UEFI_RS3RS4RS5__3.11.2550.0 and I got the same result.

The error is coming from the "old" firmware release, rather than the
one you're trying to install. Unfortunately I think the only way you
can update the system bios is to install windows, install the latest
firmware using the low-level vendor tool, then reinstall Linux and see
if the latest firmware is able to do capsule updates. Sorry that's not
more helpful.

Richard.

superm1, 2021-04-05:
> Any ideas, what I could try?

I glanced through commits and didn't notice anything notable that should change things, but you can certainly try to use a current version of fwupd instead of 1.3.11.  1.5.8 is the most recently released version.  You can experiment pretty easily using a [snap package](https://github.com/fwupd/fwupd/wiki/fwupd-snap).

But this really looks like a firmware bug to me.  I would expect that Windows Update fails just the same too.

closed 2021-04-06 by superm1