fwupd/fwupd #3122: Can't update the DBX with fwupdmgr -v update (dual boot)

fredoche, 2021-04-08:
**Describe the bug**
Can't update the DBX with fwupdmgr -v update
Note that I have a dual boot, with maybe two ESP.

**Steps to Reproduce**
fwupdmgr -v update

**Expected behavior**
should update the dbx

**fwupd version information**
Please provide the version of the daemon and client.
```shell
fwupdmgr --version
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.8

fwupdate -l
failed: More than one available ESP

sudo /usr/bin/fwupdtool esp-list --verbose
19:59:04:0196 FuDebug              Verbose debugging enabled (on console 1)
19:59:04:0209 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda3, type: e3c9e316-0b5c-4db8-817d-f92df00215ae, internal: 1, fs: 
19:59:04:0211 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda5, type: de94bba4-06d1-4d40-a16a-bfd50179d6ac, internal: 1, fs: ntfs
19:59:04:0213 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
19:59:04:0215 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
19:59:04:0217 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
19:59:04:0218 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda4, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7, internal: 1, fs: ntfs
19:59:04:0220 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda1, type: de94bba4-06d1-4d40-a16a-bfd50179d6ac, internal: 1, fs: ntfs
19:59:04:0221 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: e6d6d379-f507-44c2-a23c-238f2a3df928, internal: 1, fs: LVM2_member
Choisir un volume :
0.	Annuler
1.	/org/freedesktop/UDisks2/block_devices/sda2
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
System Product Name
│
├─Ellesmere [Radeon RX 470/480/570/570X/580/580X/590]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Version actuelle:   e7
│     Fournisseur:        Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              58371b3f-4af4-5403-9ab4-971a697ad503 ← PCI\VEN_1002&DEV_67DF&SUBSYS_14623417&REV_E7
│                         0bf43304-48c4-5c01-ba09-c8f46f54c0be ← PCI\VEN_1002&DEV_67DF&SUBSYS_14623417
│                         68958f93-be94-5a98-a06d-a4da21a38135 ← PCI\VEN_1002&DEV_67DF&REV_E7
│                         55746bd7-4196-594a-9d39-0808222765d1 ← PCI\VEN_1002&DEV_67DF
│     Device Flags:       • Périphérique interne
│                         • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i7-6700K CPU @ 4.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Version actuelle:   0x000000e2
│     Fournisseur:        Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         215d01de-fd1a-57cf-9153-72e29c0b7ffe ← CPUID\PRO_0&FAM_06&MOD_5E
│                         c944e5cd-430c-5731-a9d5-c1b63c61cf27 ← CPUID\PRO_0&FAM_06&MOD_5E&STP_3
[... 221 lines cut ...]
├── System.map-5.11.10-300.fc34.x86_64
├── System.map-5.11.11-300.fc34.x86_64
├── System.map-5.11.9-300.fc34.x86_64
├── vmlinuz-0-rescue-3c3f2bd1b6994d4cb45d2a2c0229bcc9
├── vmlinuz-5.11.10-300.fc34.x86_64
├── vmlinuz-5.11.11-300.fc34.x86_64
└── vmlinuz-5.11.9-300.fc34.x86_64

16 directories, 105 files

```

**Additional questions**
- Operating system and version:
- fedora 34 (beta)
- Have you tried rebooting?
- yes
- Is this a regression?
- no
- Are you using an NVMe disk?
- yes, and a sata disk as well (dual boot with apparently two EFI partitions)
- Is secure boot enabled?
- yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
- not a Lenovo (desktop PC with Asus motherboard)

superm1, 2021-04-09:
> Remote peer disconnected

This sounds like the daemon crashed or timed out.  Do you have a crash report or core dump by chance?   Can you report the ABRT into Fedora?  @hughsie can see it then from there.

fredoche, 2021-04-09:
I don't have such a crash dump in ABRT. I dont know how to make one.

hughsie, 2021-04-09:
> WARNING: UEFI ESP partition not detected or configured

I think this is the root of the problem. Can you try:

    sudo /usr/libexec/fwupd/fwupd --verbose

and then in another terminal do `fwupdmgr update` -- and then paste the full daemon log here. Thanks!

superm1, 2021-04-09:
But esp isn't used for dbx update, why should it actually matter?

hughsie, 2021-04-09:
Hmm, indeed. The other thing that seemed odd was the `Vérification` line; we don't have any kind of verification for DBX update...

fredoche, 2021-04-09:
Before update:
```
shell
sudo /usr/libexec/fwupd/fwupd --verbose
15:52:52:0359 FuDebug              Verbose debugging enabled (on console 1)
15:52:52:0359 FuConfig             loading config values from /etc/fwupd/daemon.conf
15:52:52:0359 FuConfig             using autodetected max archive size 18,4 Eo
15:52:52:0359 FuRemoteList         loading /usr/share/fwupd/metainfo
15:52:52:0359 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:52:52:0359 XbSilo               file: 2780f553-0733-789a-4738-89f6fe556548, current:2780f553-0733-789a-4738-89f6fe556548, cached: (null)
15:52:52:0359 XbSilo               loading silo with file contents
15:52:52:0359 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dbx.conf
15:52:52:0359 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:52:52:0360 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
15:52:52:0360 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
15:52:52:0360 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
15:52:52:0360 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:52:52:0360 FuRemoteList         ordering dbx=lvfs+1
15:52:52:0360 FuRemoteList         ignoring unfound remote fwupd
15:52:52:0360 FuRemoteList         ordering lvfs-testing=lvfs+1
15:52:52:0360 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:52:52:0360 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
15:52:52:0360 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:52:52:0360 Jcat                 loaded 1 certificates
15:52:52:0360 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:52:52:0360 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:52:52:0360 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
15:52:52:0360 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:52:52:0360 Jcat                 loaded 1 certificates
15:52:52:0360 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:52:52:0360 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
15:52:52:0360 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:52:52:0365 FuEngine             client certificate exists and working
15:52:52:0365 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:52:52:0365 FuHistory            got schema version of 6
15:52:52:0365 FuIdle               setting timeout to 7200s
15:52:52:0365 FuHwids              smbios property Manufacturer=System manufacturer
15:52:52:0365 FuHwids              smbios property EnclosureKind=3
15:52:52:0365 FuHwids              smbios property Family=To be filled by O.E.M.
15:52:52:0365 FuHwids              smbios property ProductName=System Product Name
15:52:52:0365 FuHwids              smbios property ProductSku=SKU
15:52:52:0365 FuHwids              smbios property BiosVendor=American Megatrends Inc.
15:52:52:0365 FuHwids              smbios property BiosVersion=2606
15:52:52:0365 FuHwids              smbios property BiosMajorRelease=05
15:52:52:0365 FuHwids              smbios property BiosMinorRelease=0b
15:52:52:0365 FuHwids              smbios property BaseboardManufacturer=ASUSTeK COMPUTER INC.
15:52:52:0365 FuHwids              smbios property BaseboardProduct=B150I PRO GAMING/AURA
15:52:52:0365 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
15:52:52:0365 XbSilo               file: 5e05b07d-00bd-8837-ac73-b10c0f4a8023, current:5e05b07d-00bd-8837-ac73-b10c0f4a8023, cached: (null)
15:52:52:0365 XbSilo               loading silo with file contents
15:52:52:0366 FuEngine             556 components now in silo
15:52:52:0373 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_flashrom.so)
15:52:52:0373 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_logind.so)
15:52:52:0375 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_uefi_capsule.so)
15:52:52:0375 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_synaptics_cxaudio.so)
15:52:52:0375 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_pixart_rf.so)
15:52:52:0375 FuPlugin             added udev subsystem watch of hidraw
15:52:52:0376 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_linux_lockdown.so)
15:52:52:0376 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_upower.so)
15:52:52:0376 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_linux_sleep.so)
15:52:52:0376 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_solokey.so)
15:52:52:0377 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_elantp.so)
15:52:52:0377 FuPlugin             added udev subsystem watch of i2c-dev
15:52:52:0377 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_vli.so)
15:52:52:0377 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_wacom_usb.so)
15:52:52:0378 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_thunderbolt.so)
15:52:52:0378 FuPlugin             added udev subsystem watch of thunderbolt
[... cut ...]
15:54:11:0425 FuPluginUefiDbx      SHA256 was 99b89b31b77bace5827e170a2573dc00f4b83c43a86b67de236d02f6ef815d59
15:54:11:0425 FuPluginUefiDbx      fn=/run/media/root/D67F-86C7/EFI/Microsoft/Boot/bootmgr.efi, checksum=99b89b31b77bace5827e170a2573dc00f4b83c43a86b67de236d02f6ef815d59
15:54:11:0425 FuPluginUefiDbx      getting Authenticode hash of /run/media/root/D67F-86C7/EFI/Microsoft/Boot/BCD.LOG1
15:54:11:0425 FuCommon             fu_common_read_uint16_safe: assertion 'buf != NULL' failed
Erreur de segmentation
```
Erreur de segmentation = segfault

hughsie, 2021-04-09:
Can you upload /run/media/root/D67F-86C7/EFI/Microsoft/Boot/BCD.LOG1 here please.

hughsie, 2021-04-09:
I'm guessing it's an empty file. TIL: from `g_bytes_get_data`:

    NULL may be returned if size is 0. This is not guaranteed, as the GBytes may represent an empty string with data non-NULL and size as 0. NULL will not be returned if size is non-zero.

I think we probably want to audit every place we do `g_bytes_get_data()` and check for NULL. I suspect we're only hitting this for this now because `bytes` was constructed using g_mapped_file_get_bytes() i.e. was mmapped.

hughsie, 2021-04-09:
I wonder if we want a `fu_bytes_get_data_safe()` which takes a `GError`...

superm1, 2021-04-09:
> I wonder if we want a `fu_bytes_get_data_safe()` which takes a `GError`...

sounds like a good idea to me.

hughsie, 2021-04-09:
@fredoche the branch wip/hughsie/fu_bytes_get_data_safe should fix your problem in the short term, and I'll fix that branch up tomorrow to be the more complete fix.

fredoche, 2021-04-09:
The file is empty.

superm1, 2021-04-09:
I think in the short term the file can probably be deleted to fix it too.

fredoche, 2021-04-09:
Tried deleting it, it didnt work.

fredoche, 2021-04-09:
```
shell
***@***.*** ~]$ sudo rm
/run/media/root/D67F-86C7/EFI/Microsoft/Boot/BCD.LOG1
[sudo] Mot de passe de fred :
***@***.*** ~]$ fwupdmgr update --
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more
information.
Devices with no available firmware updates:
 • Samsung SSD 960 EVO 500GB
 • SanDisk SD8SB8U256G1122
Upgrade available for UEFI dbx from 83 to 190
UEFI dbx and all connected devices may not be usable while updating.
Continue with update? [Y|n]: Y
Téléchargement…          [***************************************]
Décompression…           [***************************************]
Authentification…        [***************************************]
Authentification…        [***************************************]
Mise à jour de UEFI dbx… [***************************************]
Vérification…            [***************************************]Remote
peer disconnected
***@***.*** ~]$ sudo rm
/run/media/root/D67F-86C7/EFI/Microsoft/Boot/BCD.LOG1
rm: impossible de supprimer
'/run/media/root/D67F-86C7/EFI/Microsoft/Boot/BCD.LOG1': Aucun fichier ou
dossier de ce type
```

superm1, 2021-04-09:
Impossible to delete?  Maybe your ESP is mounted read only?

fredoche, 2021-04-09:
No, the file simply still does not exist. the second `rm`  is there to confirm that the file hasnt reappeared, since afaic I dont know which process creates it.

closed 2021-04-10 by hughsie

fredoche, 2021-04-11:
Thanks. Was it in any case related to the dual boot nature of my setup? Cheers.