fwupd/fwupd #3128: RFE: port TLS code from gnutls to openss

kloczek, 2021-04-10:
At the moment only handfull packages are still using gnutls and most of them openssl.
It would be good to have possiblity to choose TLS library backend on configure source tree.

hughsie, 2021-04-10:
Patches welcome!

hughsie, 2021-04-11:
Adding to this, now I'm not on mobile. Dropping the GnuTLS dep is a lot of work, and the OpenSSL API is by far inferior to the former. At least on my machine, libgnutls.so.30 is being used by dozens of things, including systemd. It's not something that's trivial, or that I want to spent time on. Sorry. If you want to do this work, please feel free to work on a pull request that passes the CI tests.

closed 2021-04-11 by hughsie

kloczek, 2021-04-11:
Even ssytemd has possibility to choose GnuTLS and openssl.
Threre are way more projects which supports only openssl than GnuTLS.
There are plenty of examples which provides how to implemet such possiblity.
In my distriution where I'm headimg to only openssl I have less than 10 packages which requires GnuTLS. One of those is fwupd.

FFY00, 2021-04-12:
Again, this is an open source project, and people's time isn't free. The software is provided as-is with no implied warranty, apart from the best-effort approach from maintainers. This feature isn't being denied, the maintainers are just saying there's no manpower to drive it, and if you want to provide that, they would be willing to review and merge the change.