fwupd/fwupd #3133: can't verify firmware on dell xps 7390

alexwms, 2021-04-11:
**Describe the bug**
I just updated the firmware on my Dell XPS 13 7390 developer edition laptop. Prior to the firmware update, I did a fresh install of Ubuntu 20.04 and installed all available os/package updates via `apt`. To install the firmware updates, I issued the commands `fwupdmgr refresh`, `fwupdmgr get-updates`, and `fwupdmgr update`. Now I want to verify the update. Here is the output that I get when I attempt to do this for the `System Firmware`:

```
$ fwupdmgr verify b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2 
failed to verify System Firmware: No device checksums for 1.7.0
```

This indicates that checksums are not provided for this firmware. But the output of `fwupdmgr get-devices` below displays `Cryptographic hash verification is available` for the `System Firmware`. 

Note: After I first installed the firmware updates, my laptop's boot time increased significantly. I issued `fwupdmgr reinstall b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2` to try to address this issue (it seems to have fixed it) . This is reflected in the output of `fwupdmgr get-history` (see below). The `System Firmware` version that was installed prior to `1.7.0` was `1.6.0` I believe.

**Steps to Reproduce**
See the above section.
 

**Expected behavior**
I expected to be able to verify the firmware update.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): I didn't, ubuntu has `fwupdate` and `fwupdmgr` installed by default.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices

XPS 13 7390
│
├─Thunderbolt Controller:
│     Device ID:           d65175765aec15ff60f3dd02148b5bd2b26b1478
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     41.02
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               7d538854-204d-51b2-8f9d-1fe881c70200
│                          8a1cafa7-2d8b-5641-ba4a-140a6497a5e9
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Display controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     04
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               f9d817fa-d9b4-5956-b0f2-9d351faaf9dd
│                          e5b3b407-04cc-518e-9230-71b318fd1bd8
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─KXG60ZNV1T02 NVMe TOSHIBA 1024GB:
│     Device ID:           c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│     Summary:             NVM Express Solid State Drive
│     Current version:     10604107
│     Vendor:              Toshiba Corporation (NVME:0x1179)
│     GUIDs:               8a412b95-5c07-5ce4-b63a-f9cbf254dd5d
│                          344c2804-4948-e811-842f-0ed5f89f718b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─ST2000LM007-1R8174:
│     Device ID:           4c623d2b0128292492d01b282d9171f50d0e1c9a
│     Summary:             ATA Drive
│     Current version:     SBK2
│     GUIDs:               da57fd23-15c1-5fe7-a7c5-6c142aed1b5e
│                          a3cbe2af-31fd-5848-a7f9-44a95fa5f44d
│                          0f5e4f1e-1732-52a1-88d9-118952f0ffb3
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│   
├─System Firmware:
│     Device ID:           b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
│     Current version:     1.7.0
│     Minimum Version:     1.7.0
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     GUIDs:               cbe49cca-492b-93e8-b0f4-f693501f271b
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a
│                          40d34c18-f6d5-54db-b81d-0aaa600c0540
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─Touchpad:
      Device ID:           b26933c085b020ecf84c490812458523aee710ac
      Current version:     1.5.2767034
      Bootloader Version:  54.0
      Vendor:              Synaptics (HIDRAW:0x06CB)
      GUIDs:               f4384034-9243-5334-8075-a534be913e46
                           424bd00e-9789-5cdf-a12a-3c81bc4676d6
                           140f4458-951b-5bb9-85e2-879bd5b02615
                           b29d3c85-cd0e-503e-9c7e-f6731c1eaf2d
      Device Flags:        • Internal device
                           • Updatable

```
Also see:
```
$ fwupdmgr get-history
XPS 13 7390
│
├─KXG60ZNV1T02 NVMe TOSHIBA 1024GB:
│ │   Device ID:           c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│ │   Previous version:    10604105
│ │   Update State:        success
│ │   Last modified:       2021-04-10 20:56
│ │   GUID:                8a412b95-5c07-5ce4-b63a-f9cbf254dd5d
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─XG6 NVMe Device Update:
│       New version:       10604107
│       Remote ID:         lvfs
│       Summary:           XG6 NVMe SSD firmware for Dell PC
│       License:           Proprietary
│       Size:              1.8 MB
│       Vendor:            KIOXIA Corporation
│       Description:       Optimize SMART thermal temperature implementation.
│     
└─System Firmware:
  │   Device ID:           b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
  │   Previous version:    1.7.0
  │   Update State:        success
  │   Last modified:       2021-04-10 23:48
  │   GUID:                cbe49cca-492b-93e8-b0f4-f693501f271b
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │ 
  └─XPS 13 7390 System Update:
        New version:       1.7.0
        Remote ID:         lvfs
        Summary:           Firmware for the Dell XPS 13 7390
        License:           Proprietary
        Size:              12.3 MB
        Vendor:            Dell Inc.
        Description:       This stable release fixes the following issues:
        
         • Firmware updates to address the Intel Technical Advisory.
         • Fixed the issue where the Data wipe option does not show in the BIOS setup after switching between RAID and AHCI mode.
         • Fixed the issue where the external monitor does not power on it is restarted.
         • Fixed the issue where the Linux Vendor Firmware Service (LVFS) does not work when Secure Boot is enabled in the BIOS setup menu.
        
        Some new functionality has also been added:
        
         • Updated the Embedded Controller Engine firmware.
         • Updated the system Power Delivery firmware.
         • Updated Enhanced Pre-Boot System Assessment (ePSA) firmware.
         • Reduced the system fan noise.  
```
**Additional questions**
- Operating system and version: `Ubuntu 20.04.2`
- Have you tried rebooting? Yes. 
- Is this a regression? I am not sure what this is asking.

alexwms, 2021-04-11:
I have an additional question: According to the output of `fwupdmgr get-history`, the firmware for `KXG60ZNV1T02 NVMe TOSHIBA 1024GB` was also updated. It seems that the checksum for this update is also not available. I guess this is simply just because dell did not provide a checksum for the firmware when they uploaded it to LVFS. Can the LVFS team contact dell to request a checksum? To my knowledge  (I am a little new to all of this), verifying these downloads is very important for the stability/security of the machines they are installed on.

superm1, 2021-04-12:
>failed to verify System Firmware: No device checksums for 1.7.0

This means that no checksums are available for this device on LVFS or locally enrolled for this device.  You can yourself run `fwupdmgr verify-update` to enroll and trust the checksum that currently exists and then you will be able to check if it changes from one boot to another using `fwupdmgr verify`

>Cryptographic hash verification is available for the System Firmware

Let me explain some background here.

0. As an FYI the "checksum" used for the system firmware device is the PCR0 represented on the TPM.  
2. You're running fwupd 1.3.x which has [known issues with the creation of TPM PCR0.](https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction)  You'll want to upgrade to a newer version if you intend to rely upon this functionality.
3. Dell does not currently publish checksum for system firmware to LVFS.  Although a majority of cases they will be the same across many machines this is not guaranteed.  Some light reading on the topic of relying upon PCR0 is available here: https://github.com/fwupd/fwupd/discussions/2548

> KXG60ZNV1T02 NVMe TOSHIBA 1024GB was also updated. It seems that the checksum for this update is also not available. I

This device update is staged from within the OS and does not support cryptographic hash verification from fwupd.  The NVME device will verify the signature on the binary at the time it is uploaded into the drive to install the new firmware.  The firmware will be rejected if it's been tampered with or otherwise the signature does not match.

>To my knowledge (I am a little new to all of this), verifying these downloads is very important for the stability/security of the machines they are installed on.

For System Firmware - Dell's firmware will verify the signature of updated firmware payload without regard to what happens from LVFS or fwupd or anything else.
For NVME firmware, the drive will verify the signature of the payload.

If you choose to use the `verify`/`verify-update` functionality for system firmware, I would suggest you enroll the new checksums with `fwupdmgr verify-update` after installing the new firmware and then use `fwupdmgr verify` to check they don't change.
Since you're on 1.3.x with the known issues, you should know that upgrading to a version with the fix WILL most likely cause them to change though.

alexwms, 2021-04-12:
Thank you for the detailed response! Just to be absolutely clear, when you say

> For System Firmware - Dell's firmware will verify the signature of updated firmware payload without regard to what happens from LVFS or fwupd or anything else.
For NVME firmware, the drive will verify the signature of the payload.

you mean that the binaries that are downloaded from LVFS by `fwupdmgr` have a signature stored inside of them, correct?

superm1, 2021-04-12:
Correct

closed 2021-04-12 by alexwms