fwupd/fwupd #3143: install/activate commands require trusted payloads

djcampello, 2021-04-14:
After 381c27ca44ac6decd7f7e6bf0999a965d1391562 install/activate commands require trusted payloads. This is opposite to what was intended with 7db8e88d98c870259963b05bfa3a87c4db0626f8

superm1, 2021-04-14:
I think that's the intention for `fwupdmgr` - see https://github.com/fwupd/fwupd/pull/2482#issuecomment-714672188.

Are you seeing problems with `fwupdtool`?  Can you share more details on what you're wanting to accomplish?

djcampello, 2021-04-14:
Ohh I missed that... That is exactly what I'm trying to do... I'm working on the daemon prototype again. I have some unsigned .cab that I tried to install via fwupdmgr update/install... Is there any way to do this?

superm1, 2021-04-14:
This is a quite intentional decision with `fwupdgmr` related to the fact that any user process could interact with it.  So you are relying upon the daemon to only allow CAB files that it trusts.

>Ohh I missed that... That is exactly what I'm trying to do... I'm working on the daemon prototype again. I have some unsigned .cab that I tried to install via fwupdmgr update/install... Is there any way to do this?

Yeah - create a local trust authority, sign the cab with it and put the authority in the local fwupd installation.

djcampello, 2021-04-14:
For the case of fwupmgr update where the .cab cannot be supplied, having the .cab being sourced from one of the enabled remotes is not enough validation?

hughsie, 2021-04-14:
The daemon just installs from a file descriptor. I think relying on a checksum matching might be fragile, and makes it difficult to explain why it "works" with up to date metadata, but not offline. I think the need to sign the firmware with *something* and if that's tricky maybe we need to make the tools better.

djcampello, 2021-04-14:
Is there any documentation that outlines the process step by step?

hughsie, 2021-04-14:
https://giphy.com/gifs/moodman-monkey-puppet-meme-reaction-cJMlR1SsCSkUjVY3iK

hughsie, 2021-04-15:
@djcampello before I start working on this, are you _aiming_ to:

 * sign just one file with a test trusted key
 * sign a directory of firmware with a possibly additional trusted key
 * sign a directory of firmware replacing the old signature and using justa new one

I think depending on the goal here the implementation would be different.

djcampello, 2021-04-15:
I believe the objective is to be able to deploy in house firmware that has not been sent to LVFS (not that I'm suggesting this is a good thing) and does not have LVFS signature in them (example: https://chromium.googlesource.com/chromiumos/overlays/board-overlays/+/HEAD/overlay-puff/sys-firmware/parade-ps175-firmware/files/generate_cab.py used to generate firmware used by the lspcon_i2c_spi device in the flashrom plugin). It is useful for firmware for in development devices.

djcampello, 2021-04-15:
@bleungatchromium and @tari for visibility

superm1, 2021-04-15:
> I believe the objective is to be able to deploy in house firmware that has not been sent to LVFS (not that I'm suggesting this is a good thing) and does not have LVFS signature in them (example: [...]

In that case, I wonder if we should have a noinst "development keypair" in tree or something and some helper scripts or a simple flow for using it.

djcampello, 2021-04-15:
If there is no much different work wise, maybe tooling around permanent keypair might be better in case of for some reason (don't have an example right now) the firmware never reaches LVFS and stays in house.

hughsie, 2021-04-15:
> I wonder if we should have a noinst "development keypair"

That's a good idea. Will ponder.

hughsie, 2021-04-16:
@djcampello I pondered something like the commits in wip/hughsie/firmware-sign -- i.e. you'd do `fwupdtool firmware-sign firmware.cab certificate.pem privatekey.pfx` on each file. I guess one optimisation would be to generate a test private key and certificate somewhere if the `pem` and `pfx` are omitted. That branch is just a quick brain dump, it's nowhere ready for review yet, but comments welcome.

djcampello, 2021-04-16:
Thanks for this!! That would be my next question: is there a good reference guide on how to "generate a test private key and certificate"

superm1, 2021-04-17:
> Thanks for this!! That would be my next question: is there a good reference guide on how to "generate a test private key and certificate"

One way to approach this could be to have fwupdtool look in a known location and if they keys exist use them, if not then prompt to generate them directly.

hughsie, 2021-04-18:
> if not then prompt to generate them directly.

I looked at this a bit yesterday; I don't think this is a good idea as it encourages people to do the wrong thing. Generating a certificate and keypair you're going to use for signing purposes is not the kind of thing to do as a quick 5 minute hack -- for instance, you need to know about and spin up a CRL in case you ever want to revoke the key. If you're distributing the binaries on a secure system you probably want to store the private key on a HSM or yubikey rather than some developers machine etc. There's also a ton of info that you need to supply to create a certificate that's actually useful. I think the best thing to do would be to use https://github.com/hughsie/libjcat/blob/master/contrib/build-certs.py for the developer example, and talk it through with the security team at $EMPLOYER$ about the best way to generate a production certificate.

superm1, 2021-04-18:
> I looked at this a bit yesterday; I don't think this is a good idea as it encourages people to do the wrong thing. Generating a certificate and keypair you're going to use for signing purposes is [...]

Great points.

Something worth keeping in mind - using an hsm or yubikey will mean you don't have access to the private key directly as an argument.

hughsie, 2021-04-19:
> using an hsm or yubikey will mean you don't have access to the private key

Yes, indeed. The first vendor that requires the HSM functionality gets the write the JCat code, perhaps.

hughsie, 2021-04-26:
We've got `fwupdtool firmware-sign` now, and I think that covers this issue now.

closed 2021-04-26 by hughsie