fwupd/fwupd #3151: fu_uefi_get_esp_path_for_os() does not handle /etc/os-release "ID_LIKE" entirely properly

bubbleguuum, 2021-04-15:
**Describe the bug**

Function `fu_uefi_get_esp_path_for_os()` (in `fu-uefi-common.c`) does not handle the `/etc/os-release` "ID_LIKE" key entirely properly:

according to `man os-release`, the `ID_LIKE` field is "A space-separated list of operating system identifiers in the same syntax as the ID= setting". However the function treat it like a single value. 

My openSUSE system has this `/etc/os-release` (removed irrelevant keys):

```
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
```

On my system, `fu_uefi_get_esp_path_for_os()` should return `/boot/efi/EFI/opensuse` but it doesn't because it does not split `ID_LIKE` by space to test "opensuse" and "suse" individually until one folder exist (the first). Instead it just tests the presence of folder `/boot/efi/EFI/opensuse suse`

**fwupd version information**
```
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5
daemon version:	1.5.7
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): zypper

- Operating system and version: openSUSE Tumbleweed

superm1, 2021-04-15:
Thanks for reporting, would you by chance be able to work out and submit a patch for it?

bubbleguuum, 2021-04-15:
I've made a patch for it which is simple enough, but I have no way to test it.
Is there a procedure to test a firmware update with a fake firmware (just to check if on reboot the secure boot shim is started) ?

superm1, 2021-04-15:
Great!

you can do `fwupdmgr reinstall` and it will try to to reinstall

bubbleguuum, 2021-04-15:
Well, it turns out that my system refuses to apply firmware update (have been testing with BIOS) is not caused by this issue after all: if I modify ID to "opensuse" in /etc/os-release, it still does not work (system boots as normal). It's Secure Boot specific as if I disable it there is no issue.
So it's something else, although the parsing issue of ID_LIKE is probably a real issue. 
I'm at a loss at how to debug this. Is there a way to have g_print() statement logged to a file so I can have a chance to see them ? I do not even know at which stage `fu_uefi_get_esp_path_for_os()` is supposed to be called...

superm1, 2021-04-15:
> I'm at a loss at how to debug this. Is there a way to have g_debug() statements logged to a file so I can have a chance to see them ? I do not even know at which stage fu_uefi_get_esp_path_for_os() is supposed to be called...

So the best thing to do is decouple the part "in" OS versus on the reboot.  When you get to the part where there is the prompt to reboot, say "NO".  Check your EFI binaries on the ESP - you can compare the checksums on the ESP to those that are in the fwupd package.  Then you can tell what has been installed to the ESP, and make sure the right thing got updates.

You can also check `efibootmgr` output to confirm which binary is being selected.  It could be that the wrong binary is getting picked from all this `ID_LIKE` and `ID` stuff.

If all the right binaries are getting picked and installed, then the next thing I would suggest to check the signed binary is signed with the proper keys (did you forget to do an update?  or something related to a new shim and bad timing of an update, or busted SBAT or something?

bubbleguuum, 2021-04-15:
Thanks for the explanation.

here's efibootmgr output after answering N for reboot:

```
BootNext: 0004
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,0003,0001,001B,0010,0011,0012,0013,0014,0000,0018,0019,001A,001C,001D,001E,001F,0020,0021,0022,0027,0004
Boot0000* Windows Boot Manager	HD(8,GPT,ab16b902-fdd2-f945-b235-bf0f99a6d098,0x72853800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...e................
Boot0001* openSUSE Secure Boot	HD(1,GPT,9a54bd14-137a-4800-9494-38ec72b3809a,0x800,0x82000)/File(EFI\opensuse\shim.efi)
Boot0002* opensuse-secureboot	HD(8,GPT,ab16b902-fdd2-f945-b235-bf0f99a6d098,0x72853800,0x82000)/File(\EFI\opensuse\shim.efi)
Boot0003* openSUSE TEST	HD(8,GPT,ab16b902-fdd2-f945-b235-bf0f99a6d098,0x72853800,0x82000)/File(\EFI\opensuse\shim.efi)
Boot0004* Linux-Firmware-Updater	HD(8,GPT,ab16b902-fdd2-f945-b235-bf0f99a6d098,0x72853800,0x82000)/File(\EFI\opensuse\shim.efi)\.f.w.u.p.d.x.6.4...e.f.i...
```

BootNext seems correct and it's a Lenovo laptop thus the `Linux-Firmware-Updater`. The path to the shim on the Boot0004 line is (\EFI\opensuse\shim.efi) is correct too. 
Not sure about SBAT, although this failure to do firmware update is not recent (I tried it first a few months ago). I just read about briefly about SBAT and it does not seem to be a piece of cake :) but the openSUSE fwupd package seems to set it up with `-Defi_sbat_*` statements: https://build.opensuse.org/package/view_file/home:bobbie424242:branches:openSUSE:Factory/fwupd/fwupd.spec?expand=1.

So it's getting a bit complex for me at this point. It is either: 

- a ThinkPad specific issue
- an issue in the packaging of fwupd in openSUSE Tumbleweed
- something related to the shim and fwupd

bubbleguuum, 2021-04-19:
Mystery solved.

It turns out it is a problem with openSUSE Tumbleweed shim:
https://bugzilla.suse.com/show_bug.cgi?id=1177404

Shall I still submit a PR fixing the ID_LIKE parsing ?

hughsie, 2021-04-19:
> Shall I still submit a PR fixing the ID_LIKE parsing ?

Please.

closed 2021-04-19 by bubbleguuum