fwupd/fwupd #3154: CLI says SSD was updated after MEFW update

der-eismann, 2021-04-16:
**Describe the bug**
After checking today I saw that there is an update for the Management Engine Firmware available. When entering `Y` I got the progress lines and then it said it successfully updated the SSD as well, which isn't correct:
```shell
$ fwupdmgr
Devices with no available firmware updates: 
 • USB3.1 Hub
 • USB3.1 Hub
 • UEFI Device Firmware
 • UEFI dbx
 • USB2.0 Hub
 • USB2.0 Hub
 • VMM5322
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • Embedded Controller
Upgrade available for Intel Management Engine from 184.82.3838 to 184.83.3874
20L7001LGE must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Herunterladen …          [***************************************]
Entpacken …              [***************************************]
Authentifizierung …      [***************************************]
Authentifizierung …      [***************************************]
Intel Management Engine wird aktualisiert …**********************]
Einplanen …              [***************************************]
Successfully installed firmware
 • SAMSUNG MZVLB512HAJQ-000L7
 • System Firmware

Ein Neustart ist erforderlich, um eine Aktualisierung abzuschließen. Jetzt neu starten? [y|N]:
```

**Steps to Reproduce**
`fwupdmgr update` after available MEFW update.

**Expected behavior**
Only a message that the system firmware was updated, not the SSD.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.8
```

Installed via dnf

**fwupd device information**

```shell
$ fwupdmgr get-devices --show-all-devices
20L7001LGE
│
├─Thunderbolt host controller:
│     Device ID:          729dda7cd055f3947a1de93c8f2bbcabd7dfdf94
│     Zusammenfassung:    Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              3849839e-d8b2-5941-9e69-5acc397d3ed3
│                         7e3190a2-d5a5-5a9e-b582-494b47f7288c
│                         061f2847-061b-5cfb-9c39-3dbfe2b49d69
│                         65a3d897-4a18-5d27-83d8-eb8a231b7aa4
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─USB3.1 Hub:
[... 179 lines cut ...]
│                         e2c96871-8be3-530e-a796-5652cee64eeb
│                         6f3f32bd-ac98-523e-9128-ef67941fb614
│                         193e6679-e01f-5afc-a4b9-bcf31e14f09f
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
└─VMM5322:
      Device ID:          585c346c009da3cc19a400fb59a4ed8d0cbd0151
      Zusammenfassung:    Multi-Stream Transport Device
      Current version:    5.05.00
      Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:              586087d5-270b-5e36-904e-a508264287f5
                          f15aa55c-9cd5-5942-85ae-a6bf8740b96c
                          c083122a-aae0-5ce2-9f80-c65a1aa1151b
                          d65b3310-c47a-5e0c-bee0-df7f130126ec
      Device Flags:       • Updatable
                          • Device stages updates
```

**Additional questions**
- Operating system and version: Fedora 33
- Have you tried rebooting? No.
- Is this a regression? Don't know

hughsie, 2021-04-17:
It looks like we really did update the SSD firmware. I think https://github.com/fwupd/fwupd/pull/3078 would help here.

der-eismann, 2021-04-17:
I actually saw this already while doing the previous MEFW upgrade but didn't open a ticket because I forgot to save the logs. In addition I did a `fwupdmgr get-updates` and it only showed the MEFW one and none for the SSD. But again, forgot to save the output for that. Can I flash an older version easily to repeat the process?

hughsie, 2021-04-19:
> Can I flash an older version easily to repeat the process

Maybe, it's up to the OEM to set this policy. `fwupdmgr downgrade` should help.

der-eismann, 2021-07-01:
Sorry I let this go stale, but I was able to reproduce it since there was a new MEFW update available. This time I recorded it with asciinema. Again the SSD was never mentioned before the update, neither as "no update available", nor as "upgrade available". But after the MEFW upgrade finished it is shown in the "successfully installed firmware" part. Not sure if the SSD update was never really finished back then, but it's still popping up with every other update I do. 
[image]

hughsie, 2021-08-03:
I think 1.6.x will help a lot here.

closed 2021-08-03 by hughsie