fwupd/fwupd #3156: fwupd 1.5.9 cannot update capsule, flashrom or superio

Sean-StarLabs, 2021-04-19:
Error when trying to update any device:
`failed to open device for prepare: not yet initialized`

Error can be reproduced when built from source or using deb. Tested on Ubuntu

Verbose output
```
(fwupdmgr:26688): Fwupd-DEBUG: 09:46:43.944: download progress: 100%
(fwupdmgr:26688): Fwupd-DEBUG: 09:46:43.945: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
(fwupdmgr:26688): Fwupd-DEBUG: 09:46:44.003: Emitting ::status-changed() [decompressing]
Decompressing…           [***************************************](fwupdmgr:26688): Fwupd-DEBUG: 09:46:44.205: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:26688): Fwupd-DEBUG: 09:46:44.205: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [***************************************]
==== AUTHENTICATING FOR org.freedesktop.fwupd.downgrade-internal ===
Authentication is required to downgrade the firmware on this machine
Authenticating as: Ben McMillen,,, (ben)
(process:26690): GLib-DEBUG: 09:46:44.208: posix_spawn avoided (fd close requested) 
Password: 
==== AUTHENTICATION COMPLETE ===
(fwupdmgr:26688): Fwupd-DEBUG: 09:46:49.121: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:26688): Fwupd-DEBUG: 09:46:49.130: Emitting ::device-changed(dbee8bd3b1ae0316ad143336155651eedb495a0e)
(fwupdmgr:26688): FuMain-DEBUG: 09:46:49.130: ignoring LabTop with status idle
failed to open device for prepare: not yet initialized
```
Verbose switch-branch
```
fwupdmgr switch-branch --verbose
(fwupdmgr:26179): GLib-DEBUG: 09:42:10.256: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:26179): Fwupd-DEBUG: 09:42:10.275: Emitting ::status-changed() [idle]
Selected device: LabTop
failed to process components/component[@type='firmware']/provides/firmware[@type=$'flashed'][text()=$'230c8b18-8d9b-53ec-838b-6cfc0383493a']/../..|components/component[@type='firmware']/provides/firmware[@type=$'flashed'][text()=$'165714e1-d63b-500a-8870-b944ebaa749f']/../..|components/component[@type='firmware']/provides/firmware[@type=$'flashed'][text()=$'82553f40-b9b8-5008-96b6-96ccfdc04f43']/../..|components/component[@type='firmware']/provides/firmware[@type=$'flashed'][text()=$'69a66544-ef29-57e8-bb18-81f2e41a3c75']/../..|components/component[@type='firmware']/provides/firmware[@type=$'flashed'][text()=$'3d69caac-479a-5c2a-8978-04a356505bc5']/../..|components/component[@type='firmware']/provides/firmware[@type=$'flashed'][text()=$'e7fae735-91a4-5702-b171-a62fe746aa43']/../..|components/component[@type='firmware']/provides/firmware[@type=$'flashed'][text()=$'d33219e2-b84c-53a8-a624-27af9752dba6']/../..|components/component[@type='firmware']/provides/firmware[@type=$'flashed'][text()=$'(null)']/../..: cannot parse text or number `null`

```

Version:
```shell
client version:	1.6.0
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.6.0

```
Journalctl has nothing relavant

dmesg:
```
[ 3546.972557] fwupd[6127]: segfault at 7fd12976a240 ip 00007fd12976a240 sp 00007ffe2519f208 error 14 in libfu_plugin_nvme.so (deleted)[7fd129777000+2000]
[ 3798.508533] fwupd[24962]: segfault at 0 ip 00007f96a824a5f4 sp 00007fff75605158 error 4 in libglib-2.0.so.0.6400.6[7f96a8226000+84000]
[ 3831.255861] fwupd[25706]: segfault at 0 ip 00007f141571e5f4 sp 00007ffea205d428 error 4 in libglib-2.0.so.0.6400.6[7f14156fa000+84000]
[ 3868.358321] fwupd[25828]: segfault at 0 ip 00007fe3b7ec55f4 sp 00007ffefb6c82a8 error 4 in libglib-2.0.so.0.6400.6[7fe3b7ea1000+84000]
[ 3891.383757] fwupd[25886]: segfault at 7fa1a44eb240 ip 00007fa1a44eb240 sp 00007fff6fcc54e8 error 14 in libfu_plugin_nvme.so[7fa1a44f8000+2000]
[ 3895.004960] fwupd[26016]: segfault at 7fc769801240 ip 00007fc769801240 sp 00007fff17e93d48 error 14 in libfu_plugin_nvme.so[7fc76980e000+2000]
```

```shell
LabTop
│
├─LabTop:
│     Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│     Current version:    4
│     Vendor:             Star Labs (DMI:coreboot)
│     Release Branch:     coreboot
│     Update State:       Failed
│     Update Error:       Cannot install update when not on AC power
│     Last modified:      2021-04-19 08:41
│     GUIDs:              230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         165714e1-d63b-500a-8870-b944ebaa749f
│                         82553f40-b9b8-5008-96b6-96ccfdc04f43
│                         69a66544-ef29-57e8-bb18-81f2e41a3c75
│                         3d69caac-479a-5c2a-8978-04a356505bc5
│                         e7fae735-91a4-5702-b171-a62fe746aa43
│                         d33219e2-b84c-53a8-a624-27af9752dba6
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Updatable
│                         • Device supports switching to a different branch of firmware
│   
├─Star Drive PCIe SSD:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express Solid State Drive
│     Current version:    ECFM12.3
│     Vendor:             Phison Electronics Corporation (NVME:0x1987)
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              38d64a8a-1cd1-589c-9f70-b0b9c4379c43
│                         94e27f4a-86e3-53a2-a728-18db5dd2be18
│                         8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
│                         a44eb54c-5441-56f2-8cc0-5e48964c6457
│                         12d61493-5ac4-57f7-a2c5-96878d115455
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs shutdown after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
└─SuperIO IT8987:
      Device ID:          1f664ad149f2151fb750b52751dbf9ba045a7846
      Summary:            Embedded Controller
      Current version:    03.12
      Vendor:             DMI:Star Labs
      Update Error:       Cannot install update when not on AC power
      GUID:               0aa07091-dc02-5e62-a0cd-9d3880721b64
      Device Flags:       • Internal device
                          • Update requires a reboot
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Updatable
    
Idle…                    [***************************************]
Successfully uploaded 1 report```

hughsie, 2021-04-19:
Okay, a lot to unpick here. Lets go to `[ 3895.004960] fwupd[26016]: segfault at 7fc769801240 ip 00007fc769801240 sp 00007fff17e93d48 error 14 in libfu_plugin_nvme.so[7fc76980e000+2000]` first -- can you get a backtrace?

hughsie, 2021-04-19:
Yup, something is adding a NULL GUID to a device, and that's making fwupd very upset in various places.

Sean-StarLabs, 2021-04-19:
> Okay, a lot to unpick here. Lets go to `[ 3895.004960] fwupd[26016]: segfault at 7fc769801240 ip 00007fc769801240 sp 00007fff17e93d48 error 14 in libfu_plugin_nvme.so[7fc76980e000+2000]` first -- can you get a backtrace?

I can't seem to get it to segfault on command (switch-branch, get-devices, update, refresh) but it's still happening:

```
[ 4210.702838] fwupd[26579]: segfault at 7f516acd7240 ip 00007f516acd7240 sp 00007fff6e4a4d68 error 14 in libfu_plugin_nvme.so[7f516ace4000+2000]
[ 4337.611925] fwupd[26643]: segfault at 7f1e80346240 ip 00007f1e80346240 sp 00007ffe1eb8f688 error 14 in libfu_plugin_nvme.so[7f1e80353000+2000]
[ 5784.674815] fwupd[26874]: segfault at 7f7892cd7240 ip 00007f7892cd7240 sp 00007fff112366b8 error 14 in libfu_plugin_nvme.so[7f7892ce4000+2000]
[ 7368.307498] fwupd[29841]: segfault at 7f76e98ff240 ip 00007f76e98ff240 sp 00007ffdc2db3fe8 error 14 in libfu_plugin_nvme.so[7f76e990c000+2000]
[ 7428.385094] fwupd[33657]: segfault at 7f1140702240 ip 00007f1140702240 sp 00007ffff7aa6948 error 14 in libfu_plugin_nvme.so[7f114070f000+2000]
[ 7512.954599] fwupd[33831]: segfault at 7f0739660240 ip 00007f0739660240 sp 00007ffd848fc998 error 14 in libfu_plugin_nvme.so[7f073966d000+2000]
```

Am I right in thinking need to somehow attach gdb to libfwupd2?

hughsie, 2021-04-19:
> Am I right in thinking need to somehow attach gdb to libfwupd2?

Nah, the main binary. e.g. `gdb /usr/libexec/fwupd/fwupd`

Sean-StarLabs, 2021-04-19:
I'm not actually sure what I'm looking for!

```
(gdb) run fwupdmgr get-devices
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /usr/libexec/fwupd/fwupd fwupdmgr get-devices
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff3ae4700 (LWP 5139)]
[New Thread 0x7ffff2fd1700 (LWP 5140)]
[New Thread 0x7ffff27c9700 (LWP 5141)]
[New Thread 0x7ffff1a8c700 (LWP 5142)]
10:50:52:0587 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: System does not support UEFI mode
10:50:52:0601 FuDevice             4 is not a valid triplet (guessed number)
[New Thread 0x7ffff128b700 (LWP 5160)]
10:50:53:0078 FuMain               another service has claimed the dbus name org.freedesktop.fwupd

Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.
0x00007ffff1f11240 in ?? ()
(gdb) backtrace
#0  0x00007ffff1f11240 in  ()
#1  0x00007ffff7c07d0e in g_object_unref () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#2  0x00007ffff7b01232 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7b026b3 in g_hash_table_unref () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff7f2c6a5 in  () at /usr/lib/x86_64-linux-gnu/libfwupdplugin.so.2
#5  0x00007ffff7c07d0e in g_object_unref () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#6  0x00007ffff7b01232 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x00007ffff7b026b3 in g_hash_table_unref () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x0000555555585792 in  ()
#9  0x00007ffff7c07d0e in g_object_unref () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x0000555555573137 in  ()
#11 0x00007ffff7c07d0e in g_object_unref () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00005555555691e8 in main ()

```

hughsie, 2021-04-19:
Can you do it with debug symbols installed for fwupd? I'm not teribly sure how to do that on Ubuntu. In the `(gdb)` prompt you can do `r --verbose` if that helps.

Sean-StarLabs, 2021-04-19:
Installing the debug symbols for fwupd, libfwupd2 and libfwupdplugin1 doesn't change the output. Do you know if there are any others that are needed?

hughsie, 2021-04-19:
Can you try with master again? You should get a critical warning rather than a crash now. Also, what hardware? Anything I've got here?

superm1, 2021-04-19:
To me this looks like it's a crash in the cleanup path, and it's happening when there is a daemon running and attempting to start another.  If I was going to guess where the crash is maybe along the context stuff? https://github.com/fwupd/fwupd/commit/b333e0045cc1b625084ba974444e6c809ba7e4fe

Sean-StarLabs, 2021-04-20:
> Can you try with master again? You should get a critical warning rather than a crash now. Also, what hardware? Anything I've got here?

Still get a seg fault:
```
06:16:16:0488 FuHistory            add device LabTop [dbee8bd3b1ae0316ad143336155651eedb495a0e]
06:16:16:0490 FuEngine             prepare -> FuFlashromInternalDevice:
LabTop
  DeviceId:             dbee8bd3b1ae0316ad143336155651eedb495a0e
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a <- main-system-firmware
  Guid:                 165714e1-d63b-500a-8870-b944ebaa749f <- Star Labs&L3&LabTop&L3-U&Star Labs&LabTop
  Guid:                 82553f40-b9b8-5008-96b6-96ccfdc04f43 <- Star Labs&L3&LabTop&L3-U
  Guid:                 69a66544-ef29-57e8-bb18-81f2e41a3c75 <- Star Labs&L3&LabTop
  Guid:                 3d69caac-479a-5c2a-8978-04a356505bc5 <- Star Labs&L3-U&Star Labs&LabTop
  Guid:                 e7fae735-91a4-5702-b171-a62fe746aa43 <- Star Labs&L3&Star Labs&LabTop
  Guid:                 d33219e2-b84c-53a8-a624-27af9752dba6 <- Star Labs&L3&LabTop&L3-U&coreboot
  Branch:               coreboot
  Plugin:               flashrom
  Protocol:             org.flashrom
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot|has-multiple-branches
  Vendor:               Star Labs
  VendorId:             DMI:coreboot
  Version:              5
  VersionFormat:        number
  Icon:                 computer
  Created:              2021-04-20
  Modified:             2021-04-20
  UpdateState:          failed
  PhysicalId:           flashrom
  LogicalId:            bios
  FirmwareSizeMax:      8388608
  Order:                0
  InternalFlags:        ensure-semver

06:16:16:0490 FuDevice             fu_device_add_instance_id_full: assertion 'instance_id != NULL' failed
06:16:16:0490 FuDevice             fu_device_ensure_id: assertion 'error == NULL || *error == NULL' failed
06:16:16:0490 FuMain               ignoring LabTop with status idle
06:16:16:0490 FuHistory            modifying device LabTop [dbee8bd3b1ae0316ad143336155651eedb495a0e]
06:16:16:0493 FuPlugin             composite_cleanup(dell_dock)
06:16:16:0493 FuIdle               uninhibiting: update
failed to open device for prepare: not yet initialized
06:16:16:0495 FuPlugin             destroy(synaptics_mst)
06:16:16:0495 FuPlugin             destroy(logind)
06:16:16:0495 FuPlugin             destroy(upower)
06:16:16:0496 FuPlugin             destroy(linux_lockdown)
06:16:16:0496 FuPlugin             destroy(tpm_eventlog)
06:16:16:0496 FuPlugin             destroy(linux_swap)
06:16:16:0496 FuPlugin             destroy(dell)
06:16:16:0496 FuPlugin             destroy(linux_tainted)
Segmentation fault
```

Hardware is a LabTop Mk III, using Ubuntu Focal. I think you've got a PCBA?

Sean-StarLabs, 2021-04-20:
That's done the trick - thank you very much as always :)

Sean-StarLabs, 2021-04-21:
Apologies didn't test enough! The segfault is 100% gone, but something isn't quite right...

```
sean@sean-LabTop:~/Documents/firmware/Star LabTop Mk IV/coreboot/4$ fwupdmgr refresh 
Firmware metadata last refresh: 2 minutes ago. Use --force to refresh again.
sean@sean-LabTop:~/Documents/firmware/Star LabTop Mk IV/coreboot/4$ fwupdmgr update 
Devices with no available firmware updates: 
 • LabTop
 • Star Drive PCIe SSD
 • SuperIO IT8987
No updatable devices
sean@sean-LabTop:~/Documents/firmware/Star LabTop Mk IV/coreboot/4$ fwupdmgr update --verbose
(fwupdmgr:11281): GLib-DEBUG: 19:53:17.325: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:11281): Fwupd-DEBUG: 19:53:17.335: Emitting ::status-changed() [idle]
Devices with no available firmware updates: 
 • LabTop
 • Star Drive PCIe SSD
 • SuperIO IT8987
No updatable devices
(fwupdmgr:11281): FuMain-DEBUG: 19:53:17.342: No updatable devices

sean@sean-LabTop:~/Documents/firmware/Star LabTop Mk IV/coreboot/4$ fwupdmgr install coreboot-L4.cab 
Decompressing…           [***************************************]
Authenticating…          [ -                                     ]
Authenticating…          [***************************************]
device ID dbee8bd3b1ae0316ad143336155651eedb495a0e was not found
sean@sean-LabTop:~/Documents/firmware/Star LabTop Mk IV/coreboot/4$ fwupdmgr install coreboot-L4.cab --verbose
(fwupdmgr:11303): GLib-DEBUG: 19:53:30.959: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:11303): Fwupd-DEBUG: 19:53:30.985: Emitting ::status-changed() [idle]
(fwupdmgr:11303): Fwupd-DEBUG: 19:53:31.009: Emitting ::status-changed() [decompressing]
Decompressing…           [ -                                     ](fwupdmgr:11303): Fwupd-DEBUG: 19:53:31.048: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:11303): Fwupd-DEBUG: 19:53:31.048: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ]
(fwupdmgr:11303): Fwupd-DEBUG: 19:53:31.050: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:11303): Fwupd-DEBUG: 19:53:31.830: Emitting ::status-changed() [device-write]
Writing…                 [  -                                    ](fwupdmgr:11303): Fwupd-DEBUG: 19:53:31.830: Emitting ::device-changed(779e8c1670b51fe042bd33cd2e350f937e28853d)
Installing on LabTop…
Writing…                 [  -                                    ]

```

Not quite sure where to go, as the `--verbose` switch seems to just make things work...

Sean-StarLabs, 2021-04-21:
Downgrade has the same behaviour:
```
sean@sean-LabTop:~$ fwupdmgr downgrade 
Choose a device:
0.	Cancel
1.	dbee8bd3b1ae0316ad143336155651eedb495a0e (LabTop)
2.	71b677ca0f1bc2c5b804fa1d59e52064ce589293 (Star Drive PCIe SSD)
3.	1f664ad149f2151fb750b52751dbf9ba045a7846 (SuperIO IT8987)
1
Choose a release:
0.	Cancel
1.	3 (
This stable release includes the following features:

• Updated VBT ID's
• Removed unused ACPI
• Rebase on upstream
• Updated FSP configuration
• Disabled S4)
2.	3 (
This stable release includes the following features:

• Updated VBT ID's
• Removed unused ACPI
• Rebase on upstream
• Updated FSP configuration
• Disabled S4)
3.	2 (
This stable release includes the following features:

• Rebase on upstream
• Changed to ASL v2 syntax)
4.	2 (
This stable release includes the following features:

• Rebase on upstream
• Changed to ASL v2 syntax)
5.	1 (
This stable release includes the following features:

• Rebase on upstream
• Updated BGRT Logo
• Reverted to FSP)
6.	1 (
This stable release includes the following features:

• Rebase on upstream
• Updated BGRT Logo
• Reverted to FSP)
3
LabTop and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading…             [****                                   ] Less than oneDownloading…             [*****                                  ] Less than oneDownloading…             [*****                                  ] Less than oneDownloading…             [*****                                  ] Less than oneDownloading…             [******                                 ] Less than oneDownloading…             [*******                                ] Less than oneDownloading…             [*******                                ] Less than oneDownloading…             [*******                                ] Less than oneDownloading…             [********                               ] Less than oneDownloading…             [********                               ] Less than oneDownloading…             [********                               ] Less than oneDownloading…             [*********                              ] Less than oneDownloading…             [*********                              ] Less than oneDownloading…             [**********                             ] Less than oneDownloading…             [**********                             ] Less than oneDownloading…             [**********                             ] Less than oneDownloading…             [***********                            ] Less than oneDownloading…             [***********                            ] Less than oneDownloading…             [************                           ] Less than oneDownloading…             [************                           ] Less than oneDownloading…             [************                           ] Less than oneDownloading…             [*************                          ] Less than oneDownloading…             [*************                          ] Less than oneDownloading…             [**************                         ] Less than oneDownloading…             [**************                         ] Less than oneDownloading…             [**************                         ] Less than oneDownloading…             [***************                        ] Less than oneDownloading…             [***************                        ] Less than oneDownloading…             [***************                        ] Less than oneDownloading…             [****************                       ] Less than oneDownloading…             [****************                       ] Less than oneDownloading…             [*****************                      ] Less than oneDownloading…             [*****************                      ] Less than oneDownloading…             [*****************                      ] Less than oneDownloading…             [******************                     ] Less than oneDownloading…             [******************                     ] Less than oneDownloading…             [*******************                    ] Less than oneDownloading…             [*******************                    ] Less than oneDownloading…             [*******************                    ] Less than oneDownloading…             [********************                   ] Less than oneDownloading…             [********************                   ] Less than oneDownloading…             [*********************                  ] Less than oneDownloading…             [*********************                  ] Less than oneDownloading…             [*********************                  ] Less than oneDownloading…             [**********************                 ] Less than oneDownloading…             [**********************                 ] Less than oneDownloading…             [***********************                ] Less than oneDownloading…             [***********************                ] Less than oneDownloading…             [***********************                ] Less than oneDownloading…             [************************               ] Less than oneDownloading…             [************************               ] Less than oneDownloading…             [************************               ] Less than oneDownloading…             [*************************              ] Less than oneDownloading…             [************************* [... cut ...]
(fwupdmgr:3176): Fwupd-DEBUG: 20:04:23.501: download progress: 100%
(fwupdmgr:3176): Fwupd-DEBUG: 20:04:23.501: download progress: 100%
(fwupdmgr:3176): Fwupd-DEBUG: 20:04:23.501: download progress: 100%
(fwupdmgr:3176): FuMain-DEBUG: 20:04:23.505: server message: 94d603aaae04bc66555c522225667d8de9d68911 replaces old report
(fwupdmgr:3176): FuMain-DEBUG: 20:04:23.505: setting flag on dbee8bd3b1ae0316ad143336155651eedb495a0e
Successfully uploaded 1 report

An update requires a reboot to complete. Restart now? [y|N]: 
```

hughsie, 2021-04-21:
> No updatable devices

what does `fwupdmgr get-devices --show-all` show when you get this?

Sean-StarLabs, 2021-04-21:
Correct GUIDs
```
fwupdmgr get-devices --show-all
LabTop
│
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    04
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              3b3450b3-f781-5288-8017-c5ae3aa186fe
│                         645b4549-8abb-54da-a460-42b387eceb33
│                         f9d817fa-d9b4-5956-b0f2-9d351faaf9dd
│                         e5b3b407-04cc-518e-9230-71b318fd1bd8
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i7-10710U CPU @ 1.10GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000e0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         a39cda30-2c5e-508a-a5ad-4836c5b7ee2d
│                         01727d98-7092-5210-ac7d-e12138ea29f4
│     Device Flags:       • Internal device
│   
├─LabTop:
│     Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│     Current version:    3
│     Vendor:             Star Labs (DMI:coreboot)
│     Release Branch:     coreboot
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         f8267299-61cc-539a-ab32-660649ddd0cf
│                         6dae6fef-f184-510d-bba7-9e89c4f4b1e0
│                         70bd7b9d-aae5-520f-91fc-aeb07c93c2f8
│                         6da40160-2553-5256-b78c-4d85349743f0
│                         0ee5867c-93f0-5fb4-adf1-9d686ea1183a
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Updatable
│   
├─Star Drive PCIe SSD:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    ECFM12.3
│     Vendor:             Phison Electronics Corporation (NVME:0x1987)
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              38d64a8a-1cd1-589c-9f70-b0b9c4379c43
│                         94e27f4a-86e3-53a2-a728-18db5dd2be18
│                         8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
│                         a44eb54c-5441-56f2-8cc0-5e48964c6457
│                         12d61493-5ac4-57f7-a2c5-96878d115455
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs shutdown after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
└─SuperIO IT8987:
      Device ID:          1f664ad149f2151fb750b52751dbf9ba045a7846
      Summary:            Embedded Controller
      Current version:    01.04
      Vendor:             DMI:Star Labs
      GUID:               0aa07091-dc02-5e62-a0cd-9d3880721b64
      Device Flags:       • Internal device
                          • Updatable
                          • Update requires a reboot
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available

```

hughsie, 2021-04-21:
> │     Update Error:       Cannot install update when not on AC power

Sean-StarLabs, 2021-04-21:
It's on charge!

```
sean@sean-LabTop:~$ upower -i /org/freedesktop/UPower/devices/battery_BAT0 | grep state
    state:               charging
sean@sean-LabTop:~$ fwupdmgr get-devices 
LabTop
│
├─LabTop:
│     Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│     Current version:    3
│     Vendor:             Star Labs (DMI:coreboot)
│     Release Branch:     coreboot
│     GUIDs:              230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         f8267299-61cc-539a-ab32-660649ddd0cf
│                         6dae6fef-f184-510d-bba7-9e89c4f4b1e0
│                         70bd7b9d-aae5-520f-91fc-aeb07c93c2f8
│                         6da40160-2553-5256-b78c-4d85349743f0
│                         0ee5867c-93f0-5fb4-adf1-9d686ea1183a
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│   
├─Star Drive PCIe SSD:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    ECFM12.3
│     Vendor:             Phison Electronics Corporation (NVME:0x1987)
│     GUIDs:              38d64a8a-1cd1-589c-9f70-b0b9c4379c43
│                         94e27f4a-86e3-53a2-a728-18db5dd2be18
│                         8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
│                         a44eb54c-5441-56f2-8cc0-5e48964c6457
│                         12d61493-5ac4-57f7-a2c5-96878d115455
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs shutdown after installation
│                         • Device is usable for the duration of the update
│   
└─SuperIO IT8987:
      Device ID:          1f664ad149f2151fb750b52751dbf9ba045a7846
      Summary:            Embedded Controller
      Current version:    01.04
      Vendor:             DMI:Star Labs
      GUID:               0aa07091-dc02-5e62-a0cd-9d3880721b64
      Device Flags:       • Internal device
                          • Updatable
                          • Update requires a reboot
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
    
sean@sean-LabTop:~$ fwupdmgr update 
Devices with no available firmware updates: 
 • LabTop
 • Star Drive PCIe SSD
Devices with the latest available firmware version:
 • SuperIO IT8987
sean@sean-LabTop:~$ 
```

hughsie, 2021-04-21:
Hm, interesting. Can you grab me three things when you see this please:

 * the full `upower -i /org/freedesktop/UPower/devices/DisplayDevice` output
 * the full `sudo ./src/fwupdtool get-devices --verbose` output pls
 * the full `sudo ./src/fwupd --verbose` output pls

Sean-StarLabs, 2021-04-22:
Sure:
```
sean@sean-LabTop:~$ upower -i /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Thu 22 Apr 2021 07:11:13 BST (105 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               charging
    warning-level:       none
    energy:              48.7768 Wh
    energy-full:         55.6396 Wh
    energy-rate:         9.9104 W
    time to full:        41.5 minutes
    percentage:          87%
    icon-name:          'battery-full-charging-symbolic'

sean@sean-LabTop:~$ sudo fwupdtool get-devices --verbose
[sudo] password for sean: 
06:13:18:0823 FuDebug              Verbose debugging enabled (on console 1)
06:13:18:0833 FuConfig             loading config values from /etc/fwupd/daemon.conf
06:13:18:0834 FuConfig             using autodetected max archive size 18.4 EB
06:13:18:0834 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
06:13:18:0834 FuRemoteList         loading /usr/share/fwupd/metainfo
06:13:18:0835 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
06:13:18:0835 XbSilo               file: 93585fd0-5db1-e07a-0536-b2b16699a25c, current:93585fd0-5db1-e07a-0536-b2b16699a25c, cached: (null)
06:13:18:0835 XbSilo               loading silo with file contents
06:13:18:0835 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
06:13:18:0835 FuRemoteList         loading remote from /etc/fwupd/remotes.d/starlabs-embargo.conf
06:13:18:0835 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
06:13:18:0835 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
06:13:18:0835 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs.conf.dpkg-dist
06:13:18:0835 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.dpkg-dist
06:13:18:0835 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
06:13:18:0836 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
06:13:18:0836 FuRemoteList         ordering starlabs-embargo=lvfs+1
06:13:18:0836 FuRemoteList         ignoring unfound remote fwupd
06:13:18:0836 FuRemoteList         ordering lvfs-testing=lvfs+1
06:13:18:0836 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
06:13:18:0836 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
06:13:18:0836 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:13:18:0836 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
06:13:18:0836 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
06:13:18:0836 Jcat                 loaded 1 certificates
06:13:18:0836 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:13:18:0836 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
06:13:18:0836 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
06:13:18:0836 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
06:13:18:0836 Jcat                 loaded 1 certificates
06:13:18:0836 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
06:13:18:0836 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
06:13:18:0843 FuEngine             client certificate exists and working
06:13:18:0843 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
06:13:18:0843 FuHistory            got schema version of 6
06:13:18:0843 FuHwids              smbios property Manufacturer=Star Labs
06:13:18:0843 FuHwids              smbios property EnclosureKind=a
06:13:18:0843 FuHwids              smbios property Family=L4
06:13:18:0843 FuHwids              smbios property ProductName=LabTop
06:13:18:0843 FuHwids              smbios property ProductSku=L4
06:13:18:0844 FuHwids              smbios property BiosVendor=coreboot
06:13:18:0844 FuHwids              smbios property BiosVersion=CB_3
06:13:18:0844 FuHwids              smbios property BiosMajorRelease=04
06:13:18:0844 FuHwids              smbios property BiosMinorRelease=0d
06:13:18:0844 FuHwids              smbios property BaseboardManufacturer=Star Labs
06:13:18:0844 FuHwids              smbios property BaseboardProduct=LabTop
06:13:18:0844 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
06:13:18:0844 XbSilo               file: e8806c52-584c-2cad-7790-de1bbd58cb9b, current:e8806c52-584c-2cad-7790-de1bbd58cb9b, cached: (null)
06:13:18:0844 XbSilo               loading silo with file contents
06:13:18:0844 FuEngine             749 components now in silo
[... 963 lines cut ...]
06:13:35:0388 FuEngine             requirement 1.2.4 ge 1.6.0 -> org.freedesktop.fwupd passed
06:13:35:0388 FuEngine             HWID provided b31aa89d-cfb8-5c67-8502-ce76a9e7b580
06:13:35:0388 FuEngine             ignoring ECFM12.3 == ECFM12.3
06:13:35:0388 FuEngine             ignoring ECFM12.1 < ECFM12.3
06:13:35:0388 FuEngine             ignoring ECFM12.3 == ECFM12.3
06:13:35:0388 FuEngine             ignoring ECFM12.1 < ECFM12.3
06:13:35:0388 FuEngine             requirement 1.2.6 ge 1.6.0 -> org.freedesktop.fwupd passed
[... cut ...]
06:13:35:0389 FuEngine             ignoring 1.04 == 01.04
06:13:35:0419 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
06:13:35:0419 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
06:13:35:0443 FuMain               Daemon ready for requests (locale en_GB.UTF-8)
06:13:35:0448 FuMain               acquired name: org.freedesktop.fwupd
```

The last one just seems to sit there indefinitely

Sean-StarLabs, 2021-04-22:
Tell a lie:
```
06:19:14:0233 FuContext            battery level now 90
06:20:03:0943 FuDevice             using 279ed287-3607-549e-bacc-f873bb9838c4 for HIDRAW\VEN_046D&DEV_C52B
06:20:03:0955 FuPlugin             ae7e4ff6-05a9-52bb-ae2e-8e3412644674,eddee2ef-5d1b-51fa-8bef-48e896801259,4fea7028-e16d-5715-b88d-7bdc3c638aa5 has no updates, so ignoring device
06:21:23:0212 FuContext            battery level now 91
06:25:23:0237 FuContext            battery level now 92
06:27:23:0241 FuContext            battery level now 93
```

hughsie, 2021-04-22:
Ahh, so the problem isn't "battery detected when on AC" -- more:

    Devices with no available firmware updates: 
     • LabTop
     • Star Drive PCIe SSD

i.e. why don't they show up with the `SUPPORTED` flag. It's actually all related, and you found an interesting bug:

 * Boot system on battery -> devices moved from UPDATABLE -> UPDATABLE_HIDDEN
 * Refresh metadata (filter on UPDATED -> SUPPORTED)
 * Plug in AC power, so UPDATABLE_HIDDEN -> UPDATABLE
 * Device not SUPPORTED as we filtered on just the UPDATABLE flag

closed 2021-04-26 by hughsie