fwupd/fwupd #3164: fwupd not detecting the ESP on Dell Precision 5530

lgp171188, 2021-04-20:
**Describe the bug**
fwupd is unable to detect the EFI System partition on my Dell Precision 5530 running an up-to-date Manjaro installation even though it has the correct GUID set.

**Steps to Reproduce**
Run `fwupdmgr refresh` and here is the output

```
$ fwupdmgr refresh                                                                                                                                                                                                                    1 ↵
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Firmware metadata last refresh: 27 minutes ago. Use --force to refresh again.
```

`/dev/nvme0n1p1` is the ESP and it has the partition type set.

```
$ sudo blkid -p /dev/nvme0n1p1                                                                                                                                                                                                        2 ↵
/dev/nvme0n1p1: UUID="ED7A-903E" VERSION="FAT32" BLOCK_SIZE="512" TYPE="vfat" USAGE="filesystem" PART_ENTRY_SCHEME="gpt" PART_ENTRY_NAME="primary" PART_ENTRY_UUID="8043f727-4088-49a4-8944-2d93827d2a29" PART_ENTRY_TYPE="c12a7328-f81f-11d2-ba4b-00a0c93ec93b" PART_ENTRY_NUMBER="1" PART_ENTRY_OFFSET="2048" PART_ENTRY_SIZE="1046528" PART_ENTRY_DISK="259:0"
```

**Expected behavior**
`fwupd` detects the ESP using the partition type.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.5.9
compile-time dependency versions
        gusb:   0.3.6

daemon version: 1.5.9
```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`pacman` on Manjaro Linux.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Precision 5530
│
├─GP107GLM [Quadro P2000 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              60b8f473-a839-5548-828f-623151572606
│                         5e9e31be-7aad-5823-9a9f-a00616951165
│                         bc8f6213-8695-5407-be5d-cd03808d40af
│                         07633373-c1be-577f-a080-1f9ae17806eb
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i7-8850H CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000de
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         809a0b93-8a12-5338-a571-ad5583acf896
│                         72ec2ff3-49ff-5ec2-bdbb-525badd47543
│     Device Flags:       • Internal device
│   
├─KXG50ZNV512G NVMe TOSHIBA 512GB:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
[... 374 lines cut ...]
    Model:                      KXG50ZNV512G NVMe TOSHIBA 512GB
    Optical:                    false
    OpticalBlank:               false
    OpticalNumAudioTracks:      0
    OpticalNumDataTracks:       0
    OpticalNumSessions:         0
    OpticalNumTracks:           0
    Removable:                  false
    Revision:                   AADA4105
    RotationRate:               0
    Seat:                       seat0
    Serial:                     88NS10KUT8JQ
    SiblingId:                  
    Size:                       512110190592
    SortKey:                    00coldplug/00fixed/nvme0n1
    TimeDetected:               1618902522237874
    TimeMediaDetected:          1618902522237874
    Vendor:                     
    WWN:                        eui.000000000000001000080d02004d9a14
```

**Additional questions**
- Operating system and version: Manjaro Linux, up-to-date.
- Have you tried rebooting? Yes
- Is this a regression? Maybe, since this was working till a recent `fwupd` upgrade?

hughsie, 2021-04-20:
The partition was not detected by UDisks:

    org.freedesktop.UDisks2.Partition:
    Flags:              0
    IsContained:        false
    IsContainer:        false
    Name:               
    Number:             1
    Offset:             1048576
    Size:               535822336
    Table:              '/org/freedesktop/UDisks2/block_devices/nvme0n1'
    Type:               
    UUID:               

The `Type` and `UUID` are not being set, and so fwupd doesn't know the partition type.

hughsie, 2021-04-20:
> is this a UDisks bug

I think so, yes.

closed 2021-06-09 by stale[bot]