fwupd/fwupd #3165: Daemon has loaded 3rd party code

lgp171188, 2021-04-20:
**Describe the bug**
`fwupd` keeps complaining that the daemon has loaded 3rd party code on my Dell Precision 5530 which has an up-to-date Manjaro installation.

**Steps to Reproduce**
Run any `fwupd` command like `fwupmgr refresh` and the following error is shown.

```shell
$ fwupdmgr refresh                       
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Firmware metadata last refresh: 35 minutes ago. Use --force to refresh again.
```

**Expected behavior**
No such error about 3rd party code is shown.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.5.9
compile-time dependency versions
        gusb:   0.3.6

daemon version: 1.5.
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`pacman`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
$ fwupdmgr get-devices --show-all-devices
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Precision 5530
│
├─GP107GLM [Quadro P2000 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              60b8f473-a839-5548-828f-623151572606
│                         5e9e31be-7aad-5823-9a9f-a00616951165
│                         bc8f6213-8695-5407-be5d-cd03808d40af
│                         07633373-c1be-577f-a080-1f9ae17806eb
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i7-8850H CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000de
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         809a0b93-8a12-5338-a571-ad5583acf896
│                         72ec2ff3-49ff-5ec2-bdbb-525badd47543
│     Device Flags:       • Internal device
│   
├─KXG50ZNV512G NVMe TOSHIBA 512GB:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:            NVM Express Solid State Drive
│     Current version:    AADA4105
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     GUIDs:              26951d55-ff62-5baf-8ead-5df1d8eee13f
│                         23e442d8-2af4-5c60-81f5-ac41aaee72c7
[... 594 lines cut ...]
  Created:              2021-04-20

07:57:02:0252 FuPlugin             destroy(platform_integrity)
07:57:02:0252 FuPlugin             destroy(synaptics_mst)
07:57:02:0252 FuPlugin             destroy(logind)
07:57:02:0252 FuPlugin             destroy(upower)
07:57:02:0252 FuPlugin             destroy(linux_lockdown)
07:57:02:0253 FuPlugin             destroy(tpm_eventlog)
07:57:02:0253 FuPlugin             destroy(linux_swap)
07:57:02:0253 FuPlugin             destroy(dell)
07:57:02:0253 FuPlugin             destroy(flashrom)
07:57:02:0253 FuPlugin             destroy(linux_tainted)
07:57:02:0253 FuPlugin             destroy(redfish)
07:57:02:0253 FuPlugin             destroy(uefi_capsule)

```
The relevant error log related to this issue appears to be
```
07:57:01:0377 FuEngine             platform_integrity has incorrect built version ffc70466b42bf732051daec870903aaab5114363ceb0caf5c1ff54c5e49ea4be
```

**Additional questions**
- Operating system and version: Up-to-date Manjaro Linux running the 5.10.30 kernel.
- Have you tried rebooting? Yes and it didn't work.
- Is this a regression? Maybe, since `fwupd` was working on the same device before?

hughsie, 2021-04-20:
> 07:57:01:0377 FuEngine             platform_integrity has incorrect built version ffc70466b42bf732051daec870903aaab5114363ceb0caf5c1ff54c5e49ea4be

I think the package manager on your system failed to delete /usr/lib/fwupd-plugins-3/libfu_plugin_platform_integrity.so on upgrade. Deleting that file will fix the issue.

closed 2021-04-20 by hughsie

lgp171188, 2021-04-20:
@hughsie, thanks for the quick response. That worked! I will check and file a bug against my distro's package for this. 👍🏼