fwupd/fwupd #3168: user facing revocation versioning

cmurf, 2021-04-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Revocation versioning is ambiguous. For systems with old shim 15-8, the version is number of dbx entries, which today could be between 77 and 120; but once new shim 15.3 and higher are installed the version is 13. Probably because it's only taking dbx into account and not sbat generation.

**Describe the solution you'd like**
We need an unambiguous way of communicating a system's revocation state, i.e. is it current? And if not current how far off is it? And it should hopefully use a scheme that's comparable across distros, i.e. revocation epoch 2021-04 should mean the same things have been revoked, no matter the distribution.

As long as it's unambiguous I don't think it really matters if it starts at some arbitrary value or is date based. But I think the less it exposes users (and even developers) to whether a version bump is due to dbx and/or sbat generation changes, the better. That is, the cause of the versioning bump doesn't need to be reflected in the scheme.

**Describe alternatives you've considered**
None so far.

**Additional context**
I'm basing this on 'dbxtool -l' and 'fwupdmgr get-devices' For example the latter uses:

  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI Revocation Database
        Current version:  13
        Minimum Version:  13

a) in theory dbx isn't going to need to be bumped anymore but it could be
b) we need something that incorporates sbat generations being bumped

superm1, 2021-04-22:
Devil's advocate here - fwupd shouldn't be advertising sbat version unless there is a way that it can update it.  `dbx` is explicitly done because the updated dbx files can come from LVFS.

>a) in theory dbx isn't going to need to be bumped anymore but it could be

Linux isn't the only thing that can cause revocations.  The UEFI 3rd party cert is used to sign more than Linux.

>b) we need something that incorporates sbat generations being bumped

Until the updated SBAT variable is distributed outside of shim I don't know I agree.

hughsie, 2023-02-20:
I'll move this to a discussion as it's not really something we can do yet.