fwupd/fwupd #3171: Failure when updating Lenovo Thinkpad Thunderbolt 3 Dock

thescouser89, 2021-04-21:
**Describe the bug**
While trying to update the Lenovo Thinkpad Thunderbolt 3 Dock (Gen 2) via `fwupdmgr` it failed with:

```
~ ❯ sudo fwupdmgr update         
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • USB2.0 Hub
Devices with the latest available firmware version:
 • Thunderbolt host controller
Upgrade available for USB3.1 Hub from 51.13 to 51.63
USB3.1 Hub and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating USB3.1 Hub…     [***************************************]
Erasing…                 [***************************************]
failed to erase FW sector @0x11000: failed to read back empty: failed to read SPI data @0x11be0: transfer failed
```

Original comment: https://github.com/fwupd/firmware-lenovo-thinkpad/issues/5#issuecomment-823668208

@hughsie replied from my comment with this:
```
This is the VLI USB hub and not the MCU -- and shouldn't have damaged anything on failure.
```
https://github.com/fwupd/firmware-lenovo-thinkpad/issues/5#issuecomment-823926074

**Steps to Reproduce**
Run `sudo fwupdmgr update`

**Expected behavior**
The update succeeds

**fwupd version information**
Please provide the version of the daemon and client.
```shell
~ ❯ fwupdmgr --version
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.8
~ ❯ 

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
~ ❯ fwupdmgr get-devices --show-all-devices
20NYS7K90Z
│
├─Thunderbolt host controller:
│     Device ID:          ccf24df0101fff1f2117a00dbbd2e82d98060109
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              ea3c7ac8-d937-568a-aab5-e053bd6af9ce
│                         3d3f953a-3408-56fa-81f3-042d73c59c16
│                         e773c51e-a20c-5b29-9f09-6bb0e0ef7560
│                         f1793149-6c12-5a7f-95db-9fefd116748d
│     Device Flags:       • Internal device
[... 170 lines cut ...]
│                         6986c21a-0f3f-5538-bcef-fe4e8cda7ab7
│                         52aa8ac7-c837-561f-af32-b7979cd2f6ed
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─Unifying Receiver:
      Device ID:          2d687c9105d12706ab306d71285f4e3cdbed39aa
      Summary:            A miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.02_B0009
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                          279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:       • Updatable
                          • Supported on remote server
    
~ ❯ 
```

**Additional questions**
- Operating system and version: Fedora 33
- Have you tried rebooting? Yup
- Is this a regression? No, it's my first time trying to update the docking station.

hughsie, 2021-04-21:
@memily hey -- does this look like something you've seen before? I have the same hardware here but can't reproduce.

SherlockChu219, 2021-04-22:
We reckon this might be a hardware issue. Please test these two methods for us:
1. Insert a pen drive to the hub and try updating with the pen drive attached.
2. Repeat fwupd update several times and record whether or not the error repeats in the same address 0x11be0 in the SPI Flash. 

Thanks.

hughsie, 2021-04-22:
@thescouser89 would you be able to try the two things kindly requested from @SherlockChu219 please. Also, any more detail about the hardware would be useful, e.g. what Lenovo power supplies you have plugged in, what USB devices and monitors were in the dock at the time of the upgrade etc. Thanks!

thescouser89, 2021-04-22:
Will do!

With pen drive inserted:
```
~ ❯ fwupdmgr update                                                      ✘ 1 9s
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • USB2.0 Hub
Devices with the latest available firmware version:
 • Thunderbolt host controller
Upgrade available for USB3.1 Hub from 51.13 to 51.63
USB3.1 Hub and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating USB3.1 Hub…     [***************************************]
Erasing…                 [*****                                  ] Less than oneErasing…                 [********                               ] Less than oneErasing…                 [*****Erasing…                 [***************************************]…
failed to erase FW sector @0x15000: failed to read back empty: failed to read SPI data @0x159a0: transfer failed
```

I tried to put the pendrive both in front, and at the back if that matters.

The address sector is always different:
- failed to erase FW sector @0x10000: failed to read back empty: failed to read SPI data @0x104e0: transfer failed
- failed to erase FW sector @0x15000: failed to read back empty: failed to read SPI data @0x159a0: transfer failed
- failed to erase FW sector @0x13000: ->spi_write_status failed: failed to write SPI status 0x0: transfer failed
- failed to erase FW sector @0x12000: failed to read back empty: failed to read SPI data @0x12560: transfer failed
- failed to erase FW sector @0x11000: failed to read back empty: failed to read SPI data @0x11720: transfer failed

I'll update my initial description with the more detailed hardware info @hughsie .

Thanks guys for your help in figuring out the issue

hughsie, 2021-04-22:
> transfer failed

Given all the address are pretty random, and we've even seen a write failure, I wonder if we need to retry some part of the control transfer on error? e.g.

	return g_usb_device_control_transfer (fu_usb_device_get_dev (FU_USB_DEVICE (self)),
					      G_USB_DEVICE_DIRECTION_DEVICE_TO_HOST,
					      G_USB_DEVICE_REQUEST_TYPE_VENDOR,
					      G_USB_DEVICE_RECIPIENT_DEVICE,
					      0xc4, value, index,
					      buf, bufsz, NULL,
					      FU_VLI_DEVICE_TIMEOUT,
					      NULL, error);

Or maybe we need to do the fu_vli_device_get_spi_cmd() command and then do the control transfer read again?

...or perhaps the timeout is too small? 3000ms should be more than enough, right?

SherlockChu219, 2021-04-23:
>I wonder if we need to retry some part of the control transfer on error? e.g.

We do not think the problem is software in nature. So we doubt retrying control transfer or adding a 3000ms timeout will address the issue because a read command only takes a few ms on U3. 

Given how the addresses are random, I assumed signal quality might be the problem...

If so then there are two types of quality fail:
1. USB signal connectivity issue related to the cable, receptor, or daisy chaining. 
2. SPI interface issue which on the PCB. 
 
We don't think it is possible to resolve this problem without contacting Lenovo customer service or having a Lenovo Repair center take a look at the device.

hughsie, 2021-04-23:
> USB signal connectivity issue related to the cable

@thescouser89 are you using the original cable supplied with the Dock? If so, could you try a different USB cable? It doesn't have to be a thunderbolt cable for the USB hub update.

> SPI interface issue which on the PCB

Plausible. Would the Windows dock upgrade tools show anything like the same level of detail on failure? I wondered if the reporter should try the windows tool and see if the same error is generated.

thescouser89, 2021-04-23:
@hughsie yup original cable.

I tried with another usb-c cable (most likely not thunderbolt) and I was able to update my audio firmware! I guess because it's not thunderbolt, I wasn't able to update the thunderbolt controller. I guess that's a cable issue then?

Unfortunately I won't get a hold of another usb-c thunderbolt cable until we're allowed back in the office. I wonder if Lenovo is willing to send me one for free... :)

Anyhow, I guess that answers why I was having issues. Thanks for the investigation!

hughsie, 2021-04-23:
>  I guess that's a cable issue then?

Well, it's certainly very odd but it might explain the failures. Can you reopen this if you see any more issues? Thanks.

closed 2021-04-23 by hughsie