fwupd/fwupd #3176: incorrectly reporting unencrypted swap

dagrh, 2021-04-22:
**Describe the bug**
fwupdmgr security --force   says 'Linux swap:                    Unencrypted'
but I'm pretty sure it is.
(Maybe it's getting confused by f34 zram??)

**Steps to Reproduce**
fwupdmgr security --force

[root@dgilbert-t580 dgilbert]# cat /proc/swaps 
Filename				Type		Size		Used		Priority
/dev/dm-2                               partition	16457724	0		-2
/dev/zram0                              partition	8388604		0		100

# dmsetup info
Name:              fedora_dgilbert--t580-swap
State:             ACTIVE
Read Ahead:        256
Tables present:    LIVE
Open count:        2
Event number:      0
Major, minor:      253, 2
Number of targets: 1
UUID: LVM-kP5olf5Yhcp9Eu1M6wB6qFzHPvqShe02SZVA4tiFRSu22eQuEfnSiCsCRwzdR2AR

[root@dgilbert-t580 dgilbert]# lsblk
NAME                                          MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
zram0                                         252:0    0     8G  0 disk  [SWAP]
nvme0n1                                       259:0    0 476.9G  0 disk  
├─nvme0n1p1                                   259:1    0   200M  0 part  /boot/efi
├─nvme0n1p2                                   259:2    0     1G  0 part  /boot
└─nvme0n1p3                                   259:3    0 475.7G  0 part  
  └─luks-db06c80d-dd03-4ca7-85e6-e7f848758ff6 253:0    0 475.7G  0 crypt 
    ├─fedora_dgilbert--t580-root              253:1    0   395G  0 lvm   /
    └─fedora_dgilbert--t580-swap              253:2    0  15.7G  0 lvm   [SWAP]

**Expected behavior**
it should be happy with my swap

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.9

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

fedora 34

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
[root@dgilbert-t580 dgilbert]# fwupdmgr get-devices --show-all-devices
20LAS3NJ0B
│
├─Thunderbolt host controller:
│     Device ID:          2de7d8754b9b5fdcc8c2c53b7b166dc0b8198739
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              d0b85dea-a21f-5d1b-a693-55a3cf1bca5e ← THUNDERBOLT\VEN_0109&DEV_1703&REV_00
│                         72021ca5-cd76-5b37-85ca-d39e7184bf77 ← THUNDERBOLT\VEN_0109&DEV_1703
│                         5bed8b5c-e5c2-547b-8806-6603930184ee ← TBT-01091703-native
│                         f69aca31-6f9e-5b23-93ec-9683b02d66eb ← TBT-01091703-native-controller0-0
[... 102 lines cut ...]
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    07
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              6e46bdd6-e93d-5f5a-a22f-9bfe138b3db5 ← PCI\VEN_8086&DEV_5917&SUBSYS_17AA225A&REV_07
                          6abd5dd1-9f3b-5d12-971c-01eeedc01e1a ← PCI\VEN_8086&DEV_5917&SUBSYS_17AA225A
                          fed2efa4-7045-55a1-b4fc-b29283d59fe5 ← PCI\VEN_8086&DEV_5917&REV_07
                          8b72a10c-1279-5f8e-a28a-34fb11a58240 ← PCI\VEN_8086&DEV_5917
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
    
________________________________________________

Devices that have been updated successfully:

 • Thunderbolt Controller (14.00 → 20.00)
 • System Firmware (0.1.18 → 0.1.21)
 • UEFI Device Firmware (184.55.3510 → 184.70.3626)

**Additional questions**
- Operating system and version:
Fedora 34
- Have you tried rebooting?
No
- Is this a regression?
Unknown

hughsie, 2021-04-22:
TBH, I didn't know you could do swap on LVM.

dagrh, 2021-04-22:
I think swap-on-LVM is what you get by default if you ask for an encrypted disk with fedora (I think!).
Anyway, it's an easy setup so I think it's a common case.

PhilippWendler, 2021-12-16:
I also have swap on LVM on encrypted physical volume, because this is the default setup of Ubuntu if you choose an encrypted disk, and fwupd reports this as unencrypted.

The bot closed this issue because of inactivity, was this really desired? I think recognizing this situation is a legitimate request and I would be glad to see this implemented (or at least changed to report "unknown" if swap is on LVM or so).

hughsie, 2021-12-16:
> swap on LVM on encrypted physical volume

Could you add support for detecting that to https://github.com/fwupd/fwupd/blob/main/plugins/linux-swap/fu-linux-swap.c ?

PhilippWendler, 2022-04-19:
This bug still exists, can this be reopened?

Side note: I find it weird that a bot seems to make the decision which problem won't be fixed. Shouldn't such decisions be made by a person? As a user or bug reporter having an issue closed with "wontfix" by a bot simply because nothing happened feels quite disappointing and certainly does not invite further bug reports / feature requests / etc.

hughsie, 2022-04-19:
> I find it weird that a bot seems to make the decision which problem won't be fixed. 

We're just a few people who contribute to a shared project, it's impossible for us to fix *every* bug with such limited resources. We need people to submit pull requests with actual code for an open source project to scale effectively. If this is important to you for $WORK then please talk with your Red Hat technical account manager and file a bug in Red Hat bugzilla so it can be prioritized accordingly. I'm sure you could do the same for Canonical or SUSE, or even hire someone from Collabora to write the code if you're unable to do so yourself. Making us feel bad on issues isn't the way to get this functionality added.

PhilippWendler, 2022-04-19:
I am not wanting to make anyone feel bad about an open issue, I am very sorry if this was caused by my comment. I also do not want to argue about the importance of this issue.

I just wanted to point out that a *bot* marking stuff as wontfix automatically may come across as negative for users.

hughsie, 2022-04-19:
What about something like this:

	 markComment: >
	   This issue has been automatically marked as stale because it has not had
	-  recent activity. It will be closed if no further activity occurs. Thank you
	-  for your contributions.
	+  recent activity. It will be closed if no further activity occurs.
	+  Please note: We are just a few people who contribute to a shared project,
	+  it's impossible for us to fix every bug with such limited resources.
	+  If this is important to you for your business then please talk with your
	+  Red Hat, Canonical or SUSE technical account manager and file a bug in
	+  bugzilla so it can be prioritized accordingly.
	+  You might even consider hiring someone from Collabora to write the code
	+  if you're unable to do so yourself.
	+

superm1, 2022-04-19:
Something like that yes, open a PR and I will give some word smithing advice.

closed 2022-07-31 by stale[bot]