fwupd/fwupd #3177: What is the difference between --offline and --no-reboot-check?

openjck, 2021-04-23:
**Describe the question**
Reading the documentation, I'm not sure I understand the difference between `fwupdmgr --offline` and `fwupdmgr --no-reboot-check`.

```
       --offline
              Schedule installation for next reboot when possible
       --no-reboot-check
              Do not check for reboot after update
```

Is the following table correct?

| Flag | Installs changes immediately | Installs changes at reboot | Prompts user to reboot |
| --- | --- | --- | --- |
| [none] | :heavy_check_mark: | :x: | :heavy_check_mark: |
| `--offline` | :x: | :heavy_check_mark: | :x: |
| `--no-reboot-check` | :heavy_check_mark: | :x: | :x: |

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
```shell
$ fwupdmgr --version
client version: 1.4.5
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.5
```

openjck, 2021-04-23:
If my table is correct, perhaps it would be more clear to have separate `--install-at-reboot` and `--no-prompt` options.

hughsie, 2021-04-23:
` --no-reboot-check` just disables the question about `Do you want to update now [Y|n]` after the update has completed -- so I think your table is correct. We didn't do `install-at-reboot` as it's not always an install -- it could be a downgrade or reinstall. `no-reboot-prompt would certainly be clearer` I guess

openjck, 2021-04-23:
I think that would be clearer. Alternatively, or in addition, perhaps the documentation could also include the word *prompt*, e.g., "Do not check for or prompt for reboot after update."

Also, would `--apply-at-reboot` work as an alternative for `--offline`?

hughsie, 2021-04-23:
> Do not check for or prompt for reboot after update

I think that's a great idea - could you perhaps do a pull request for that change and we can merge it before the 1.6.0 release next week?

> Also, would --apply-at-reboot work as an alternative for --offline?

I think I'd like to keep the `--` arguments as-is unless there's an exceptional reason for changing them -- they're kinda like API.  I'm find if you want to fix up that description too tho.

openjck, 2021-04-23:
Sounds good. I'll submit a PR.

openjck, 2021-04-23:
Opened #3178. I think the description of `--offline` is fine, so I left that untouched, but I'd be happy to change it to something else that you suggest.

closed 2021-04-26 by hughsie