fwupd/fwupd #3179: How to update a device in DFU mode?

djcb, 2021-04-23:
I'm working on an fwupd plugin to upgrade microphone firmware using DFU, but I'm hitting an error when actually doing so; not filing a bug as it's non unlikely I'm doing something wrong...:

WIth @hughsie's generous help (#3042) I can get the device into DFU mode. The next step is flashing the firmware, but  `fwupdmgr` tells me:

```
13:32:37:0237 FuDeviceList         adding GUID bb846392-89d5-5312-885e-00ceed0ad1ea to device
13:32:37:0237 FuDeviceList         copying old vendor ID USB:0x58BE to new device
13:32:37:0237 FuDeviceList         copying old custom flags manifest-poll,no-fu-dfu-runtime to new device
13:32:37:0237 FuDeviceList         ::changed 23ec719b6aabc2d2dac5176c232f0da7a21881b0
13:32:37:0237 FuDeviceList         device came back, clearing flag
13:32:37:0240 FuDeviceList         devices in wait_removed: 1 -> 0
13:32:37:0241 FuDeviceList         waited for replug
13:32:37:0241 FuEngine             update -> FuDfuDevice:
DFU in FS Mode
  DeviceId:             23ec719b6aabc2d2dac5176c232f0da7a21881b0
  Guid:                 21e3f8e5-bdb9-52b6-b2ad-f1d61395ccc5
  Guid:                 97f8269f-e7de-50fb-86e5-340d63e32767 <- USB\VID_58BE&PID_0006&REV_0200
  Guid:                 b53c995e-6ca5-53b0-a2c1-72fa4c980ee0 <- USB\VID_58BE&PID_0006
  Guid:                 bb846392-89d5-5312-885e-00ceed0ad1ea
  Serial:               2055305E394E
  Summary:              Green Ninja USB Microphone
  Plugin:               dfu
  Protocol:             com.st.dfuse
  Flags:                updatable|registered|is-bootloader|self-recovery|add-counterpart-guids
  Vendor:               Acme
  VendorId:             USB:0x58BE
  Version:              2.0
  VersionFormat:        bcd
  Icon:                 drive-harddisk-usb
  Created:              2021-04-23
  PhysicalId:           usb:01:00:04
  Order:                0
  CustomFlags:          manifest-poll,no-fu-dfu-runtime
  PossiblePlugin:       green_ninja
  PossiblePlugin:       dfu
  InternalFlags:        replug-match-guid
  State:                dfuIDLE
  Status:               OK
  DoneUploadOrDownload: false
  ClaimedInterface:     false
  Version:              0x11a
  ForceVersion:         0x0
  RuntimePid:           0xffff
  RuntimeVid:           0xffff
  RuntimeRelease:       0xffff
  TransferSize:         0x400
  IfaceNumber:          0x0
  DnloadTimeout:        0x0
  TimeoutMs:            0x5dc
    AltSetting:         0x0
    AltIdx:             0x6
    AltName:            @Internal Flash   /0x08000000/03*016Ka,01*016Kg,01*064Kg,03*128Kg
    AltNameForDisplay:  Internal Flash
      Idx00:            Zone:0, Sec#:0, Addr:0x08000000, Size:0x4000, Caps:0x1 [R]
      Idx01:            Zone:0, Sec#:0, Addr:0x08004000, Size:0x4000, Caps:0x1 [R]
      Idx02:            Zone:0, Sec#:0, Addr:0x08008000, Size:0x4000, Caps:0x1 [R]
      Idx03:            Zone:0, Sec#:1, Addr:0x0800c000, Size:0x4000, Caps:0x7 [REW]
      Idx04:            Zone:0, Sec#:2, Addr:0x08010000, Size:0x10000, Caps:0x7 [REW]
      Idx05:            Zone:0, Sec#:3, Addr:0x08020000, Size:0x20000, Caps:0x7 [REW]
      Idx06:            Zone:0, Sec#:3, Addr:0x08040000, Size:0x20000, Caps:0x7 [REW]
      Idx07:            Zone:0, Sec#:3, Addr:0x08060000, Size:0x20000, Caps:0x7 [REW]

13:32:37:0241 FuPlugin             superclassed write_firmware(dfu)
13:32:37:0241 FuEngine             Emitting PropertyChanged('Status'='decompressing')
13:32:37:0241 FuMain               Emitting PropertyChanged('Status'='decompressing')
13:32:37:0241 FuDevice             installing onto 23ec719b6aabc2d2dac5176c232f0da7a21881b0:
<firmware>
  <size>0xeb76</size>
  <data size="0xeb76">`A</data>
</firmware>
```

^^^ side-note: this may just be cosmetic, but the firmware is binary and starts with 0x60 0x41 0x00,
hence the backtick-A; but perhaps the XML shouldn't include binary-data as-is?

```
13:32:37:0244 FuPluginDfu          no dnload-timeout, using default of 5ms
13:32:37:0244 FuPluginDfu          refreshed status=OK and state=dfuIDLE (dnload=5)
13:32:37:0244 FuPlugin             running superclassed update_attach(dfu)
13:32:37:0248 FuPluginDfu          no dnload-timeout, using default of 5ms
13:32:37:0248 FuPluginDfu          refreshed status=OK and state=dfuIDLE (dnload=5)
13:32:37:0248 FuEngine             Emitting PropertyChanged('Status'='device-restart')
13:32:37:0248 FuMain               Emitting PropertyChanged('Status'='device-restart')
13:32:37:0248 FuPluginDfu          Message:
13:32:37:0254 FuPluginDfu          failed to release interface 0: USB error on device 58be:0006 : No such device (it may have been disconnected) [-4]
13:32:37:0254 FuEngine             failed to attach device after failed update: cannot get device state: device was disconnected
```

^^^ and then we fail, as does the following
```
13:32:37:0255 FuEngine             cleanup -> FuDfuDevice:
[...]
```
with a similar error. Note that 58be:0006 _does_ exist when checking with `lsusb`. 

So some questions:
- any idea why I get the USB error? What can I do about that?
- before I can reflash, I need to erase two sectors of the flash memory (Idx04/Idx05); I _suspect_ that fwupd cannot figure this out automagically, so how could I do this? Should I override the `setup_prepare` vfunc?

Thanks in advance!

**fwupd version information**
git HEAD as per today, `ae690c57b` + local plugin code.

hughsie, 2021-04-23:
> 13:32:37:0248 FuEngine             Emitting PropertyChanged('Status'='device-restart')

So this means it's abandoned writing and fwupd is trying to get the device back to runtime mode so the user doesn't think it's "broken". I'm trying to work out why fwupd stopped the write process; this seemed to work fine:

> 13:32:37:0248 FuPluginDfu          refreshed status=OK and state=dfuIDLE (dnload=5)

If you put the device into DFU mode manually, and then run with `fwupdtool install-blob foo.dfu` do you get a useful error out? Note, if you want to send me a device that would be fine, and then it would also be possible to include your hardware in the device-tests we do before each major release. If that sounds a good idea email me and I'll send you my address. Random PCB is fine.

hughsie, 2021-04-23:
I noticed this the fu-dfu-no-runtime quirk, but it was my fault. When you rebase against something that contains https://github.com/fwupd/fwupd/pull/3180 you'll need to use the (old) new quirk name.

djcb, 2021-04-26:
Okay, thanks for the amazingly quick reply! 

I tried with `fwuptool` (while in DFU mode), but that gave me the same error; the quirk didn't seem to matter for this. Could it be that the writing fails (because I need to erase first), but somehow the problems isn't recognized (by fwupd) as such? I can try to do the erasing first - what would be the logical place to hook that up?

I'll check regarding the device, it sounds like a good idea!

hughsie, 2021-04-26:
> Could it be that the writing fails (because I need to erase first),

Which STM-capable chip are you using out of interest? In `fu_dfu_target_stm_download_element` we do a first pass working out what sectors need erasing, then a 2nd pass actually erasing them, then a thirs pass of writing. Given the debug output never gets to `FWUPD_STATUS_DEVICE_ERASE` I don't think it gets past the first stage. It might be worth putting some g_warnings("XXX foo")` in `fu-dfu-target-stm.c` and seeing where the write process gets to. I even wonder if you're getting out of `fu_dfu_device_download()` -- e.g. the ALT name isn't matching or something?

djcb, 2021-04-26:
Ah... that's helpful! 

A bit of `gdb` later, I see I'm erroring out of `fu_dfu_device_download`, a bit before that ^^, specifically in line 1605:  
```
   if (firmware_vid == 0xffff) {
```

So the `.bin` I have is _not_ a dfuse-formatted file (a quick `hexdump` confirms)... `fu_dfu_device_download` apparently expects that. Guess I can wrap that... with `contrib/add_dfu_header.py`? That gets me a bit further, and I now see:
```
<flags>has-checksum|has-vid-pid</flags>
  <data size="0xeb76">`A</data>
  <vendor>0xb58e</vendor>
  <product>0x5</product>
  <release>0x406</release>
  <dfu_version>0x100</dfu_version>
</firmware>
```
but, `dfu_version` is wrong for dfuse... should be 0x11a I think? Anyhow, this gets us all the way (in `fu_dfu_device_download`) to 
```
1657		if (images->len == 0) {
```
so indeed, seems no images are detected. It appears I need to generate a dfuse-format file from the `.bin` I have (the `.bin` works for some proprietary tool). Is there a tool that can do that?

hughsie, 2021-04-26:
> So the .bin I have is not a dfuse-formatted file

So, we're actually supposed to fall back to a binary file if we can't load the DfuSe file... Can you try wip/hughsie/dfuse-from-bin with your binary file? Of course, a dfuse file is better (as it can check VID/PID), so regardless maybe create myfirmware.builder.xml:

	<firmware gtype="FuDfuseFirmware">
	  <vendor>0xb58e</vendor>
	  <product>0x5</product>
	  <release>0x406</release>
	  <firmware>
	    <idx>0x0</idx>
	    <filename>/location/to/your/file.bin</filename>
	  </firmware>
	</firmware>

Then do `fwupdtool firmware-build myfirmware.builder.xml file.dfuse` and that should spit out a valid file.

djcb, 2021-04-26:
I tried `wip/hughsie/dfuse-from-bin` as well, but that didn't quite work with a raw binary file.
Looking at the change, it handles the case where there are zero images, but the error I was seeing was earlier, in line 1600 of that same function. `if (firmware_vid == 0xffff) {`.

Perhaps we need to enable `FU_TARGET_TRANSFER_FLAG_WILDCARD_VID` automatically in this case?

hughsie, 2021-04-26:
> Perhaps we need to enable FU_TARGET_TRANSFER_FLAG_WILDCARD_VID automatically in this case?

Agree. Added to the branch.

hughsie, 2021-04-26:
> I did need to add <addr>

Ahh yes, if you're not offset 0x0 then indeed. Sorry!

> usb 1-3: Not enough bandwidth for new device state

That smells like a firmware issue; not sure fwupd can do much about that :)

> No such device (it may have been disconnected)

We should probably hide that warning if the device just went away. Fixed in 87fd2260f8bdb86e8879687ec5ddd4d2d46c5085

djcb, 2021-04-26:
So with the updated `wip/hughsie/dfuse-from-bin` it _almost_ works; I needed to added
```
modified   plugins/dfu/fu-dfu-device.c
@@ -1595,6 +1595,7 @@ fu_dfu_device_download (FuDfuDevice *self,
 		firmware_pid = fu_dfu_firmware_get_pid (FU_DFU_FIRMWARE (firmware));
 	} else {
 		flags |= DFU_TARGET_TRANSFER_FLAG_WILDCARD_VID;
+		flags |= DFU_TARGET_TRANSFER_FLAG_WILDCARD_PID;
```

but then it does:
```
15:04:32:0622 FuPluginDfu          downloading chunk at 0x0000
15:04:32:0622 FuPluginDfu          fixing up firmware address from 0x0 to 0x8000000
```

so I guess I do need the:
```
<addr>0x08010000</addr>
```
No idea if that could be auto-detected?

hughsie, 2021-04-26:
> flags |= DFU_TARGET_TRANSFER_FLAG_WILDCARD_PID

Urgh, yes, we need that too of course. Will consume more coffee.

> 15:04:32:0622 FuPluginDfu          fixing up firmware address from 0x0 to 0x8000000

That's expected; we just find the first writeable sector as 0x0 isn't available.

> 0x08010000

so you're explicitly adding an extra 0x10000 to the base address? Are you trying to avoid a bootloader or something?

djcb, 2021-04-26:
Yeah, I think so, that would be hard to guess for `fwupd`.

hughsie, 2021-04-26:
> that would be hard to guess for fwupd

Yes, impossible to guess, but possible to _quirk_. Are you using something like the reference ST bootloader? e.g. we could add a quirk key for the start offset to avoid when matching a specific VID/PID.

closed 2021-06-09 by stale[bot]