fwupd/fwupd #3186: fwupd sometimes removes GRUB from the UEFI boot order

mgedmin, 2021-04-24:
**Describe the bug**

Sometimes I attempt to upgrade the system firmware on my Lenovo ThinkPad X390 with fwupd, and the upgrade doesn't happen, and GRUB gets removed from the boot order, and I end up booting into Windows.  This is likely not a fwupd bug, but I did discover a fwupd bug while trying to recover from the situation described above.

The bug: fwupdmgr replaced the boot entry pointing to grub with one pointing to fwupd itself.
(Originally filed at https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1925813)

**Steps to Reproduce**

- Boot GRUB from a USB drive, load the grub.cfg from the primary OS install
- Run `grub-install` to make sure GRUB is in your boot order list
- Run `fwupdmgr upgrade`
- Inspect boot variables with `efibootmgr`

When I did that I saw that the regular grub boot entry was replaced by a second Linux-Firmware-Updater entry.

I have three `efibootmgr -v` dumps made between each step, but they don't fit neatly into this bug report template here.  You can find them in [the linked Ubuntu bug](https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1925813), as well as the full output of `fwupdmgr upgrade`.

**Expected behavior**

The boot order contains grub, BootNext points to Linux-Firmware-Updater.

**fwupd version information**

1.4.5-1 from Ubuntu 20.10.

I cannot run `fwupdmgr --version` because I've upgraded to Ubuntu 21.04 since encountering this bug.

**fwupd device information**

```shell
$ fwupdmgr get-devices --show-all-devices
20Q0CTO1WW
│
├─Thunderbolt host controller:
│     Device ID:          8685856be84028011ec101dbce36216f842bd287
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              ea3c7ac8-d937-568a-aab5-e053bd6af9ce
│                         3d3f953a-3408-56fa-81f3-042d73c59c16
│                         e773c51e-a20c-5b29-9f09-6bb0e0ef7560
│                         f1793149-6c12-5a7f-95db-9fefd116748d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          09ce7908b2d90a3b53473c3f328410590a527f29
│     Current version:    0.1.15
│     Minimum Version:    0.1.15
│     Vendor:             DMI:LENOVO
│     GUIDs:              be9a5f93-930b-40bf-8864-815268ea4ea3
│                         98129d04-2653-5233-89f1-8aa461e54a25
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          07edea1cf12d20814225b44de22b33758db1a1c8
│     Current version:    192.72.1757
│     Minimum Version:    192.72.1757
│     Vendor:             DMI:LENOVO
│     GUIDs:              ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
│                         056f0b06-1562-50f4-aea6-0d52f6fd4be5
[... 690 lines cut ...]
│       └── zstd.mod
├── initrd.img -> initrd.img-5.11.0-16-generic
├── initrd.img-5.11.0-16-generic
├── initrd.img-5.8.0-50-generic
├── initrd.img.old -> initrd.img-5.8.0-50-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.11.0-16-generic
├── System.map-5.8.0-50-generic
├── vmlinuz -> vmlinuz-5.11.0-16-generic
├── vmlinuz-5.11.0-16-generic
├── vmlinuz-5.8.0-50-generic
└── vmlinuz.old -> vmlinuz-5.8.0-50-generic

56 directories, 458 files
```

**Additional questions**
- Operating system and version: Ubuntu 20.04
- Have you tried rebooting? um, yes? after I fixed the boot order with `grub-install /dev/nvme0n1`
- Is this a regression? can't say, booting GRUB from USB is a rare occurrence
- Are you using an NVMe disk? yes
- Is secure boot enabled? yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? It is a Lenovo system but Boot Order Lock is turned off in the BIOS, I noticed this setting while I was trying to restore the ubuntu entry in the BIOS setup.

hughsie, 2021-04-26:
@mgedmin would it be possible to test out with fwupd 1.5.x or 1.6.0? Then we can add some more detained logging to try to debug what's happening. My gut instinct says "firmware bug" but I have no proof for that yet! :)

superm1, 2021-04-26:
I think that trying it with self built will change things too significantly - he reproduced it with secure boot on and would need to turn it off for a hand built.

I'm with you though, it looks like firmware bug to me. The efibootmgr output is enough to prove that isn't it?

superm1, 2021-04-26:
To add more color to that - look at what is getting booted into. It's not even the Ubuntu entry, it's the autogenerated entry by the ibv boot manager.

mgedmin, 2021-04-27:
How is this a firmware bug?  Running `fwupdmgr upgrade` changed Boot0001 from

> Boot0001* ubuntu HD(1,GPT,efd54dec-f70c-4b32-9480-c2dd0043ea0d,0x800,0x82000)/File(\EFI\ubuntu\shimx64.efi)

to

> Boot0001* Linux-Firmware-Updater HD(1,GPT,efd54dec-f70c-4b32-9480-c2dd0043ea0d,0x800,0x82000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...

Where does firmware get a chance to do anything?

> look at what is getting booted into. It's not even the Ubuntu entry, it's the autogenerated entry by the ibv boot manager.

If you mean the `BootCurrent: 001E`, that was because I booted from a removable USB drive (because my `ubuntu` boot entry was gone).  I did restore Boot0001 by running `grub-install` before I ran `fwupdmgr upgrade`, which promptly overwrote it.

I get the feeling I'm not describing the bug well enough.  I'm sorry :(

hughsie, 2021-04-27:
@mgedmin okay, perhaps two bugs indeed. Can you compile git master fwupd [ https://lvfs.readthedocs.io/en/latest/custom-plugin.html#building-fwupd ], and then can you run the new instance with: `./src/fwupd --verbose` and then in a different terminal then do `./src/fwupdmgr reinstall" or something that will trigger the firmware update action to jump to life and overwrite the Boot0001 entry. We then need all the debug output since you triggered the update, probably starting around the `updating existing boot entry` line. Thanks!

superm1, 2021-04-27:
Ah I thought that meant it was after the reboot. It's literally right after calling update? Then yeah it seems like two bugs I agree.

mgedmin, 2021-04-28:
> Ah I thought that meant it was after the reboot. It's literally right after calling update?

Precisely!  The sequence of commands was (from my notes)

```
efibootmgr -v
grub-install /dev/nvme0n1
efibootmgr -v           # <-- here I have Boot0001: ubuntu, Boot0002: Linux-Firmware-Updater
fwupdmgr upgrade
tree /boot/efi/
ls -l /boot/efi/EFI/ubuntu/fw
efibootmgr -v           # <-- here both Boot0001 and Boot0002 are Linux-Firmware-Updater
grub-install /dev/nvme0n1
efibootmgr -v           # <-- now I have Boot001 and Boot0002: Linux-Firmware-Updater, Boot0003 is ubuntu, BootNext is 0001
reboot
```

I have the outputs of each command recorded (excepting grub-install which says nothing very interesting, and for the tree I only recorded the /boot/efi/EFI/ubuntu subtree) in my notes, if that would be helpful.  It's mostly very verbose and I think I already included all the relevant information earlier.

mgedmin, 2021-04-28:
I wondered if this bug was fixed by 59e375707aab4c3e95934ef88b81becee9931d42, but Ubuntu's fwupd 1.4.5-1 already contains that commit.

My guess that BootCurrent could've had any relevance seems entirely wrong: the fwupd source tree doesn't even mention BootCurrent once.

hughsie, 2021-04-28:
Ya, `fwupd --verbose` would give us the information needed to debug this further.

dkwo, 2021-05-27:
Same issue here.

dkwo, 2021-06-28:
How can this be debugged further? I have no fwupd command on my system.. do you mean fwupdmgr?

frittentheke, 2021-07-30:
Is this solved, unreproducable or just abandoned?

hughsie, 2021-07-30:
We need the `fwupd --verbose` log -- the actual fwupd binary is usually in /usr/libexec/fwupd, although that differs between distributions.

dkwo, 2021-08-18:
@hughsie 
I have `/usr/libexec/fwupd/fwupd`
Just to make sure: what does this perform?

superm1, 2021-08-18:
That's the daemon, and we need the verbose logs from it. Launch manually adding --verbose to the command line or modify the systemd unit to do the same.

dkwo, 2021-08-19:
This is my log (I terminated it with ctrl-c):
```
$ doas /usr/libexec/fwupd/fwupd --verbose
10:45:25:0816 FuDebug              Verbose debugging enabled (on console 1)
10:45:25:0816 FuConfig             loading config values from /etc/fwupd/daemon.conf
10:45:25:0816 FuConfig             using autodetected max archive size 18.4 EB
10:45:25:0817 FuRemoteList         loading /usr/share/fwupd/metainfo
10:45:25:0817 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
10:45:25:0817 XbSilo               file: 2c16f58b-992c-1737-0f4d-319652253150, current:2c16f58b-992c-1737-0f4d-319652253150, cached: (null)
10:45:25:0817 XbSilo               loading silo with file contents
10:45:25:0817 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
10:45:25:0817 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
10:45:25:0818 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
10:45:25:0818 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
10:45:25:0818 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
10:45:25:0818 FuRemoteList         ordering lvfs-testing=lvfs+1
10:45:25:0818 FuRemoteList         ignoring unfound remote fwupd
10:45:25:0818 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
10:45:25:0818 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
10:45:25:0818 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
10:45:25:0818 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
10:45:25:0818 Jcat                 loaded 1 certificates
10:45:25:0818 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
10:45:25:0818 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
10:45:25:0818 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
10:45:25:0818 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
10:45:25:0819 Jcat                 loaded 1 certificates
10:45:25:0819 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
10:45:25:0819 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
10:45:25:0825 FuEngine             client certificate exists and working
10:45:25:0826 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
10:45:25:0826 FuHistory            got schema version of 6
10:45:25:0826 FuIdle               setting timeout to 7200s
10:45:25:0826 FuHwids              smbios property Manufacturer=LENOVO
10:45:25:0826 FuHwids              smbios property EnclosureKind=a
10:45:25:0826 FuHwids              smbios property Family=ThinkPad T460s
10:45:25:0826 FuHwids              smbios property ProductName=20F9CTO1WW
10:45:25:0826 FuHwids              smbios property ProductSku=LENOVO_MT_20F9_BU_Think_FM_ThinkPad T460s
10:45:25:0826 FuHwids              smbios property BiosVendor=LENOVO
10:45:25:0826 FuHwids              smbios property BiosVersion=N1CET83W (1.51 )
10:45:25:0826 FuHwids              smbios property BiosMajorRelease=01
10:45:25:0826 FuHwids              smbios property BiosMinorRelease=33
10:45:25:0826 FuHwids              smbios property BaseboardManufacturer=LENOVO
10:45:25:0826 FuHwids              smbios property BiosMinorRelease=33                                                                               [272/1066]
10:45:25:0826 FuHwids              smbios property BaseboardManufacturer=LENOVO
10:45:25:0826 FuHwids              smbios property BaseboardProduct=20F9CTO1WW
10:45:25:0827 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
10:45:25:0827 XbSilo               file: aaab8fe5-5703-32ca-4a66-30f5280ae164, current:aaab8fe5-5703-32ca-4a66-30f5280ae164, cached: (null)
10:45:25:0827 XbSilo               loading silo with file contents
10:45:25:0828 FuEngine             1071 components now in silo
10:45:25:0832 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_pci_bcr.so)
10:45:25:0832 FuPlugin             added udev subsystem watch of pci
10:45:25:0832 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_colorhug.so)
10:45:25:0832 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_vli.so)
10:45:25:0832 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_acpi_dmar.so)
10:45:25:0832 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_amt.so)
10:45:25:0833 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_superio.so)
10:45:25:0833 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_iommu.so)
10:45:25:0833 FuPlugin             added udev subsystem watch of iommu
10:45:25:0833 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_hailuck.so)
10:45:25:0833 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_ep963x.so)
10:45:25:0833 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_linux_tainted.so)
10:45:25:0833 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_rts54hid.so)
10:45:25:0833 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_bios.so)
10:45:25:0835 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_tpm.so)
10:45:25:0835 FuPlugin             added udev subsystem watch of tpm
[... cut ...]
10:45:26:0720 FuEngine             ignoring 0.1.35 < 0.1.51
10:45:26:0720 FuEngine             ignoring 0.1.33 < 0.1.51
10:45:26:0731 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
10:45:26:0731 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
10:45:26:0744 FuMain               Daemon ready for requests (locale (null))
10:45:26:0746 FuMain               acquired name: org.freedesktop.fwupd
^CInterrupt
```

dkwo, 2021-08-19:
I'm on Void Linux
```
$ xuname
Void 5.13.12_1 x86_64-musl GenuineIntel uptodate rrnFFFFFF
```
Can provide more details if needed.

dkwo, 2021-09-20:
This is with
```
$ fwupdmgr --version
client version: 1.6.3
compile-time dependency versions
        gusb:   0.3.7

daemon version: 1.6.3
$ xuname
Void 5.13.19_1 x86_64-musl GenuineIntel uptodate rrnFFFFFF
```
Just to keep the bug from going stale, from newer version:
```
$ doas /usr/libexec/fwupd/fwupd --verbose
07:44:42:0213 FuDebug              Verbose debugging enabled (on console 1)
07:44:42:0214 FuConfig             loading config values from /etc/fwupd/daemon.conf
07:44:42:0214 FuConfig             using autodetected max archive size 18.4 EB
07:44:42:0214 FuRemoteList         loading /usr/share/fwupd/metainfo
07:44:42:0214 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
07:44:42:0214 XbSilo               file: 4f7eaefe-ce90-0f7e-b3f6-918b406e207f, current:4f7eaefe-ce90-0f7e-b3f6-918b406e207f, cached: (null)
07:44:42:0214 XbSilo               loading silo with file contents
07:44:42:0214 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
07:44:42:0215 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
07:44:42:0215 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
07:44:42:0215 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
07:44:42:0215 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
07:44:42:0215 FuRemoteList         ordering lvfs-testing=lvfs+1
07:44:42:0215 FuRemoteList         ignoring unfound remote fwupd
07:44:42:0215 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
07:44:42:0215 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
07:44:42:0215 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
07:44:42:0215 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
07:44:42:0215 Jcat                 loaded 1 certificates
07:44:42:0215 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
07:44:42:0215 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
07:44:42:0215 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
07:44:42:0215 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
07:44:42:0216 Jcat                 loaded 1 certificates
07:44:42:0216 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
07:44:42:0216 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
07:44:42:0223 FuEngine             client certificate exists and working
07:44:42:0223 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
07:44:42:0223 FuHistory            got schema version of 6
07:44:42:0223 FuIdle               setting timeout to 7200s
07:44:42:0223 FuSmbios             trying to read /sys/class/dmi/id
07:44:42:0224 FuHwids              smbios property Manufacturer=LENOVO
07:44:42:0224 FuHwids              smbios property EnclosureKind=a
07:44:42:0224 FuHwids              smbios property Family=ThinkPad T460s
07:44:42:0224 FuHwids              smbios property ProductName=20F9CTO1WW
07:44:42:0224 FuHwids              smbios property ProductSku=LENOVO_MT_20F9_BU_Think_FM_ThinkPad T460s
07:44:42:0224 FuHwids              smbios property BiosVendor=LENOVO
07:44:42:0224 FuHwids              smbios property BiosVersion=N1CET83W (1.51 )
07:44:42:0224 FuHwids              smbios property BiosMajorRelease=01
07:44:42:0224 FuHwids              smbios property BiosMinorRelease=33
07:44:42:0224 FuHwids              smbios property FirmwareMajorRelease=01
07:44:42:0224 FuHwids              smbios property FirmwareMinorRelease=0e
07:44:42:0224 FuHwids              smbios property BaseboardManufacturer=LENOVO
07:44:42:0224 FuHwids              smbios property BaseboardProduct=20F9CTO1WW
07:44:42:0224 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
07:44:42:0224 XbSilo               file: b224b834-57a1-a2e3-b35c-d12de88c462a, current:b224b834-57a1-a2e3-b35c-d12de88c462a, cached: (null)
07:44:42:0224 XbSilo               loading silo with file contents
07:44:42:0225 FuEngine             1115 components now in silo
07:44:42:0231 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_pci_bcr.so)
07:44:42:0231 FuContext            added udev subsystem watch of pci
07:44:42:0232 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_colorhug.so)
07:44:42:0231 FuContext            added udev subsystem watch of pci                                                                                                                                                                                                    [365/754]
07:44:42:0232 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_colorhug.so)
07:44:42:0232 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_vli.so)
07:44:42:0232 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_acpi_dmar.so)
07:44:42:0232 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_amt.so)
[... 420 lines cut ...]
07:44:43:0500 FuEngine             ignoring 0.1.51 == 0.1.51
07:44:43:0500 FuEngine             ignoring 0.1.50 < 0.1.51
07:44:43:0500 FuEngine             ignoring 0.1.49 < 0.1.51
07:44:43:0500 FuEngine             ignoring 0.1.47 < 0.1.51
[... cut ...]
07:44:43:0500 XbSilo               ignoring for OR statement: indexed string '8eba3da0-2998-58bf-a7d7-210c15b3a3a8' was unfound
07:44:43:0512 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
07:44:43:0512 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
07:44:43:0526 FuMain               Daemon ready for requests (locale (null))
07:44:43:0528 FuMain               acquired name: org.freedesktop.fwupd
^CInterrupt
```

frittentheke, 2021-11-10:
I don't think this is stale....

hughsie, 2022-01-21:
@dkwo we need the daemon log when you've actually scheduled the update (using fwupdmgr) -- not just the startup phase. Thanks.

dkwo, 2022-01-21:
Do you mean: start logging, schedule update, stop logging ?

There's a new update for my firmware,
so I can do that in the next couple of days.

dkwo, 2022-01-22:
@hughsie Done, here's the log: https://paste.c-net.org/BeatriceDelay
Hope it helps.

dkwo, 2022-01-24:
Here it is:
```
doas efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C
Boot0000* GRUB  HD(1,GPT,4a04d763-6e6c-4f50-a11b-c5d1092ccee5,0x800,0x100000)/File(\EFI\GRUB\grubx64.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D* IDER BOOT CDROM       PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot001E* IDER BOOT Floppy      PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot001F* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0020* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

dkwo, 2022-01-24:
(Notice I had to chroot and reinstall grub after the firmware update, with
`grub-install --target=x86_64-efi --efi-directory=/boot --bootloader-id="GRUB"`
otherwise nothing  to boot from..)

dkwo, 2022-03-21:
@hughsie Could it be that this issue depends on the details of how grub is installed, whether with efi-directory=/boot or /boot/efi?
I updated firmware on a Dell machine yesterday, with similar setup, and it did not wipe out grub, which was installed to /boot/efi, while the lenovo laptop where the issue arises uses /boot. Not sure if relevant.

hughsie, 2022-03-21:
> efi-objcopy

hughsie, 2022-03-21:
> Could it be that this issue depends on the details of how grub is installed

I don't think so.