fwupd/fwupd #3187: fwupd automatically claims some USB devices and attempts bulk transfers, thus preventing programs like fastboot from working

sandy-8925, 2021-04-25:
**Describe the bug**
In recent versions of fwupd (probably February/March onwards), fwupd automatically connects to USB devices and seems to attempt USB bulk transfers. However, this fails for some devices (e.g Android device in fastboot mode), and this prevents other programs like fastboot from working.

**Steps to Reproduce**
- Run fwupd
- After a few seconds, attempt to use fastboot to flash an image to an Android device partition (e.g fastboot flash boot boot.img)
- Following error is observed in journalctl logs: 
`fwupd[2248]: 21:08:27:0117 FuEngine             failed to add device usb:03:00:04: failed to do bulk transfer: USB error on device 8087:0fff : Input/Output Error [-1]`

**Expected behavior**
Fastboot should succeed, because fwupd shouldn't try to connect to random devices and leave them hanging.

**fwupd version information**
client version: 1.5.9
compile-time dependency versions
        gusb:   0.3.6

daemon version: 1.5.9

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`pacman`

**Additional questions**
- Operating system and version: Arch Linux, latest software
- Have you tried rebooting? Yes

hughsie, 2021-04-26:
> failed to do bulk transfer

This comes from `plugins/fastboot/fu-fastboot-device.c` -- which certainly does `g_usb_device_release_interface()` on `->close`.

> fwupd shouldn't try to connect to random devices

We're explictly connecting to fastboot devices specified with `USB\CLASS_FF&SUBCLASS_42&PROT_03`

>  and leave them hanging

When you say that `fastboot is stuck in disk sleep mode` what does that mean? The device is in deep sleep (as in no USB interface) or is fwupd not disconnecting from fastboot properly? We do query the device as soon as it appears in fastboot mode, but only the common stuff like `product`, `serialno` etc.

sandy-8925, 2021-04-26:
fwupd is not disconnecting from fastboot (on the device) properly, and when I attempt to run a fastboot command from my computer, the fastboot process on my computer was stuck in disk sleep mode, because it was waiting for fwupd to relinquish it's claim on the device.

This is very frustrating, because fastboot commands just failed with no clue as to what was going on.

Is there a way to prevent fwupd from automatically scanning USB devices?

hughsie, 2021-04-26:
> because it was waiting for fwupd to relinquish it's claim on the device

What is the fastboot CLI waiting for? We release the libusb handle as part of `->close()`

> Is there a way to prevent fwupd from automatically scanning USB devices?

Sure, you can use `DisabledPlugins` in `/etc/fwupd/daemon.conf` but that's just a workaround. It's better to fix the actual issue!

sandy-8925, 2021-04-26:
> What is the fastboot CLI waiting for? We release the libusb handle as part of `->close()`
> 

Well that didn't happen here. fwupd consistently fails to release the libusb handle. I was only able to execute fastboot commands after killing the fwupd process.

hughsie, 2021-04-26:
> USB error on device 8087:0fff : Input/Output Error

Maybe this is the root cause? e.g. the device gets confused, or maybe fwupd doesn't `->close` or something bazaar. Could you add some debugging in `fu_fastboot_device_setup` and see if it's one of the commands that's causing fwupd to lock up? Thanks.

sandy-8925, 2021-04-26:
Sure, I'll do that later and report back.

sandy-8925, 2021-04-30:
I debugged, and `fu_fastboot_device_close()` never gets called. Will try to figure out why.

sandy-8925, 2021-04-30:
I wonder if commit 2e1245728f47c4342cce53ef39388f027ed4dea6 has something to do with it? It's the only change that looks relevant, although I could be wrong.

hughsie, 2021-05-01:
@sandy-8925 does ^^^ work? I'm amazed we've never hit this kind of failure before.

sandy-8925, 2021-05-06:
Unfortunately, it didn't work, the bug still persists.

hughsie, 2021-05-06:
@sandy-8925 does ->close() not get called on setup() failure? Did you `ninja install`?

sandy-8925, 2021-05-06:
No, close() doesn't get called. Yes, I made sure the code with the patch applied, was installed.

hughsie, 2021-05-06:
Can you add some debugging and see why self->close_func isn't being called? I've just put my phone (Google Pixel 3a) into fastboot mode, and added some patches like:

    @@ -648,6 +658,8 @@ fu_fastboot_device_close (FuDevice *device, GError **error)
            FuFastbootDevice *self = FU_FASTBOOT_DEVICE (device);
            GUsbDevice *usb_device = fu_usb_device_get_dev (FU_USB_DEVICE (device));
     
    +       g_warning ("CLOSE");

And with that PR I get:

	Loading…                 [-                                      ]15:49:04:0545 FuEngine             using plugins: fastboot
	15:49:04:0551 FuPluginFastboot     PROBE
	15:49:04:0551 FuDevice             using 983c3cffc6fd36d32b00b62928d30721eaeb93db for usb:03:00:01
	15:49:04:0551 FuPluginFastboot     OPEN
	15:49:05:0180 FuPluginFastboot     CLAIM
	15:49:05:0180 FuPluginFastboot     SETUP
	Loading…                 [  -                                    ]15:49:05:0230 FuDevice             983c3cffc6fd36d32b00b62928d30721eaeb93db device overwriting name value: Pixel 3a->Fastboot sargo
	Loading…                 [   -                                   ]15:49:05:0300 FuPluginFastboot     CLOSE
	15:49:05:0300 FuEngine             failed to add device usb:03:00:01: failed to getvar version: failed to read response: variable not found
	15:49:05:0336 FuEngine             Emitting PropertyChanged('Status'='idle')
	[hughsie@hughsie-work build (wip/hughsie/close-on-aborted-open *%)]$ sudo fastboot getvar product
	product: sargo
	Finished. Total time: 0.017s

closed 2021-05-12 by hughsie

aretw0, 2021-11-16:
Hi everyone,

Unfortunately, the same problem already happens here:

```
client version:	1.2.14
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.14
```

**Is this was fixed?
How can I update to the version that was fixed?**

I have to admit that is very sad the fact that the company I'm recently working for was just avoiding this the whole time by just killing the fwupd service and nothing was done to report and later fix it.

In addition, thanks for all the efforts done here.

superm1, 2021-11-16:
Sadly that fwupd version is quite ancient.  I think you'll need to upgrade to a newer fwupd version to get the proper solution for this on your system.  If that's not possible you can block the plugin from loading in `/etc/fwupd/daemon.conf` in your version (see this key: https://github.com/fwupd/fwupd/blob/1_2_X/data/daemon.conf#L9)

aretw0, 2021-11-17:
> Sadly that fwupd version is quite ancient. I think you'll need to upgrade to a newer fwupd version to get the proper solution for this on your system. If that's not possible you can block the plugin [...]

How can I upgrade? Is this something an `apt update / upgrade` should solve?

I'm on ubuntu 18, however, I'm seeing this issue on ubuntu 20 too with:
```
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.11
```
Anyway, how exactly is the plugin name to put on the conf file you mentioned?
Thanks in advance.

superm1, 2021-11-17:
There isn't a 1.5.12 release yet, which is the release in the 1.5.x series that would fix it. 1.6.1 in the 1.6.x series or later would also fix it.

Assuming it's the exact same failure (which I can't confirm since you didn't post logs) - for now you can put the plugin name "fastboot" in `/etc/fwupd/daemon.conf`.  If that doesn't work, I suggest you open a new issue with all the details requested in the template because this may be a different issue.

aretw0, 2021-11-17:
Sorry for not providing logs.
The original discussion here fits exactly with the behavior I have been facing.
I'll try that thanks!

aretw0, 2021-11-17:
So, there is no fastboot plugin here :(
```
nov 17 12:13:56 pinturas-ubu fwupd[19880]: 15:13:56:0380 FuEngine             failed to find specified plugin fastboot: no plugin fastboot found
```

superm1, 2021-11-17:
Then we need your logs in your own issue.