fwupd/fwupd #3190: fwupd crash on boot constantly

knstn-project, 2021-04-26:
**Describe the bug**
fwupd crash on boot constantly after the last update of it with Fedora 33

**Steps to Reproduce**
After a `dnf upgrade` those packages were installed:
    
```
Upgrade  fwupd-1.5.9-1.fc33.x86_64                                        @updates
Upgraded fwupd-1.5.8-1.fc33.x86_64                                        @@System
Upgrade  fwupd-plugin-flashrom-1.5.9-1.fc33.x86_64                        @updates
Upgraded fwupd-plugin-flashrom-1.5.8-1.fc33.x86_64                        @@System
Upgrade  fwupd-plugin-modem-manager-1.5.9-1.fc33.x86_64                   @updates
Upgraded fwupd-plugin-modem-manager-1.5.8-1.fc33.x86_64                   @@System
Upgrade  fwupd-plugin-uefi-capsule-data-1.5.9-1.fc33.x86_64               @updates
Upgraded fwupd-plugin-uefi-capsule-data-1.5.8-1.fc33.x86_64               @@System
```

Simply reboot and check journalctl and abrt.

**Expected behavior**
Not crashing

**fwupd version information**
fwupd-1.5.9-1.fc33.x86_64

```shell
$ fwupdmgr --version
client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.9

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Using dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
System Product Name
│
├─400 Series Chipset SATA Controller:
│     Device ID:          7d29f2075dcafb4488b40c73f199cf46bb76bddb
│     Current version:    01
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              24f9394e-567d-573a-9e22-4ce1d0668241
│                         b315bc76-8da0-5bc6-82e8-296673694802
│                         37d930ee-e811-585c-a094-f430aa4d42fb
│                         40dd51ce-4d2b-5863-a55b-e5e05b828e18
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─AMD Ryzen 5 3600 6-Core Processor              :
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         b54f3c89-8a41-5b9f-9a5c-0367c19a9451
│                         d951a23c-a920-50ac-ad89-0660aa0be95c
│     Device Flags:       • Internal device
│   
├─Baffin [Radeon RX 550 640SP / RX 560/560X]:
│     Device ID:          ce6d39e7c84738d80b6899f99329c88aacd67559
│     Current version:    cf
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              ca4bc559-c415-502f-a9c6-7800b9a9ff0f
[... 143 lines cut ...]
                                               #5  0x00007f543b7833f9 start_thread (libpthread.so.0 + 0x93f9)
                                               #6  0x00007f543b966b53 __clone (libc.so.6 + 0x101b53)
Apr 26 13:25:21 fedora systemd[1]: systemd-coredump@0-2555-0.service: Succeeded.
Apr 26 13:25:21 fedora audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@0-2555-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Apr 26 13:25:21 fedora audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Apr 26 13:25:21 fedora systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Apr 26 13:25:21 fedora systemd[1]: fwupd.service: Failed with result 'core-dump'.
Apr 26 13:25:21 fedora audit: BPF prog-id=52 op=UNLOAD
Apr 26 13:25:21 fedora audit: BPF prog-id=51 op=UNLOAD
Apr 26 13:25:21 fedora audit: BPF prog-id=50 op=UNLOAD
Apr 26 13:25:21 fedora audit[2585]: AVC avc:  denied  { read } for  pid=2585 comm="gdb" name="quirks.xmlb" dev="sda3" ino=1310829 scontext=system_u:system_r:abrt_t:s0-s0:c0.c1023 tcontext=system_u:object_r:fwupd_cache_t:s0 tclass=file permissive=0
Apr 26 13:25:21 fedora audit[2585]: AVC avc:  denied  { read } for  pid=2585 comm="gdb" name="metadata.xmlb" dev="sda3" ino=1310856 scontext=system_u:system_r:abrt_t:s0-s0:c0.c1023 tcontext=system_u:object_r:fwupd_cache_t:s0 tclass=file permissive=0
Apr 26 13:25:21 fedora audit[2585]: AVC avc:  denied  { read } for  pid=2585 comm="gdb" name="metainfo.xmlb" dev="sda3" ino=1310832 scontext=system_u:system_r:abrt_t:s0-s0:c0.c1023 tcontext=system_u:object_r:fwupd_cache_t:s0 tclass=file permissive=0
Apr 26 13:25:22 fedora abrt-server[2564]: Deleting problem directory ccpp-2021-04-26-13:25:21.454435-2397 (dup of ccpp-2021-04-25-12:22:48.412567-2378)
Apr 26 13:25:22 fedora abrt-notification[2621]: Process 2378 (fwupd) crashed in g_propagate_error()
Apr 26 13:25:22 fedora abrt-dump-journal-oops[858]: abrt-dump-journal-oops: Found oopses: 1
Apr 26 13:25:22 fedora abrt-dump-journal-oops[858]: abrt-dump-journal-oops: Creating problem directories
Apr 26 13:25:22 fedora abrt-server[2623]: Can't find a meaningful backtrace for hashing in '.'
Apr 26 13:25:22 fedora abrt-server[2623]: Preserving oops '.' because DropNotReportableOopses is 'no'
Apr 26 13:25:22 fedora abrt-notification[2640]: System encountered a non-fatal error in ??()
Apr 26 13:25:23 fedora abrt-dump-journal-oops[858]: Reported 1 kernel oopses to Abrt
```

[image]
[image]

hughsie, 2021-04-26:
This is the fix for that issue: https://bodhi.fedoraproject.org/updates/FEDORA-2021-25a414fb4e -- it went to stable yesterday.

knstn-project, 2021-04-27:
I can confirm. It got fixed.

closed 2021-04-27 by knstn-project