fwupd/fwupd #3192: Regression in the build system breaks discovery of EFI app dir

blechta, 2021-04-26:
**Describe the bug**
`/snap/bin/fwupdmgr update` fails with `(null)/fwupdx64.efi cannot be found` apparently because of [this regression](https://github.com/fwupd/fwupd/commit/99832622e157067e40896815ace0f8f7a153e8af#diff-30d8f6be6320feeacf686be94f48c70869b52630e01ea625f0f15adc0d57c3e4L375). The variable `EFI_APP_LOCATION` has been removed from config which has the unintended consequence that `fu_common_get_path (FU_KIND_PATH_EFIAPPDIR)` [returns NULL here](https://github.com/fwupd/fwupd/blob/5ad6fb2a2708f78f8c29fbfe2c41e6b211f18fd7/libfwupdplugin/fu-common.c#L1233).

**Steps to Reproduce**
Assuming there is an update with EFI on hold:
```
snap refresh fwupd --revision 3039
/snap/bin/fwupdmgr refresh
/snap/bin/fwupdmgr get-updates
/snap/bin/fwupdmgr update
```
results in
```
(null)/fwupdx64.efi cannot be found
```

Rolling back to 1.5.9:
```
snap refresh fwupd --revision 3013
/snap/bin/fwupdmgr refresh
/snap/bin/fwupdmgr get-updates
/snap/bin/fwupdmgr update
```
fixes the problem.

**Expected behavior**
Update proceeds without error.

**fwupd version information**
Please provide the version of the daemon and client.
```
$ /snap/bin/fwupdmgr --version
client version:	1.6.0
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.6.0
$ snap list fwupd 
Name   Version  Rev   Tracking       Publisher       Notes
fwupd  1.6.0    3039  latest/stable  richard-hughes  classic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): snap

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```$ /snap/bin/fwupdmgr get-devices --show-all-devices
WARNING: This package has not been validated, it may not work properly.
20S1S06C00
│
├─Thunderbolt host controller:
│     Device ID:          052931e2eaccbc4d4c28d1f25b6f815eccef8d36
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              982e2ab2-6207-554c-974d-6300e30cd597
│                         d6591a43-ce08-57d7-8f00-a4f0cb625e01
│                         d868740a-5d9b-56e0-88b6-c7f76a26d0fa
│                         c4bbfb26-644e-50ee-b0dc-1edbc4e6c033
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Apacer AS2280P4 480GB:
│     Device ID:          599ead362818be284e800cbc30d4c29f934086c3
│     Summary:            NVM Express Solid State Drive
│     Current version:    AP123PH0
│     Vendor:             Phison Electronics Corporation (NVME:0x1987)
[... 564 lines cut ...]
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal
```

- Have you tried rebooting?

Yes.

- Is this a regression?

Yes: [the removal of conf variable `EFI_APP_LOCATION`](https://github.com/fwupd/fwupd/commit/99832622e157067e40896815ace0f8f7a153e8af#diff-30d8f6be6320feeacf686be94f48c70869b52630e01ea625f0f15adc0d57c3e4L375) which is used from `libfwupdplugin/fu-common.c`.

- Are you using an NVMe disk?

Yes.

- Is secure boot enabled?

No.

- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

No.

superm1, 2021-04-27:
Thanks for digging in and sleuthing, it looks like a real regression, and simple enough fix to add those two lines back in `meson.build`.  Would you mind opening a PR with the change to add them back in and we can re-promote the snap when they're included.

blechta, 2021-04-28:
PR opened: #3206

closed 2021-04-29 by superm1