fwupd/fwupd #3193: fu_cabinet_parse_release assumes sha1 regardless of 'type'

djcb, 2021-04-27:
**Describe the bug**
It seems the checksum-verification in `fu_cabinet_parse_release` does check the `type` of the checksum, and hard-codes `sha1`. 
 
**Steps to Reproduce**

A metadata.xml file with
```
<checksum filename="file.bla" type="sha256" target="content">e1eabc2deac089b2b7c0ecba90bb064f99b412f33cbadeac8f885823799d90b5</checksum>
```
yields

```
contents checksum invalid, expected ca21ab1324299c7e353bf5bdd1ae2bcbf3ea8660, got e1eabc2deac089b2b7c0ecba90bb064f99b412f33cbadeac8f885823799d90b5
```
where the `ca21a...` is the sha1 checksum.

**fwupd version information**
git HEAD.

fu-cabinet.c:289 has
```
		checksum = g_compute_checksum_for_bytes (G_CHECKSUM_SHA1, blob);
```
so I guess this should use the 'type' attribute instead.

hughsie, 2021-04-27:
It's actually _two bugs in a trenchcoat_, as `<releases><checksum>` is deprecated anyway. I can have a look at this now if you want.

hughsie, 2021-04-27:
Ahh, so the reason we didn't switch to artifact was because the AppStream spec doesn't have a "pointing at" concept. I've started a discussion on the AppStream mailing list and cc'd you there.

djcb, 2021-04-27:
Oh, cool! 

The issue is easy to work around of course (just use sha1!) but I figured it's good to document here.

hughsie, 2021-04-27:
> issue is easy to work around of course

Yes, but I'm a great believer in fixing the root cause, so I appreciated you opening the issue. :)

closed 2021-05-06 by hughsie