fwupd/fwupd #3196: Dell WD19 device missing from fwupd

CragW, 2021-04-27:
**Describe the bug**
WD19TB device missing if fwupd is launched in systemd. Issue gone away if fwupd starts manually through command line. Issue is reproducible with dock firmware that already in LVFS stable channel.

Log: [fwupd-dock-missing.log](https://github.com/fwupd/fwupd/files/6385665/fwupd-dock-missing.log)

**Steps to Reproduce**
Check fwupd service is active and running, found WD19TB dock is missing
```shell
$ systemctl status fwupd
$ fwupdmgr get-devices
```

**Expected behavior**
WD19TB is seen in `fwupdmgr get-devices` output

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.6.0-1-g19e77ba4
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.6.0-1-g19e77ba4
```

**Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):**
From source commit 5ad6fb2a

**fwupd device information**
Please provide the output of the external fwupd devices recognized in your system.
```shell
$ fwupdmgr get-devices --filter=~internal
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: This package has not been validated, it may not work properly.
No hardware detected with firmware update capability

```

**Dock SKU**
Please mention which module is installed in your WD19.

- [ ] WD19 (Single-C)
- [x] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**
No extended peripherals

**Verbose daemon logs**
First enable daemon verbose logs collection.
```shell
fwupdmgr modify-config "VerboseDomains" "*"
```

Then try to reproduce the issue.  Even if it doesn't reproduce, please attach the
daemon verbose logs collected from the system journal.
```shell
journalctl -b -u fwupd.service
```
Log: [journalctl-b-u-fwupd.service.log](https://github.com/fwupd/fwupd/files/6385659/journalctl-b-u-fwupd.service.log)

**Additional questions**
- Operating system and version: Ubuntu 20.04.2 LTS
- Have you tried unplugging the dock or any peripherals from your machine? Yes, issue persisted.
- Have you tried to power cycle the dock from the AC adapter?  Yes, issue persisted.
- Is this a regression? Not sure.

hughsie, 2021-04-27:
If you do `fwupdmgr get-devices --show-all` does that show it? Weird.

superm1, 2021-04-27:
```Failed to query dock info: write over HID-I2C failed: failed after 5 retries: failed to SetReport: transfer timed out```

I noticed this in one of your logs.  That's pretty weird - particularly only happening with systemd.

Can you add `FWUPD_PROBE_VERBOSE=1` to your systemd unit and share the new journal log from that test run?

CragW, 2021-04-27:
It's still unseen from `fwupdmgr get-devices --show-all`  :(
[fwupdmgr-get-devices-show-all.log](https://github.com/fwupd/fwupd/files/6385782/fwupdmgr-get-devices-show-all.log)

CragW, 2021-04-27:
I've added environment `FWUPD_PROBE_VERBOSE=1 `to systemd unit, hereunder the updated journal log:
[probe-verbose-journalctl-b-u-fwupd.service.log](https://github.com/fwupd/fwupd/files/6385858/probe-verbose-journalctl-b-u-fwupd.service.log)

superm1, 2021-04-27:
```
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]: 15:38:16:0925 FuEngine             usb added FuUsbDevice:
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]: Unknown Device
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   InstanceIds:          USB\VID_413C&PID_B06E&REV_0101
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   InstanceIds:          USB\VID_413C&PID_B06E
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   Flags:                none
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   VendorId:             USB:0x413C
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   Version:              1.1
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   VersionFormat:        bcd
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   PhysicalId:           usb:05:00:02:05
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]: 15:38:16:0925 FuEngine             usb added FuUsbDevice:
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]: Unknown Device
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   Flags:                none
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   PhysicalId:           usb:05:00:02:03:05
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]: 15:38:16:0926 FuEngine             usb added FuUsbDevice:
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]: Unknown Device
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   InstanceIds:          USB\VID_413C&PID_B06F&REV_0101
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   InstanceIds:          USB\VID_413C&PID_B06F
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   Flags:                none
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   VendorId:             USB:0x413C
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   Version:              1.1
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   VersionFormat:        bcd
Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]:   PhysicalId:           usb:05:00:02:03:05
```

I expected those to show `Unprobed Dell accessory endpoint`

I wonder if the quirks aren't being loaded when launched from systemd.

superm1, 2021-04-27:
Can you try to launch this way and see if it still reproduces?

```
# CONFIGURATION_DIRECTORY=/etc/fwupd STATE_DIRECTORY=/var/lib/fwupd CACHE_DIRECTORY=/var/cache/fwupd /usr/local/libexec/fwupd/fwupd -v
```

Launching from systemd overrides those variables, and I'm wondering if perhaps something you have there is playing into the issue.  If you can reproduce with that still, it will be easier to figure out with the debugger.

djcampello, 2021-04-28:
I had a similar issue... While fwupdtool seem to work properly, fwupdmgr is having issues with the WD19TB when providing GUID:
```
localhost ~ # fwupdmgr update 8ceeeffd-51b6-580c-9b75-69143227aff8
WARNING: This package has not been validated, it may not work properly.
Selected device: Package level of Dell dock
failed to find any device with composite ID 947b690100e39186e0f84e1a7e65c4c1805321fe
```

hughsie, 2021-04-28:
I've got a WD19TB here (albeit an DVT sample...) and that seems to enumerate just fine even using systemd. I'm using Fedora 34 with all updates and fwupd from master, c6bb21356f2b940e6230fe205be6d7b4c77c2517 commit. I do get the `failed to find any device with composite ID` error -- investigating now.

hughsie, 2021-04-28:
^^ fixes the `failed to find any device with composite ID` failure -- but not sure about the systemd issue tho as it's completely unrelated code.

CragW, 2021-04-28:
> Can you try to launch this way and see if it still reproduces?
> 
> ```
> # CONFIGURATION_DIRECTORY=/etc/fwupd STATE_DIRECTORY=/var/lib/fwupd CACHE_DIRECTORY=/var/cache/fwupd /usr/local/libexec/fwupd/fwupd -v
> ```
Issue is reproducible with above command, that helped to connecting a debugger, thanks.

I can see fwupd engine load `plugins` and `quirks` fine, and `usb` `backend` `coldplug` (WD19 already connected) emits the enumeration but `dell_dock` `cb` keeps silent without responses.

As fwupd reuses the cached files, I deleted `/var/cache/fwupd/*` to force `XbSilo` re-compilation. Afterwards WD19 is seen in fwupd as well as client side `fwupdmgr get-devices`.

I have no idea why but I guess there was a buggy `quirks.xmlb` in my cached dir.

hughsie, 2021-04-28:
> but I guess there was a buggy quirks.xmlb in my cached dir

Hmm, interesting, well done on tracking that down. I see in your log:

> Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]: 15:38:16:0661 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
> Apr 27 23:38:16 crabx-XPS-15-7590 fwupd[50759]: 15:38:16:0661 XbSilo               file: c781187f-5ddd-8479-3f9c-13b89aa141aa, current:c781187f-5ddd-8479-3f9c-13b89aa141aa, cached: (null)

so libxmlb seems to think that all the silo is up to date. We use the filename and the mtime/ctimes on all the .quirk files to know when to regenerate the silo -- is there any specific about your system that might be relevant to that? e.g. are you on a system with no mtimes or ctimes perhaps?

CragW, 2021-04-28:
My system supports timestamp. I can only guess that the cached quirks file for some reason missed incorporating the elements in quirks.d at the very beginning. I will distribute this try of deletion to some people who are facing this issue too. Will back with updates.

superm1, 2021-04-28:
@hughsie does it look for only newer, or any change?  I'm wondering if this can be a side effect of switching branches that might have had changes in the quirk files.

hughsie, 2021-04-29:
Here's a good example:

    ./src/xb-tool compile test.xmlb /usr/share/app-info/xmls/*.xml

    (xb-tool:188173): 09:35:30.201: en_GB&en&C&/usr/share/app-info/xmls/adobe-flash.xml:ctime=1618235273.938047&/usr/share/app-info/xmls/fedora-categories.xml:ctime=1618235273.939047&...etc

So the "GUID" we compare at startup is made up of things like the current locale (as we strip other languages out of the blob by default, although not super relevant for fwupd), the full filename and path and the ctime -- which slightly confusingly in this instance means "changed time" rather than "created time". It should really be called `mtime` but whatever.

superm1, 2021-04-29:
So basically anything that changes for ctime should cause a recompile?

hughsie, 2021-04-29:
Exactly that.

cata0309, 2021-05-21:
I am using Manjaro Linux and I still get the `failed to find any device with composite ID` with the latest version.

cata0309, 2021-06-20:
Did the change fix the issue?

CragW, 2021-06-21:
Partner verified this issue is gone, thank you all!

closed 2021-06-21 by superm1

krzyk, 2021-12-20:
Strange, I have similar issue.

My WD19TB was at 1.00.00.1 version (if I remember correctly) and was being listed.
I upgraded to the latest firmware from Dell site (1.00.21) and now fwupdmgr doesn't list dock at all.

CragW, 2021-12-20:
> My WD19TB was at 1.00.00.1 version (if I remember correctly) and was being listed. I upgraded to the latest firmware from Dell site (1.00.21) and now fwupdmgr doesn't list dock at all.

Did you unplug the dock cable upon the completion of firmware update? Would it show up if you added `--show-all`?

krzyk, 2021-12-22:
Yes, as I was instructed in few places.

What is strange is that the dock worked for some time (when I plugged it back in, still no show in fwmgr) until the laptop went to sleep.

And then no device from it was detected ever again.

I also tried boltctl forget (because it was always shown there as disconnected) and now it doesn't show there also.

krzyk, 2021-12-22:
BTW. The device is visible in `/sys/bus/thunderbolt/devices` and it periodically throws to dmesg:
```
[ 1021.877880] thunderbolt 0000:00:0d.3: no switch exists at 3, ignoring
```