fwupd/fwupd #3197: fwupd crashes on startup

gwduvalljr, 2021-04-27:
Dell XPS 13 9310
Ubuntu 20.04
Crash Dump Attached

I am happy to provide anything else needed.

Error is:
```
fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static; vendor preset: enabled)
     Active: failed (Result: core-dump) since Tue 2021-04-27 15:01:39 EDT; 5min ago
       Docs: https://fwupd.org/
    Process: 14289 ExecStart=/usr/libexec/fwupd/fwupd (code=dumped, signal=SEGV)
   Main PID: 14289 (code=dumped, signal=SEGV)

Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Failed to read oom_kill field of memory.events cgroup attribute: No such file or directory
Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Child 14289 belongs to fwupd.service.
Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Failed with result 'core-dump'.
Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Service will not restart (restart setting)
Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Changed start -> failed
Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Job 4570 fwupd.service/start finished, result=failed
Apr 27 15:01:39 garlandXPS13 systemd[1]: Failed to start Firmware update daemon.
Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Unit entered failed state.
Apr 27 15:01:39 garlandXPS13 systemd[1]: fwupd.service: Control group is empty.

```
[_usr_libexec_fwupd_fwupd.0.zip](https://github.com/fwupd/fwupd/files/6387021/_usr_libexec_fwupd_fwupd.0.zip)

hughsie, 2021-04-27:
You didn't fill in any of the requested details in the issue template. Backtrack please.

gwduvalljr, 2021-04-27:
When I first clicked "New Issue" I wasn't shown a template. Just now found them. Interesting. Anyway...

**Describe the bug**
fwupd crashes on start

**Steps to Reproduce**
Attempted to start on boot. Attempted restart/start using systemctl. Attempted to run sudo /usr/libexec/fwupd/fwupd.

**Expected behavior**
Expected fwupd to start and continue running.

**fwupd version information**
1.3.11

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.
Since fwupd is not running all I get is a timeout.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version: Ubuntu 20.04
- Have you tried rebooting? you bet
- Is this a regression? no

hughsie, 2021-04-27:
Can you get a backtrace please.

gwduvalljr, 2021-04-28:
Hopefully I did this correctly. 
[gdb-fwupd.txt](https://github.com/fwupd/fwupd/files/6393805/gdb-fwupd.txt)

hughsie, 2021-04-29:
Hmm, a crash in fu_plugin_list_depsolve() doesn't ring any bells for me. Would it be possible for you to install the debug symbols for fwupd and get the same backtrace again please? Then we can see what line number it crashed at. Thanks!

gwduvalljr, 2021-04-29:
Considering I have never done this before, I gave it my best shot and got the below. What did I do wrong?

```
gwduvalljr@garlandXPS13:~$ sudo apt install fwupd-dbgsym
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 fwupd-dbgsym : Depends: fwupd (= 1.3.9-4) but 1.3.11-1~focal1 is to be installed
E: Unable to correct problems, you have held broken packages.

```

hughsie, 2021-04-29:
No idea about Debian I'm afraid, @superm1 might know the magic incantation.

superm1, 2021-04-29:
Here's the package you can grab from LP.
https://launchpad.net/ubuntu/focal/amd64/fwupd-dbgsym/1.3.11-1~focal1

I don't know that it publishes to the archive.

gwduvalljr, 2021-04-29:
Thank you. I downloaded and installed that package then ran the same steps as yesterday to produce the attached. Again, I hope that I did it correctly. 

[gdb-fwupd.txt](https://github.com/fwupd/fwupd/files/6399243/gdb-fwupd.txt)

superm1, 2021-04-29:
If I was gonna guess it's this https://github.com/fwupd/fwupd/commit/11c5941f2318bad7ece410f989b31884215e1164#diff-1c27ba2c31831489067a41880123772bd5d086e9d46182df933236440f8c0ffcR148 commit which fixed it later on.

hughsie, 2021-04-29:
Line 147 is:

			deps = fu_plugin_get_rules (plugin, FU_PLUGIN_RULE_RUN_AFTER);
			for (guint j = 0; j < deps->len && !changes; j++) {

...but that's before the lazy loading stuff was introduced. I do wonder if an older plugin is being loaded or something, something with a smaller %FU_PLUGIN_RULE_LAST. It's certainly an odd one.

@gwduvalljr would you be able to test with 1_4_X or 1_5_X? Thanks.

superm1, 2021-04-29:
Two ideas

If you add `--verbose` to the fwupd daemon launch and share the logs we might see which are loading for you last before the crash.

I wonder if in the context of that crash it's possible to call `fu_plugin_get_name (plugin)` to tell us which one is getting loaded causing havok.  There isn't anything in Ubuntu that distributes plugins other than `fwupd`, so whatever problem this is I expect is fully in tree.

superm1, 2021-04-29:
One more idea - is it possible you didn't update all your libraries?  Make libfwupdplugin is out of date relative to fwupd?

Can confirm like this:
`dpkg -l | grep fwupd`

gwduvalljr, 2021-04-29:
Here you go. Let me know if you need something else or what I need to do next.

```
gwduvalljr@garlandXPS13:~$ dpkg -l | grep fwupd
ii  fwupd                                      1.3.11-1~focal1                                           amd64        Firmware update daemon
ii  fwupd-dbgsym                               1.3.11-1~focal1                                           amd64        debug symbols for fwupd
ii  fwupd-signed                               1.27.1ubuntu2+1.3.11-1~focal1                             amd64        Linux Firmware Updater EFI signed binary
ii  libfwupd2:amd64                            1.5.8-0ubuntu1pop0~1617221312~20.04~a9de48d               amd64        Firmware update daemon library
ii  libfwupdplugin1:amd64                      1.5.8-0ubuntu1pop0~1617221312~20.04~a9de48d               amd64        Firmware update daemon plugin library
gwduvalljr@garlandXPS13:~$ 

```

```
[gdb-fwupd.txt](https://github.com/fwupd/fwupd/files/6400333/gdb-fwupd.txt)
```

gwduvalljr, 2021-04-29:
[gdb-fwupd.txt](https://github.com/fwupd/fwupd/files/6400340/gdb-fwupd.txt)

I think I hosed up the upload, so trying again.

superm1, 2021-04-29:
Ah!  There's the problem.

```
ii  libfwupd2:amd64                            1.5.8-0ubuntu1pop0~1617221312~20.04~a9de48d               amd64        Firmware update daemon library
ii  libfwupdplugin1:amd64                      1.5.8-0ubuntu1pop0~1617221312~20.04~a9de48d               amd64        Firmware update daemon plugin library
```

You've got some libraries from somewhere else.

gwduvalljr, 2021-04-29:
Interesting. Got any idea how I get to where I should be?

superm1, 2021-04-29:
This should do it.
```
apt install libfwupd2=1.3.11-1~focal1 libfwupdplugin1=1.3.11-1~focal1
```

gwduvalljr, 2021-04-29:
That certainly seems to have helped. The service is running and responding to queries through fwupdmgr. So, now I wonder where those "rogue" libs came from?

Thank you all for your help. I will try to discover the source of those libs so I won't do whatever it was I did again.

superm1, 2021-04-29:
Some PPA probably.

superm1, 2021-04-29:
No idea how we could...

hughsie, 2021-04-29:
It's a link time thing, perhaps the daemon should check the file mtime is the same? Maybe we could put some more explicit requirements in the control file?

joakim-tjernlund, 2021-04-30:
Just install Gentoo instead, no more binary lib problem again :)

closed 2021-05-06 by hughsie