fwupd/fwupd #3202: UEFI update 0.1.16 for my Thinkpad P52 has reset the UEFI boot records

rugubara, 2021-04-28:
**Describe the bug**
The UEFI firmware 0.1.16 for my Lenovo Thinkpad P52 20M90019RT wiped the grub boot entry rendering the system unbootable. While trying to fix this by booting from a rescue CD, I discovered that this firmware also reset the boot type to legacy. Changing the boot type to UEFI only allowed me to fix the boot records by re-installing grub. 

**Steps to Reproduce**
Install UEFI firmware 0.1.16 for a compatible hardware.

**Expected behavior**
Firmware update must leave the boot settings and boot records intact.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
PF16W6Y2 ~ # fwupdmgr --version
client version:1.5.9
compile-time dependency versions

daemon version:1.5.9

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): installed from sources (Gentoo emerge)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
PF16W6Y2 ~ # fwupdmgr get-devices --show-all-devices
WARNING: This package has not been validated, it may not work properly.
20M90019RT
│
├─Embedded Controller:
│     Device ID:          0f3c64df11304fc5711ef77a93e3e042e2738907
│     Current version:    0.1.16
│     Minimum Version:    0.1.16
│     Vendor:             DMI:LENOVO
│     GUIDs:              88b8ba7e-296b-4f9e-929f-dcc19318dbcf
│                         629122db-a18d-5770-a3f3-1e3d42907f7c ← UEFI\RES_{88B8BA7E-296B-4F9E-929F-DCC19318DBCF}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─GP107GLM [Quadro P2000 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              5b054688-b689-5dd0-993b-102add23543a ← PCI\VEN_10DE&DEV_1CBA&SUBSYS_17AA225F&REV_A1
│                         b5dcc529-a180-5fa2-9c4d-e3e1688148c4 ← PCI\VEN_10DE&DEV_1CBA&SUBSYS_17AA225F
│                         bc8f6213-8695-5407-be5d-cd03808d40af ← PCI\VEN_10DE&DEV_1CBA&REV_A1
│                         07633373-c1be-577f-a080-1f9ae17806eb ← PCI\VEN_10DE&DEV_1CBA
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─Intel Management Engine:
│     Device ID:          619057f1d0da67b3fa4e9f74e09fc9218ce32643
│     Current version:    192.70.1652
│     Minimum Version:    192.70.1652
│     Vendor:             DMI:LENOVO
│     GUIDs:              971682af-21a0-46e3-91c0-825702c2ba70
│                         b892b4fd-841b-5ec5-a700-437299fb233f ← UEFI\RES_{971682AF-21A0-46E3-91C0-825702C2BA70}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel(R) Core™ i7-8850H CPU @ 2.60GHz:
[... 568 lines cut ...]
├── vmlinuz-5.11.13-gentoo
├── vmlinuz-5.11.15-gentoo
├── vmlinuz-5.11.16-gentoo
├── vmlinuz-5.11.4-rt-rt11
├── vmlinuz-5.11.5-gentoo
├── vmlinuz-5.11.5-gentoo.old
├── vmlinuz-5.11.6-gentoo
├── vmlinuz-5.11.7-gentoo
├── vmlinuz-5.11.8-gentoo
├── vmlinuz-5.11.9-gentoo
├── vmlinuz-5.4.84-rt-rt47
├── vmlinuz-5.9.1-rt-rt20
└── vmlinuz-5.9.1-rt-rt20.old

15 directories, 438 files
PF16W6Y2 ~ #
```

**Additional questions**
- Operating system and version: gentoo linux
- Have you tried rebooting? yes
- Is this a regression? 
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Lenovo system w/o Boot Order Lock

hughsie, 2021-04-28:
What was the *old* version of the system firmware? Not the version you installed. Thanks!

rugubara, 2021-04-28:
```
PF16W6Y2 ~ # fwupdmgr get-history
WARNING: This package has not been validated, it may not work properly.
20M90019RT
│
├─UEFI Device Firmware:
│ │   Device ID:          0f3c64df11304fc5711ef77a93e3e042e2738907
│ │   Previous version:   0.1.15
│ │   Update State:       Success
│ │   Last modified:      2019-12-21 13:21
│ │   GUID:               88b8ba7e-296b-4f9e-929f-dcc19318dbcf
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.16
│       License:          Unknown

```

hughsie, 2021-04-29:
@mrhpearson could you ask internally and ask if there are any EFI BootXXXX bugs that might affect ThinkPad P52 system BIOS 0.1.15 -> 0.1.16 updates please? This is the first bug like this I've seen on this hardware. Thanks!