fwupd/fwupd #3204: TPM PCR0 differs from reconstruction

Dmole, 2021-04-28:
**Describe the bug**

```
• USB4 Controller has no available firmware updates
• USB4 Controller has no available firmware updates
• PC SN730 NVMe WDC 512GB has the latest available firmware version
• TPM 2.0 has no available firmware updates
• Touchpad has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
XPS 13 9310
│
└─System Firmware:
  │   Device ID:           070a7ea3ab21dbc8d5c913fe39e3fb636b985492
  │   Current version:     2.1.1
  │   Minimum Version:     2.1.1
  │   Vendor:              Dell Inc. (DMI:Dell Inc.)
  │   Update Error:        TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
  │   GUID:                cfb08d7c-8d3d-41b1-846b-48f634dfac9d
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │ 
  └─XPS 13 9310 System Update:
        New version:       2.2.0
        Remote ID:         lvfs
        Summary:           Firmware for the Dell XPS 13 9310
        License:           Proprietary
        Size:              26.6 MB
        Vendor:            Dell Inc.
        Flags:             is-upgrade
        Description:       This stable release fixes the following issues:
        
         • Updated the Thermal Algorithm to reduce the system temperature and fan noise.
```

**Steps to Reproduce**

```
sudo fwupdmgr get-updates
```

**Expected behavior**

Upgrades should be applied.

**fwupd version information**

```
fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

fwupd is included in the default install of Ubuntu managed by apt

**fwupd device information**

```
fwupdmgr get-devices --show-all-devices
XPS 13 9310
│
├─USB4 Controller:
│     Device ID:           d94ca2fd291f19be85edf1f7ab30b82d6b5d7732
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     00.00
│     Vendor:              TBT:0x00D4
│     Update Error:        Device is in safe mode
│     GUID:                4d86f168-e1cc-5995-afd3-ae9df6a14f5e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│   
├─USB4 Controller:
│     Device ID:           540fd5e379d7f616a4e4c4b0005d7432830c72a5
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     00.00
│     Vendor:              TBT:0x00D4
│     Update Error:        Device is in safe mode
│     GUID:                4d86f168-e1cc-5995-afd3-ae9df6a14f5e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│   
├─Display controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     01
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               3c11d563-4f83-5bd1-8369-75359ea6b6f6
│                          54ff3ed7-1ea3-5279-975f-93c13bf2aa53
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─PC SN730 NVMe WDC 512GB:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:             NVM Express Solid State Drive
│     Current version:     11150012
│     Vendor:              Sandisk Corp (NVME:0x15B7)
│     GUIDs:               10215a6e-5eb4-5fe4-a3d0-1ff4b3295840
│                          e3a1928f-2cb0-4b92-8526-9368d8c0c2d1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           070a7ea3ab21dbc8d5c913fe39e3fb636b985492
│     Current version:     2.1.1
│     Minimum Version:     2.1.1
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Error:        TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUID:                cfb08d7c-8d3d-41b1-846b-48f634dfac9d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM 2.0:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Summary:             Platform TPM device
│ │   Current version:     0.1.1.1
│ │   Vendor:              Dell Inc. (PCI:0x1028)
│ │   GUIDs:               1c510e19-0b20-5a48-aa73-cba9daabdead
│ │                        ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        73730635-f6c2-53da-9df2-948bb5ac1022
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       Vendor:            Dell Inc. (PCI:0x1028)
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─Touchpad:
│     Device ID:           0eeea9b449795ed307fcba6fa46f596d9679b87b
│     Current version:     1.3.3040414
│     Bootloader Version:  54.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               76018764-887c-558b-a6c0-ae31e4522d98
│                          f6982c83-dec9-5961-a250-384441c12205
│                          d69eed5b-0b5e-53c7-ac04-8e7a672f0365
│                          913a517f-55c6-5f1d-b018-405af36fcbc0
│     Device Flags:        • Internal device
│                          • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:           6b9d757c1f64d696e7af69942c2d6ed3993479d4
│     Current version:     0.0.134
│     Minimum Version:     0.0.134
│     Vendor:              DMI:Dell Inc.
│     GUID:                176e090d-0ddb-495e-8173-bc998ccfecd0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           5b1ec19a3f6b7f8dc4e7dec250ecbf3ea1be366e
      Current version:     17.21.18
      Minimum Version:     17.21.18
      Vendor:              DMI:Dell Inc.
      GUID:                e3a1928f-2cb0-4b92-8526-9368d8c0c2d1
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version:

```
lsb_release -d
Description:	Ubuntu 20.04.2 LTS
```

- Have you tried rebooting?

No

- Is this a regression?

Yes, it would be nice id there were a /var/log/fwupd.log like the apt log, but without that all that can be said is previous runs did upgrade things without issue;
 
```
zgrep fwupd /var/log/auth.log*
 /var/log/auth.log:Apr 27 10:27:28 XPS sudo:  fwupdmgr get-updates
/var/log/auth.log:Apr 28 09:39:39 XPS sudo:  fwupdmgr get-updates
/var/log/auth.log:Apr 28 10:02:50 XPS sudo:  fwupdmgr get-updates
/var/log/auth.log:Apr 28 10:04:54 XPS sudo:  fwupdmgr get-updates
/var/log/auth.log.1:Apr 20 16:52:21 XPS sudo:  fwupdmgr get-updates
/var/log/auth.log.1:Apr 21 09:59:34 XPS dbus-daemon[1066]: [system] Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
/var/log/auth.log.1:Apr 23 16:01:58 XPS sudo:  fwupdmgr get-updates
/var/log/auth.log.2.gz:Apr 15 10:05:39 XPS sudo:  fwupdmgr get-updates
/var/log/auth.log.2.gz:Apr 15 10:37:25 XPS sudo:  fwupdmgr get-updates
/var/log/auth.log.2.gz:Apr 15 23:07:40 XPS sudo:  fwupdmgr get-updates

```

- Other:
```
cat /sys/class/power_supply/BAT*/status
Charging
```

hughsie, 2021-04-28:
Please read https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction

Dmole, 2021-04-28:
@hughsie 

> Please read https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction

I did are you saying I have one of

 -   A hardware failure
 -   Presence of malware on the system

?

and that 

 -   An error in the firmware TPM event log.
 -   An error in the fwupd reconstruction of the TPM PCR0

are not possible or not bug worthy?

Please explain.

superm1, 2021-04-28:
Your fwupd version has known bugs with this (1.3.11) as mentioned on that page.

Dmole, 2021-04-28:
@superm1 Thanks,
I think I somehow confused the firmware version with the fwupd version.
Is there an ETA for the Ubuntu maintainer to get a new version or is it recommended I uninstall fwupd and reinstall it from source or using a different Linux distribution?

superm1, 2021-04-28:
There are open bugs in Ubuntu discussing moving to a newer version, you can look for those and follow them.

Alternatives would be to switch to non lts release or another distro or compiling by hand.