fwupd/fwupd #3209: Thunderbolt controller on Lenovo P52S not being detected

azonenberg, 2021-04-30:
**Describe the bug**
A clear and concise description of what the bug is.

The Thunderbolt controller on my laptop isn't showing up as an updateable device.

**Steps to Reproduce**
Steps to reproduce the behavior.

lspci output:
`
09:00.0 System peripheral: Intel Corporation JHL6240 Thunderbolt 3 NHI (Low Power) [Alpine Ridge LP 2016] (rev 01)
	Subsystem: Device 2222:1111
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at ea000000 (32-bit, non-prefetchable) [size=256K]
	Memory at ea040000 (32-bit, non-prefetchable) [size=4K]
	Capabilities: [80] Power Management version 3
	Capabilities: [88] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [c0] Express Endpoint, MSI 00
	Capabilities: [a0] MSI-X: Enable+ Count=16 Masked-
	Capabilities: [100] Device Serial Number 82-35-c5-61-98-b7-d0-00
	Capabilities: [200] Advanced Error Reporting
	Capabilities: [300] Virtual Channel
	Capabilities: [400] Power Budgeting <?>
	Capabilities: [500] Vendor Specific Information: ID=1234 Rev=1 Len=088 <?>
	Capabilities: [600] Latency Tolerance Reporting
	Kernel driver in use: thunderbolt
	Kernel modules: thunderbolt
`

dmesg output
`
Apr 29 20:28:37 cottonmouth fwupd[12227]: 03:28:37:0525 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1d.0/0000:07:00.0/0000:08:00.0/0000:09:00.0/domain0/0-0: cannot ensure ID: FuThunderboltDevice:
Apr 29 20:28:37 cottonmouth fwupd[12227]: Unknown Device
Apr 29 20:28:37 cottonmouth fwupd[12227]:   Protocol:             com.intel.thunderbolt
Apr 29 20:28:37 cottonmouth fwupd[12227]:   Flags:                require-ac
Apr 29 20:28:37 cottonmouth fwupd[12227]:   VersionFormat:        pair
Apr 29 20:28:37 cottonmouth fwupd[12227]:   Icon:                 thunderbolt
Apr 29 20:28:37 cottonmouth fwupd[12227]:   BackendId:            /sys/devices/pci0000:00/0000:00:1d.0/0000:07:00.0/0000:08:00.0/0000:09:00.0/domain0/0-0
Apr 29 20:28:37 cottonmouth fwupd[12227]:   PossiblePlugin:       thunderbolt
Apr 29 20:28:37 cottonmouth fwupd[12227]:   SysfsPath:            /sys/devices/pci0000:00/0000:00:1d.0/0000:07:00.0/0000:08:00.0/0000:09:00.0/domain0/0-0
Apr 29 20:28:37 cottonmouth fwupd[12227]:   Subsystem:            thunderbolt
Apr 29 20:28:37 cottonmouth fwupd[12227]:   Device Type:          Thunderbolt device controller
Apr 29 20:28:37 cottonmouth fwupd[12227]:   Safe Mode:            false
Apr 29 20:28:37 cottonmouth fwupd[12227]:   Native mode:          false
Apr 29 20:28:37 cottonmouth fwupd[12227]:   Generation:           0
Apr 29 20:28:37 cottonmouth fwupd[12227]:   AuthAttribute:

`

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.6.0
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.6.0

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Source from github

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.
[... 99 lines cut ...]
│     GUIDs:              c271910d-0695-4b4b-bccf-83550d87a59a
│                         99be9cf7-217b-5215-8ea4-6fcd0d7442e1 ← UEFI\RES_{C271910D-0695-4B4B-BCCF-83550D87A59A}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UHD Graphics 620:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    07
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              ba1ad5fe-baff-5f91-8600-cf7ceafa75ea ← PCI\VEN_8086&DEV_5917&SUBSYS_17AA225B&REV_07
                          36b5cd31-2a2a-508e-abf5-3ad4d882d8b5 ← PCI\VEN_8086&DEV_5917&SUBSYS_17AA225B
                          fed2efa4-7045-55a1-b4fc-b29283d59fe5 ← PCI\VEN_8086&DEV_5917&REV_07
                          8b72a10c-1279-5f8e-a28a-34fb11a58240 ← PCI\VEN_8086&DEV_5917
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available

```

**Additional questions**
- Operating system and version: Debian stable
- Have you tried rebooting? Yes
- Is this a regression? No

azonenberg, 2021-04-30:
Possibly related to #32. I've had Thunderbolt disabled in my BIOS for security since I got the laptop and am now trying to use it for the first time.

hughsie, 2021-04-30:
You need to plug in a thunderbolt device for the controller to appear. The device does its own power saving and only wakes up when a device is inserted.

azonenberg, 2021-04-30:
There is a device plugged in (Sonnet SOLO10G-SFP+ 10G ethernet adapter).

As soon as I plug it in, the "unknown device" message appears in dmesg indicating that the controller has woken up, but fwupd doesn't know what to do with it.

hughsie, 2021-04-30:
@superm1 any idea?

superm1, 2021-04-30:
Can you plug in a thunderbolt or usb4 device instead?

azonenberg, 2021-04-30:
Also relevant: the device is not starting up (which is why I was trying to update the controller firmware in the first place);

dmesg:

`
Apr 30 13:21:16 cottonmouth kernel: [47881.710530] pcieport 0000:00:1d.0: PME: Spurious native interrupt!
Apr 30 13:21:21 cottonmouth kernel: [47886.665955] usb usb4-port1: Cannot enable. Maybe the USB cable is bad?
`

superm1, 2021-05-01:
Oh I didn't realize that was actually tbt3 at first. I would suggest trying to boot with it connected to see if it's any better.

azonenberg, 2021-05-02:
No change. Still the same message in syslog.

It seems like fwupd sees the device, recognizes it as a thunderbolt controller, but doesn't detect it as a potentially updateable device for reasons unknown

hughsie, 2021-05-03:
Can you grab us the output of `sudo FWUPD_PROBE_VERBOSE=1 fwupdtool --plugins thunderbolt --verbose get-devices` please.

azonenberg, 2021-05-03:
<pre>
18:06:12:0968 FuDebug              Verbose debugging enabled (on console 1)
18:06:12:0980 FuConfig             loading config values from /usr/local/etc/fwupd/daemon.conf
18:06:12:0981 FuConfig             using autodetected max archive size 18.4 EB
18:06:12:0984 FuRemoteList         loading /usr/local/share/fwupd/metainfo
18:06:12:0985 XbSilo               attempting to load /var/local/cache/fwupd/metainfo.xmlb
18:06:12:0985 XbSilo               failed to load silo: Failed to open file “/var/local/cache/fwupd/metainfo.xmlb”: open() failed: No such file or directory
18:06:12:0985 XbSilo               compiling /usr/local/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs.metainfo.xml:ctime=1619752633.904607…
18:06:12:0986 XbSilo               compiling /usr/local/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs-testing.metainfo.xml:ctime=1619752633.904607…
18:06:12:0987 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor.conf
18:06:12:0987 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs.conf
18:06:12:0988 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
18:06:12:0988 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/dell-esrt.conf
18:06:12:0988 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
18:06:12:0989 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
18:06:12:0989 FuRemoteList         ignoring unfound remote fwupd
18:06:12:0989 FuRemoteList         ordering lvfs-testing=lvfs+1
18:06:12:0989 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
18:06:12:0989 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
18:06:12:0989 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd/LVFS-CA.pem
18:06:12:0989 Jcat                 reading /usr/local/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
18:06:12:0989 Jcat                 loaded 1 certificates
18:06:12:0989 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
18:06:12:0989 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
18:06:12:0989 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem
18:06:12:0989 Jcat                 reading /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
18:06:12:0989 Jcat                 loaded 1 certificates
18:06:12:0989 Jcat                 generating a 3072 bit RSA private key...
18:06:13:0340 Jcat                 creating path /var/local/lib/fwupd/pki
18:06:13:0340 Jcat                 writing /var/local/lib/fwupd/pki/secret.key with 2455 bytes
18:06:13:0344 Jcat                 creating path /var/local/lib/fwupd/pki
18:06:13:0344 Jcat                 writing /var/local/lib/fwupd/pki/client.pem with 1383 bytes
18:06:13:0349 FuEngine             client certificate exists and working
18:06:13:0349 FuHistory            trying to open database '/var/local/lib/fwupd/pending.db'
18:06:13:0349 FuHistory            no schema version: no such table: schema
18:06:13:0349 FuHistory            got schema version of 0
18:06:13:0349 FuHistory            building initial database
18:06:13:0368 FuHwids              smbios property Manufacturer=LENOVO
18:06:13:0368 FuHwids              smbios property EnclosureKind=a
18:06:13:0368 FuHwids              smbios property Family=ThinkPad P52s
18:06:13:0368 FuHwids              smbios property ProductName=20LB001FUS
18:06:13:0368 FuHwids              smbios property ProductSku=LENOVO_MT_20LB_BU_Think_FM_ThinkPad P52s
18:06:13:0368 FuHwids              smbios property BiosVendor=LENOVO
18:06:13:0368 FuHwids              smbios property BiosVersion=N27ET38W (1.24 )
18:06:13:0368 FuHwids              smbios property BiosMajorRelease=01
18:06:13:0368 FuHwids              smbios property BiosMinorRelease=18
18:06:13:0368 FuHwids              smbios property BaseboardManufacturer=LENOVO
18:06:13:0368 FuHwids              smbios property BaseboardProduct=20LB001FUS
18:06:13:0368 XbSilo               attempting to load /var/local/cache/fwupd/metadata.xmlb
18:06:13:0368 XbSilo               failed to load silo: Failed to open file “/var/local/cache/fwupd/metadata.xmlb”: open() failed: No such file or directory
18:06:13:0368 XbSilo               compiling /usr/local/share/fwupd/remotes.d/dell-esrt/metadata.xml:ctime=1619752634.40612:func-id=AppStreamUpgrade@3:value=/usr/local/share/fwupd/remotes.d/dell-esrt/metadata.xml:value=dell-esrt…
18:06:13:0369 FuEngine             1 components now in silo
18:06:13:0373 FuPlugin             init(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_thunderbolt.so)
18:06:13:0373 FuContext            added udev subsystem watch of thunderbolt
18:06:13:0373 FuEngine             plugins disabled: altos, dell, tpm, analogix, iommu, goodixmoc, dell_dock, msr, pci_mei, ebitdo, uefi_recovery, hailuck, logind, uefi_dbx, vli, logitech_hidpp, pci_bcr, steelseries, upower, optionrom, redfish, synaptics_mst, linux_sleep, bcm57xx, elantp, synaptics_prometheus, dfu, wacom_usb, linux_tainted, fastboot, dfu_csr, cros_ec, ata, rts54hub, nitrokey, acpi_dmar, bios, linux_swap, dell_esrt, thelio_io, colorhug, fresco_pd, nvme, tpm_eventlog, ccgx, superio, cpu, amt, system76_launch, emmc, ep963x, wacom_raw, uefi_capsule, uefi [... cut ...]
  Device Type:          Thunderbolt device controller
  Safe Mode:            false
  Native mode:          false
  Generation:           0
  AuthAttribute:

18:06:13:0422 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
18:06:13:0423 FuCommon             creating path /var/local/cache/fwupd/motd.d
WARNING: This package has not been validated, it may not work properly.
No detected devices
18:06:13:0423 FuMain               No detected devices
</pre>

azonenberg, 2021-05-03:
It seems like the controller isn't awake despite me having the TBT3 Ethernet dongle plugged into it?

The port works fine in USB mode, my phone is plugged in and charging on it right now and it recognizes the laptop as a debug host.

hughsie, 2021-05-04:
What's the output of `ls /sys/devices/pci0000:00/0000:00:1d.0/0000:07:00.0/0000:08:00.0/0000:09:00.0/domain0/0-0` -- is there a `generation` file there? if you cat the generation fiel is it really `0`?

azonenberg, 2021-05-04:
<pre>
# ls /sys/devices/pci0000:00/0000:00:1d.0/0000:07:00.0/0000:08:00.0/0000:09:00.0/domain0/0-0
power  subsystem  uevent  wakeup
</pre>

No "generation" file, with or without the Ethernet dongle attached.

hughsie, 2021-05-04:
Hmm, I just did this on a P53 on Fedora and it worked perfectly. Would it be possible for you to try with a Fedora 34 LiveUSB to rule out any Debian-specificness please?

azonenberg, 2021-05-14:
I'll see if I can find time to try that this weekend.

At this point my best guess is that this is some side effect of the P52S TBT firmware bugs and that my controller is already fried (thus nothing to do with fwupd).

hughsie, 2021-05-14:
> my best guess is that this is some side effect of the P52S TBT firmware bugs

I forgot all about that -- maybe?

closed 2021-06-26 by stale[bot]