fwupd/fwupd #3215: fwupdmgr does not support mirrors created from PULP_MANIFEST

zeromind, 2021-04-30:
**Describe the bug**
At least with `jcat` support, `fwupdmgr` updates the `MetadataURI` to use the Item's `Id` of the corresponding `.jcat`.
However, the `PULP_MANIFEST` of the official LVFS repo does not contain the "cdn-safe" `Id` referenced in the `.jcat` file, which makes mirrors created from that file not work with `fwupdmgr`.

```shell
$ wget https://cdn.fwupd.org/downloads/PULP_MANIFEST -q -O - | grep ^firmware
firmware.xml.gz,3a94ce1340f790608c9025cd61a6b3b3f6b0b6a0f41736c2b23c477bfa689393,601340
firmware.xml.gz.asc,567cda3a9cc237a92bcd435d99ceb8595d72848ef16b443da7ca17497db490dc,488
firmware.xml.gz.jcat,9b3e4074bfb017f722e84be23528f0051dfb5943bff41be398b369ee7e3c1d0f,2229
```

```shell
$ zcat /mirrors/lvfs/firmware.xml.gz.jcat | grep -oP '^.{200}'
{"JcatVersionMajor": 0, "JcatVersionMinor": 1, "Items": [{"Id": "firmware-01736-stable.xml.gz", "AliasIds": ["firmware.xml.gz"], "Blobs": [{"Kind": 4, "Flags": 1, "Timestamp": 1619761014, "Data": "ee1
```

Copying `firmware.xml.gz` to the filename from the `.jcat` (`firmware-01736-stable.xml.gz`) makes the mirror work with `fwupdmgr`.

**Steps to Reproduce**

1. Mirror LVFS repo using [sync-pulp.py](https://gitlab.com/fwupd/lvfs-website/-/blob/master/contrib/sync-pulp.py) via `./sync-pulp.py https://cdn.fwupd.org/downloads /mirrors/lvfs/` and serve it via HTTP
2. Configure client to use the newly created mirror
3. run `fwupdmgr refresh`: `fwupdmgr` tries to download `firmware-01736-stable.xml.gz` from the mirror, which it does not contain. Throws `Failed to update metadata for lvfs: checksum failure: failed to verify data, expected d64bae91091eda6a7532ebec06aa70893b79e1f8`

```shell
# fwupdmgr -v refresh
(fwupdmgr:23644): GLib-DEBUG: 17:44:20.150: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.191: Emitting ::status-changed() [idle]
Updating lvfs
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.201: downloading http://mirror.local/lvfs/firmware.xml.gz.jcat
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.201: Emitting ::status-changed() [downloading]
Downloading…             [ -                                     ](fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.234: download progress: 100%
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.234: download progress: 100%
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.234: download progress: 100%
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.234: download progress: 100%
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.234: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.235: changing metadata URI from http://mirror.local/lvfs/firmware.xml.gz to http://mirror.local/lvfs/firmware-01736-stable.xml.gz
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.236: downloading http://mirror.local/lvfs/firmware-01736-stable.xml.gz
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.236: Emitting ::status-changed() [downloading]
Downloading…             [***************************************](fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.255: download progress: 100%
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.255: download progress: 100%
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.255: download progress: 100%
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.256: download progress: 100%
(fwupdmgr:23644): Fwupd-DEBUG: 17:44:20.256: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected d64bae91091eda6a7532ebec06aa70893b79e1f8
```

**Expected behavior**
1. If `fwupd` enounters an `.jcat` file: try all possible sources for an Item, until one succeeds, something like; download from URIs based on the Item's `Id`, *and* each of its `AliasIds` until it get a response that matches the Item's checksum/signature. Fail if all possible URIs have been tried without success.
2. `PULP_MANIFEST` to reference *all* files required for a mirror, so that mirrors created from it work with previous and current versions of `fwupdmgr`.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.9
```
installed from the official Fedora 34 repos

**Additional questions**
- Operating system and version:
Fedora 34
- Have you tried rebooting?
no
- Is this a regression?
This is related to adding `jcat` support added for #391 via https://github.com/fwupd/fwupd/commit/a1de20665a90decd6b3c5699016dbda42acd4746, not sure whether that qualifies as a regression though.

hughsie, 2021-04-30:
Would you mind moving this issue here please: https://gitlab.com/fwupd/lvfs-website/-/issues -- I think it should be good to include the CDN-safe versions, but I'm not immediately sure how many I need to include. I *think* just the most current would be okay.

zeromind, 2021-04-30:
Can do, I'll have to sign up there first.
Adding the lastest one (matching the firmware.xml.gz) to the Pulp manifest as suggested, indeed.
But it still seems like `fwupdmgr` should try the aliases in addition to that, or am I misunderstanding what aliases are for?

hughsie, 2021-05-01:
Thanks, I'll work on that bug, and close this one. Thanks!

closed 2021-05-01 by hughsie