fwupd/fwupd #3217: Specified firmware is older than installed '0x9ed87a < 10410105'

sagitter, 2021-05-01:
**Describe the bug**

`fwupdmgr` shows an available update of my SSD driver but does not install it because says "**Specified firmware is older than installed**"

**Steps to Reproduce**

```
# fwupdmgr get-updates
Devices with no available firmware updates: 
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Inspiron 3501
│
└─KBG40ZNS512G NVMe KIOXIA 512GB:
  │   Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
  │   Summary:            NVM Express Solid State Drive
  │   Current version:    10410105
  │   Vendor:             KIOXIA Corporation (NVME:0x1E0F)
  │   Serial Number:      Y0MPGQFLQUN1
  │   GUIDs:              35de42d4-bd35-5bcd-89fa-6e71cbd4d0ed ← STORAGE-DELL-108349
  │                       5264a089-86d6-e811-9f8b-f2801f1b9fd1
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─BG4 Non SED Device Update:
        New version:      10410106
        Remote ID:        lvfs
        Summary:          BG4 NVMe SSD firmware for Dell PC
        License:          Proprietary
        Size:             815.1 kB
        Created:          2021-04-30
        Urgency:          Medium
        Vendor:           KIOXIA Corporation
        Flags:            is-upgrade
        Description:      
        Improve compatibility issue
      
# fwupdmgr update
Devices with no available firmware updates: 
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Upgrade available for KBG40ZNS512G NVMe KIOXIA 512GB from 10410105 to 10410106
Inspiron 3501 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Specified firmware is older than installed '0x9ed87a < 10410105'
```

**Expected behavior**

Firmware updated as proposed

**fwupd version information**

```
$ fwupdmgr --version
client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.9

```

Installed by DNF.

**fwupd device information**

```
$ fwupdmgr get-devices --show-all-devices
Inspiron 3501
│
├─11th Gen Intel(R) Core™ i7-1165G7 @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000066
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba
│                         1df99df5-22f4-5505-98cc-22dcc53b132a
│     Device Flags:       • Internal device
│   
├─KBG40ZNS512G NVMe KIOXIA 512GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    10410105
│     Vendor:             KIOXIA Corporation (NVME:0x1E0F)
│     GUIDs:              35de42d4-bd35-5bcd-89fa-6e71cbd4d0ed
│                         5264a089-86d6-e811-9f8b-f2801f1b9fd1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          08e848caed01b2fc9ee4ecb0ba4936827f0095b3
│ │   Current version:    66048
│ │   Minimum Version:    66048
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              5436e16d-4788-6c95-22d6-e5a0c1f9935a
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       7cb79bc9-7e91-55ff-929b-f1ecbdca20e7
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─TPM:
│ │   Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Summary:            TPM 2.0 Device
│ │   Current version:    600.7.0.0
│ │   Vendor:             Intel (TPM:INTC)
│ │   GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                       34801700-3a50-5b05-820c-fe14580e4c2d
│ │                       00d06c7f-2244-5a35-aa98-5d60704b3f5d
│ │                       03f304f4-223e-54f4-b2c1-c3cf3b5817c6
│ │                       9c3106c7-42ca-5099-9363-96ef5c22cbf7
│ │   Device Flags:       • Internal device
│ │ 
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:     • Internal device
│     
├─TigerLake GT2 [Iris Xe Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    01
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              3345b221-7679-5bf1-8ae1-e9865178e105
│                         2425c383-8884-57d6-a3bf-7e8aa25e26e4
│                         3c11d563-4f83-5bd1-8369-75359ea6b6f6
│                         54ff3ed7-1ea3-5279-975f-93c13bf2aa53
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─UEFI Device Firmware:
│     Device ID:          6b9d757c1f64d696e7af69942c2d6ed3993479d4
│     Current version:    102
│     Minimum Version:    102
│     Vendor:             DMI:Dell Inc.
│     GUIDs:              176e090d-0ddb-495e-8173-bc998ccfecd0
│                         3e1d83c5-31af-55ac-9f88-8be02bda9596
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          fa97fc7974f3a4408aafaafac6a63b4f090eae84
      Current version:    272695557
      Minimum Version:    272695557
      Vendor:             DMI:Dell Inc.
      GUIDs:              5264a089-86d6-e811-9f8b-f2801f1b9fd1
                          45023d5b-942b-54ae-8963-7f13e618b58c
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
```

**Additional questions**

- Operating system and version: Fedora 34

- Have you tried rebooting?
Yes.

hughsie, 2021-05-01:
I think the uploader made a mistake. I've fixed it manually. Can you wait 2 hours, then `fwupdmgr refresh --force` then try again please?

superm1, 2021-05-01:
Presumably you also notified uploader of the mistake?

hughsie, 2021-05-01:
> Presumably you also notified uploader of the mistake?

Yup.

sagitter, 2021-05-01:
> I think the uploader made a mistake. I've fixed it manually. Can you wait 2 hours, then `fwupdmgr refresh --force` then try again please?

```
# fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 1 local device supported

# fwupdmgr update
Devices with no available firmware updates: 
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Upgrade available for KBG40ZNS512G NVMe KIOXIA 512GB from 10410105 to 10410106
Inspiron 3501 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading…             [************************************** ]checksum invalid, expected 54968997d0b974f9eec2d2dc9fe20b7b16bec2c35183cb2c5373e4b35a81529c got b8cdbe4278eb281acf5572ab31547204fe5e15da6729d9defbf802b90bab8aec
```

sagitter, 2021-05-02:
My device is now updated and i have uploaded 1 report:

```
# fwupdmgr get-updates
Devices with no available firmware updates: 
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • KBG40ZNS512G NVMe KIOXIA 512GB
________________________________________________

Devices that have been updated successfully:

 • KBG40ZNS512G NVMe KIOXIA 512GB (10410105 → 10410106)
```

Thank you.

closed 2021-05-02 by superm1

leogama, 2021-08-28:
Dell messing up with NVMe SSD firmwares again, this time with an Intel drive (this is GNOME Software):

[image]

How can I (or we, Dell users in general) report these issues directly, at the right channel?

superm1, 2021-08-28:
> Dell messing up with NVMe SSD firmwares again, this time with an Intel drive (this is GNOME Software):
> 
> [image]
> 
> How can I (or we, Dell users in general) report these issues directly, at the right channel?

Can you please open a separate report with all the circumstances of your issue and follow the template?  Specifically we need to understand what `fwupdmgr get-devices` and `fwupdmgr get-updates` show to see what's really going on.