fwupd/fwupd #3219: A firmware update will not install on my Lenovo Thinkpad X1 Carbon 7th running Ubuntu 21.04

cdysthe, 2021-05-02:
The 'Ubuntu Software' application tells me me I have a firmware update available (screenshot below). I have installed many of those on this laptop without problems but this time when I click "Install" the update notification disappears and then shows up again as if I haven't done anything. I can keep clicking 'Install' but it keeps coming back after a couple of seconds. I then thought I would reboot and maybe the update would install before boot as it has in the past. Nothing, just a normal boot, but I notice a lockup/pause for a couple of seconds during shutdown I haven't noticed before.

I also tried using fwupd but It tells me I have an update, I choose to install it and it's being downloaded. Now I am getting a reboot y/n option so I reboot. Same thing, nothing other than the pause during shutdown. I boot normally again and still have the same update available right after I log in to my user account.

This is the update in question:

[image]

hughsie, 2021-05-02:
You didn't enter any of the details in the issue template.

cdysthe, 2021-05-03:
> You didn't enter any of the details in the issue template.

Sorry, it's all there now.

hughsie, 2021-05-04:
> Cannot update as UEFI Device Firmware [317fd9ef3f069f67047ab1163f8c040c3853cc38] needs reboot

Did you reboot and do the EC update? Or does the EC update not apply? Lenovo asked to limit the number of updates to 1 at a time on some hardware.

cdysthe, 2021-05-04:
I did reboot with just a normal boot, nothing is being applied or installed. When back up I'm told I have the exact same update available. I do not know much about this but I've completed many updates on this laptop without problems.

cdysthe, 2021-05-05:
I have looked at this some more. Two files are downloaded to /boot/efi/EFI/ubuntu/fw
```
fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
fwupd-1181d850-c3be-4b1d-985a-b28dca6ba6d8.cap
```
But they are not installed during reboot as I assume they should. The files has the exact same modified time as the last 'sudo fwupdmgr update' was run and I can see them being downloaded on the console when running fwupdmgr. So the problem is that they are not installed for some reason. As a layman I have no idea why. I did do an update from Ubuntu 20.10 to 21.04 a week or so ago. Could that have anything to do with it?

hughsie, 2021-05-07:
> No updates to process. Called in error

Somehow the FWUPDATE_ATTEMPT_UPDATE flag isn't being set. Can you describe exactly what you did please?

johnnyapol, 2021-05-07:
`fwupdmgr refresh && fwupdmgr update`
After accepting the prompts to download the updates then I reboot.

hughsie, 2021-05-07:
@johnnyapol does the same behaviour happen with Fedora 34? You've got a weird failure and I want to rule out software before we dive into firmware debugging. Thanks.

cdysthe, 2021-05-10:
@johnnyapol I "fixed" it this morning by deleting the two .cap files. Removed the fwupd packages using apt with 'sudo apt purge package'. I rebooted and installed fwupd again. Did the update and they were installed normally during boot. I have no idea why this resolved the issue. The only change to my system has been the upgrade from Ubuntu 20.10 to 21.04.

saikocat, 2021-05-13:
I have a similar issues on my T480s running latest Arch and the suggested workaround didn't work.

- Deleting 2 caps file didn't help in my case (first cap file is the same, 2nd one is different from cdysthe). Reinstalled package and reboot and retry didn't help.
- `fwupdate -L` will fail with `failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory`
- When prompted to update and restart, the bootscreen will flash `Reset System` then just a blank screen and hang there infinitely. Press and hold the power button didn't shutdown the system, and I have to use a pin to activate the Emergency Reset Hole. And a last update log I found is like below:

```
"TpmEventLog" : "0x00000008 bef593cfe02e26ff49b0659f42a0fb5c670cb065 [N.2.2.E.T.6.5.W. ...]\n0x80000008 4ce2e18338a06fd390f856ce7217fd32acaa63d1\n0x80000008 5c0f8b87ba4ca7b89918e75b81f43baa259df79b\n0x80000008 89d6094a189cd610dfb89cb1df39bad84232d59f\n0x80000008 0c6a03ef6f51b4273bde90be83473642558a2b53 [..........C.....]\n0x00000001 3666adabd611f55bd0fc8251bb1671c51804d971 [ACPI DATA]\n0x00000001 b0469aa139a98ebeae9693588554bab11e19724c [ACPI DATA]\n0x00000004 9069ca78e7450a285173431b3e52c5c25299e473\nPCR0: 91bb68cd5712bc990739af5ce9c0fcc6285705ce\nPCR0: f139f1dc9b1df88b482d9f6e112c01dcf4920456bf9c02945c81c5460ae9b08c",
```

If needed I can create a separate issue.

LinuxOnTheDesktop, 2021-05-15:
Locating and deleting the one .cap file I found did not help with the problem I have (#3231) either.

Issen007, 2021-05-20:
I have the same issue with Fedora 34 and Lenovo X390.

Delete All firmware updates
```
$ cd /boot/efi/EFI/fedora/fw
$ ls
fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap	fwupd-6300ba52-d9ea-4ca0-88f9-f89ef73cc07b.cap
$ rm *
```

Uninstall fwupd
`$ sudo dnf remove fwupd`

Deleting old EFI Boot Option
```
$ sudo efibootmgr --bootnum 0000 --delete-bootnum
$ sudo reboot
```

Reinstalling fwupd and update the bios one more time.
```
$ sudo dnf install fwupd
$ sudo fwupdmgr update ca8e899e46f1da1776d16410bc92f2cf0d963433
20Q00056MX must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[***************************************]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]:
```

Verify that nextboot should be Linux-Firmware-Updater and then reboot
```
$ sudo efibootmgr 
BootNext: 0000
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0001,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0011,0000
Boot0000* Linux-Firmware-Updater
Boot0001* Fedora
Boot0010  ThinkShield secure wipe
Boot0011* LENOVO CLOUD
Boot0013  Setup
Boot0014  Boot Menu
Boot0015  Diagnostic Splash Screen
Boot0016  Lenovo Diagnostics
Boot0017  Regulatory Information
Boot0018  Startup Interrupt Menu
Boot0019  Rescue and Recovery
Boot001A  MEBx Hot Key
Boot001B* USB CD
Boot001C* USB FDD
Boot001D* NVMe0
Boot001E* NVMe1
Boot001F* ATA HDD0
Boot0020* ATA HDD1
Boot0021* USB HDD
Boot0022* PXE BOOT
Boot0023  Other CD
Boot0024  Other HDD
Boot0025* IDER BOOT CDROM
Boot0026* IDER BOOT Floppy
Boot0027* ATA HDD
Boot0028* ATAPI CD

$ sudo reboot
```

Verify boot after rebooting the system and it tells me it has bootup the Firmware boot but it hasn't update the firmware. 

```
$ sudo efibootmgr 
BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0001,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0011,0000
Boot0000* Linux-Firmware-Updater
Boot0001* Fedora
```